solectrus/solectrus #3754: EV wird immer als "verbunden" angezeigt

gereons, 2024-12-07:
### Beschreibung (auf Deutsch oder Englisch)

Mit PR #3747 wird mir mein Auto immer als "verbunden" angezeigt, obwohl in der InfluxDB korrekt "ev_connected: false" ankommt.

### Welche Version von SOLECTRUS verwendest du?

pr-3747

### Auf welchem Gerät läuft bei dir SOLECTRUS?

cloud

StevieC121176, 2024-12-07:
Also mit dem neuesten Update von #3747 funktioniert es bei mir.

ledermann, 2024-12-07:
Ja, danke für den Hinweis, das ist mit `custom-consumers-74` behoben.

commit 6daddbe7 (2024-12-07): Fix wallbox_car_connected icon

Resolves #3754
closed 2024-12-07 by ledermann