solectrus/solectrus #3756: Gehäuse Temperatur wird nicht mehr angezeigt (#3747)

unclej84, 2024-12-07:
### Beschreibung (auf Deutsch oder Englisch)

Auf der "Aktuell"-Ansicht fehlt über dem Ladestand die Temperatur des SENEC Gehäuses - im Diagramm wird eine korrekte Temperatur angezeigt

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-74-g98787099

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Docker

### Gibt es sonst noch etwas?

[image]

unclej84, 2024-12-07:
Related to #3747

unclej84, 2024-12-07:
Ich habe auch die Vermutung, dass der Ladestand nicht korrekt angezeigt werden könnte (bleibt auf 0%) - kann ich aber gerade nicht testen aufgrund fehlender Sonne (fiel mir gestern auf als ich das Image schon einmal getestet hatte - war deshalb temporär auf 3727 gewechselt)

ledermann, 2024-12-07:
Sorry, einige Sensoren fehlten in der Strombilanz, das ist mittlerweile aber korrigiert (in `custom-consumers-75`). Der Ladestand ist meines Erachtens aber korrekt - momentan auch bei mir 0% und keine Sonne.

closed 2024-12-07 by ledermann

gereons, 2024-12-07:
@unclej84 Ladestand konnte ich heute morgen dank Notladung beobachten - der funktioniert 👍

commit 5e8b087a (2024-12-07): Fix missing case_temp

Closes #3756