solectrus/solectrus #3768: Verbrauchsaufteilung wird nicht angezeigt, wenn Hausverbrauch kleiner ist

patricknitsch, 2024-12-12:
### Beschreibung (auf Deutsch oder Englisch)

In der neuen Beta-Version habe ich viele Verbraucher die hohe Leistungen ziehen. Diese lasse ich aus dem Verbrauch herausrechnen, um zu sehen, was am Ende noch anderweitig verbraucht wird.
Die Ansicht funktioniert dann aber nicht, da die anderen Verbräuche höher sind, als der Hausverbrauch

[image]

[image]

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-96

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Proxmox

ledermann, 2024-12-12:
Ich habe in diesem Zusammenhang einen Fehler korrigiert, der zu einer falschen Berechnung der Summe von "Andere" geführt hat und dann Folgefehler nach sich zog. Zu sehen in `custom-consumers-100` (bitte updaten).

Ich bin mir jedoch nicht sicher, ob das auch das von dir beschriebene Problem löst. Insbesondere irritiert mich, dass in deinen Screenshots das Ausrufezeichen-Icon zu sehen ist. Dies weist auf ein Problem hin, das **außerhalb** von SOLECTRUS liegt. Je nach Setup ist es denkbar, dass der gemeldete Hausverbrauch **kleiner** ist als die Summe der einzeln gemessenen Verbraucher. 

Dies ist beispielsweise bei meinem SENEC-Speicher der Fall, wenn die Wallbox in Benutzung ist. Während dieser Zeit meldet der SENEC-Speicher oftmals einen falschen Hausverbrauch von 0, die (unabhängig davon gemessenen) Einzelverbraucher melden aber weiter einen positiven (korrekten) Wert - es stimmen dann einige Berechnungen nicht und SOLECTRUS signalisiert das über dieses Icon. Einen Korrekturversuch unternimmt SOLECTRUS **nicht**.

Falls das Problem bei dir auch nach einem Update weiterhin auftritt, müssen wir das genauer untersuchen. Zunächst solltest du prüfen, ob es einen Grund außerhalb von SOLECTRUS gibt, der zu einem zu geringen (und somit falschen) Hausverbrauch führt.

patricknitsch, 2024-12-12:
Jetzt ist das Ausrufezeichen weg und er zeigt Andere an. Die anderen aktiven Verbraucher werden aber nicht angezeigt. Liegt das daran, dass diese aus dem Hausverbrauch herausgerechnet sind?

[image]

ledermann, 2024-12-12:
Das sieht nun korrekt aus. Ja, wenn ein Verbraucher aus dem Hausverbrauch herausgerechnet werden soll, ist er nicht mehr in dieser Darstellung enthalten. Meiner Ansicht sinnvoll, denn es wird hier ja der Hausverbrauch angezeigt.

Wenn du keine weitere (Klein-)Verbraucher hast, bringt diese Darstellung natürlich keinen Mehrwert für dich, aber du hast dafür die selbstdefinierten (Groß-)Verbraucher direkt in der Strombilanz.

patricknitsch, 2024-12-12:
Das passt für mich. Kann dann hier geschlossen werden.

closed 2024-12-12 by patricknitsch