solectrus/solectrus #3769: Photovoltaik % übersteigt 100% bei Einspeisung

unclej84, 2024-12-12:
### Beschreibung (auf Deutsch oder Englisch)

ich weiß nicht, ob es am Powersplitter oder Summarizer liegt, aber teilweise wird der Photovoltaik-Anteil mit mehr als 100% angezeigt:

[image]

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-100-g19eb3b95

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Docker

unclej84, 2024-12-12:
[image]

unclej84, 2024-12-12:
Betrifft die Ansichten Woche, Monat, Jahr und Gesamt

ledermann, 2024-12-12:
Da wir kürzlich bei dir den PowerSplitter in mehreren Runden (https://github.com/solectrus/power-splitter/issues/157, https://github.com/solectrus/power-splitter/issues/156, https://github.com/solectrus/power-splitter/issues/155) korrigiert haben, könnten dadurch defekte PowerSplitter-Berechnungen entstanden sein.

Ich schlage daher zunächst zwei Maßnahmen vor:

1) Sichererstellen, dass der PowerSplitter in der aktuellen `develop`-Fassung wirklich alle Tage korrekt berechnet hat, im Zweifelsfall einen erneuten Durchlauf starten (mit `docker compose kill --signal USR1 power-splitter`)
2) Danach: Tageszusammenfassungen zurücksetzen (in den Einstellungen).

Unabhängig davon bin ich derzeitig noch dabei, den Summarizer für die neuen Custom-Sensoren anzupassen. Ich bin mir gerade nicht sicher, ob die noch ausstehende Anpassung zu dem von dir beschriebenen Artefakt führen kann. Falls ja, dann wären die beiden Schritte unnötig bzw. werden nicht helfen. Ich hoffe, die Anpassung in Kürze fertigstellen zu können.

unclej84, 2024-12-12:
Hab es gerade mal versucht mit den beiden Schritten, aber es ist unverändert - ich hoffe die Anpassung richtet es dann

ledermann, 2024-12-13:
Da ich das noch nicht reproduzieren kann, muss ich nochmal nachhaken:

1) Stehen den beiden Docker-Containern für PowerSplitter und Dashboard wirklich die gleichen Sensoren zur Verfügung? Es ist wichtig, dass beide ihre Berechnungen auf der gleichen Grundlage erledigen können, also über alle Sensoren Bescheid wissen (`INFLUX_SENSOR_*`) und insbesondere auch wissen, was aus dem Hausverbrauch herausgerechnet werden muss (`INFLUX_EXCLUDE_FROM_HOUSE_POWER`).

2) Nach einer Änderung der Berechnungsgrundlage (also den genannten Variablen) muss zwingend sowohl der Power-Splitter erneut rechnen (alles im Measurement `power_splitter` wird ungültig) und danach, wenn der PowerSplitter fertig ist, müssen die Tageszusammenfassungen neu berechnet werden. Ist das wirklich so auch passiert?

Nebenbei bemerkt: Das offenbart ein grundsätzliches Problem: Die Variable `INFLUX_EXCLUDE_FROM_HOUSE_POWER` darf man nicht so einfach ändern, weil dadurch alle Berechnungen ungültig werden. Hierzu habe ich leider keine bessere Idee - der Power-Splitter muss wissen, welcher Verbraucher wie viel Strom gezogen hat. Außerdem gibt es dadurch eine gewisse Inkonsistenz in InfluxDB: Im Field `house_power` sind die heraus zu rechnenden Sensoren **enthalten**, im Field `house_power_grid` hingegen **nicht**. Auch hierfür sehe ich keine andere Möglichkeit. Für den Enduser ist das natürlich unbedeutend.

Derzeitig berechnet das Dashboard nicht den PV/Netz-Anteil der selbstdefinierten Sensoren. Das ist das, was ich noch anpassen muss. Vorher würde ich aber gerne dieses Problem hier geklärt haben.

unclej84, 2024-12-13:
Kein Thema - bin gerne bereit soweit wie möglich bei der Ursachenforschung zu helfen.
Beide Container sind in der selben Compose-Datei definiert und verwenden die gleiche .env Datei. 
Hier einmal die Auszüge für die environment Blöcke:
Dashboard:
`    environment:
      ADMIN_PASSWORD:
      APP_HOST:
      CO2_EMISSION_FACTOR:
      DB_HOST: ${POSTGRES_HOST}
      DB_PASSWORD: ${POSTGRES_PASSWORD}
      DB_USER: ${POSTGRES_USERNAME}
      FORCE_SSL:
      # FRAME_ANCESTORS:
      <<: *environment_influxdb
      INFLUX_BUCKET:
      INFLUX_EXCLUDE_FROM_HOUSE_POWER:
      INFLUX_MEASUREMENT:
      INFLUX_POLL_INTERVAL:
      INFLUX_SENSOR_BATTERY_CHARGING_POWER:
      INFLUX_SENSOR_BATTERY_DISCHARGING_POWER:
      INFLUX_SENSOR_BATTERY_SOC:
      INFLUX_SENSOR_CAR_BATTERY_SOC:
      #INFLUX_SENSOR_CAR_MILEAGE:
      INFLUX_SENSOR_CASE_TEMP:
      # Custom sensors
      INFLUX_SENSOR_CUSTOM_POWER_01:
      INFLUX_SENSOR_CUSTOM_POWER_02:
      INFLUX_SENSOR_CUSTOM_POWER_03:
      INFLUX_SENSOR_CUSTOM_POWER_04:
      INFLUX_SENSOR_CUSTOM_POWER_05:
      INFLUX_SENSOR_CUSTOM_POWER_06:
      INFLUX_SENSOR_CUSTOM_POWER_07:
      INFLUX_SENSOR_CUSTOM_POWER_08:
      INFLUX_SENSOR_CUSTOM_POWER_09:
      INFLUX_SENSOR_CUSTOM_POWER_10:
      INFLUX_SENSOR_GRID_EXPORT_LIMIT:
      INFLUX_SENSOR_GRID_EXPORT_POWER:
      INFLUX_SENSOR_GRID_IMPORT_POWER:
      #INFLUX_SENSOR_HEATPUMP_HEATING_POWER:
      #INFLUX_SENSOR_HEATPUMP_LEAVING_TEMP:
      INFLUX_SENSOR_HEATPUMP_POWER:
      #INFLUX_SENSOR_HEATPUMP_SCORE:
      #INFLUX_SENSOR_HEATPUMP_STATUS:
      INFLUX_SENSOR_HOUSE_POWER:
      INFLUX_SENSOR_INVERTER_POWER_FORECAST:
      INFLUX_SENSOR_INVERTER_POWER:
      INFLUX_SENSOR_OUTDOOR_TEMP:
      INFLUX_SENSOR_SYSTEM_STATUS_OK:
      INFLUX_SENSOR_SYSTEM_STATUS:
      INFLUX_SENSOR_WALLBOX_CAR_CONNECTED:
      INFLUX_SENSOR_WALLBOX_POWER:
      INFLUX_TOKEN:
      INSTALLATION_DATE:
      REDIS_URL:
      SECRET_KEY_BASE:
      TZ:
      WEB_CONCURRENCY:`

Power-Splitter:
`    environment:
      INFLUX_BUCKET:
      INFLUX_EXCLUDE_FROM_HOUSE_POWER:
      INFLUX_HOST:
      INFLUX_ORG:
      INFLUX_PORT:
      INFLUX_SCHEMA:
      INFLUX_SENSOR_BATTERY_CHARGING_POWER:
      # Custom sensors
      INFLUX_SENSOR_CUSTOM_POWER_01:
      INFLUX_SENSOR_CUSTOM_POWER_02:
      INFLUX_SENSOR_CUSTOM_POWER_03:
      INFLUX_SENSOR_CUSTOM_POWER_04:
      INFLUX_SENSOR_CUSTOM_POWER_05:
      INFLUX_SENSOR_CUSTOM_POWER_06:
      INFLUX_SENSOR_CUSTOM_POWER_07:
      INFLUX_SENSOR_CUSTOM_POWER_08:
      INFLUX_SENSOR_CUSTOM_POWER_09:
      INFLUX_SENSOR_CUSTOM_POWER_10:
      INFLUX_SENSOR_GRID_IMPORT_POWER:
      INFLUX_SENSOR_HEATPUMP_POWER:
      INFLUX_SENSOR_HOUSE_POWER:
      INFLUX_SENSOR_WALLBOX_POWER:
      INFLUX_TOKEN: ${POWER_SPLITTER_INFLUX_TOKEN}
      POWER_SPLITTER_INTERVAL:
      REDIS_URL:
      TZ:`

Soweit ich sehe sind die Sensoren identisch - hoffe ich habe keinen irgendwo vergessen.

Die Berechnungen hatte ich genau in der Reihenfolge durchgeführt: Power-Splitter zurücksetzen (Influx Measurement war leer) und neu berechnen lassen bis er wieder in den Normalbetrieb gewechselt ist. Danach dann im Dashboard über die Einstellungen die Tageswerte zurückgesetzt und über den Aufruf von Gesamt neu erstellen lassen.

Interessanterweise tritt der Effekt ja schon 2022 auf wo für die Custom-Sensoren noch keine Daten erfasst wurden - diese sind erst seit ca. September 2023 vorhanden.

Bei Bedarf könnte ich einen Datenextrakt eines Tages mit diesem Verhalten zur Verfügung stellen

ledermann, 2024-12-13:
Ok, vielen Dank! Mich irritiert zwar etwas, dass die Variablen alle keine Werte enthalten, aber das ist vermutlich nur ein Abbild der `compose.yaml`. In diesem Zusammenhang: Ich finde den Befehl `docker-compose config` ganz praktisch, weil man damit eine Auflistung der Variablen mit ihren tatsächlichen Werten erhält. Man merkt dann, ob die `.env` korrekt eingelesen und alles korrekt zugeordnet wurde.

> Interessanterweise tritt der Effekt ja schon 2022 auf wo für die Custom-Sensoren noch keine Daten erfasst wurden - diese sind erst seit ca. September 2023 vorhanden.

Das ist wirklich auffällig. Dann kann es mit den Custom-Sensoren eigentlich nichts zu tun haben. Jetzt verstehe ich aber noch weniger, was da schiefgelaufen ist.

Seit wann tritt das Problem eigentlich auf? Wurde das Jahr 2022 kürzlich noch korrekt angezeigt? Ist es erst durch die kürzlichen Änderungen des Power-Splitters entstanden?

Außerdem auffällig: Da stehen Variablen `HEATPUMP_*`. Wo kommen die hier? Hattest du mal #3732 in Betrieb? Das war ein experimenteller Branch, den ich nicht zu installieren empfohlen hätte. Möglicherweise hast du jetzt PostgreSQL in einem Stand, der früher oder später Probleme bereiten wird, denn der Branch hatte Datenbankänderungen vorgenommen, die man jetzt nicht so einfach rückgängig machen kann. Ob das eine Ursache für die falsche Darstellung in 2022 ist, sehe ich momentan aber nicht.

Falls möglich, würde ich dennoch empfehlen, die PostgreSQL-Datenbank vollständig zurückzusetzen. Du müsstest dann allerdings die Einstellungen manuell wieder eintragen (insbesondere die Strompreise) und die Instanz auch wieder registrieren. Wenn das für dich tragbar ist, ginge das so:

1. `cd solectrus`
2. `docker-compose down`
3. `mv postgresql postgresql.bak`
4. `mkdir postgresql`
5. `docker-compose up -d`

Anschließend Einstellungen und Strompreise eintragen, Instanz registrieren (mit gleicher E-Mail-Adresse) und dann auf "Gesamt" klicken und die Berechnung der Tageszusammenfassungen abwarten.

Dann sind wir vermutlich zwar keinen Schritt weiter, aber du hättest die PostgreSQL-Datenbank wieder in einem definierten Zustand und das würde als mögliche Ursache wegfallen.

Ich werde derweil weiter darüber nachdenken, wie das Problem entstanden sein könnte.

Und ja, ein Auszug aus InfluxDB als CSV eines einzelnen Tages (wo das Problem auftritt, also aus 2022) wäre sehr nützlich.

unclej84, 2024-12-13:
Ja genau, das war der Auszug aus der `compose.yaml` - da sie ja aus der gleichen `.env` geht es ja nur um die Namen. Dass die Werte richtig ankommen hatte ich nach dem Start des Splitters geprüft - am Anfang listet er ja auf was er findet - das passte soweit.

Ich denke es hatte mit der letzten Version 0.5.3 noch richtig funktioniert - könnte ich aber noch einmal testen, wenn es hilft?

Ja, ich hatte #3732 ausprobiert. Die Änderungen aus dem Migrationsfile hatte ich in der Datenbank aber wieder alle Rückgängig gemacht - es waren ja nur neue Felder hinzugekommen, die ich einfach in der Struktur gelöscht habe bevor ich auf #3747 gewechselt bin. Sollte eigentlich kein Problem darstellen damit. Habe die Variablen dann einfach erstmal auskommentiert drin gelassen, wenn die Wärmepumpensachen dann kommen.

Hier ein Extrakt vom 2.10.2022
[data.csv](https://github.com/user-attachments/files/18128474/data.csv)

unclej84, 2024-12-13:
Habe die Datenbank zurückgesetzt und in den Einstellungen wieder alles eingetragen, allerdings, wird das Sponsoring noch vor der Registrierung als aktiv angezeigt und ich bekomme beim Klick auf Registrierung einen Fehler 500:

`E, [2024-12-13T17:31:00.705988 #1] ERROR -- : [192.168.30.49]
[192.168.30.49] ArgumentError (ArgumentError):
[192.168.30.49]
[192.168.30.49] app/services/magic_id.rb:3:in `encode'
[192.168.30.49] app/controllers/registration_controller.rb:39:in `magic_id'
[192.168.30.49] app/controllers/registration_controller.rb:35:in `registration_url'
[192.168.30.49] app/controllers/registration_controller.rb:8:in `show'
[192.168.30.49] app/controllers/concerns/auto_locale.rb:14:in `switch_locale'
I, [2024-12-13T17:31:00.834265 #1]  INFO -- : [192.168.30.49] method=GET path=/500 format=html controller=ErrorsController action=show status=500 allocations=10364 duration=123.86 view=121.38 db=0.00
I, [2024-12-13T17:31:00.873460 #1]  INFO -- : method= path= format= params={} controller=ApplicationCable::Connection action=disconnect status=200 allocations=17 duration=0.27`

ledermann, 2024-12-13:
Oh, du findest zielsicher alle Bugs ;-) Ja, `setup_id` und `setup_token` eintragen sollte helfen. Ansonsten Redis zurücksetzen, das hatte ich leider vergessen zu erwähnen. Das wäre bei Schritt 3 noch zu ergänzen gewesen. Die Datenbank war weg, aber der Cache noch da.

unclej84, 2024-12-13:
Das passiert mir öfter - bin Fluch und Segen zugleich für jeden Software-Entwickler :D Das mit dem Cache fiel mir auch gerade ein, aber das wird der Power-Splitter ja für mich gleich übernehmen.
Habe gerade nochmal die `0.5.3` gestartet (natürlich ohne die Custom im Exclude). Werde das prüfen und danach auch ohne Exclude die `develop` nochmal laufen lassen. Wenn die beide passen sollten, muss es am Exclude der Customs liegen - ich werde berichten. Wird aber wegen der HDD etwas dauern leider

unclej84, 2024-12-13:
Nach meinen Versuchen kann ich jetzt sagen, dass mit `INFLUX_EXCLUDE_FROM_HOUSE_POWER=` bei der `0.5.3` alles korrekt dargestellt wird, die `develop` allerdings den Anzeige-/Berechnungsfehler hat. Es wurden zwischen den Versuchen keine Konfigurationsänderungen durchgeführt und auch immer die Tageswerte, nach der Berechnung der historischen Werte, zurückgesetzt

unclej84, 2024-12-14:
Mit dem letzten Fix scheint es richtig bzw. besser zu sein:
[image]

ledermann, 2024-12-14:
Besten Dank für deine Mühe, das war sehr hilfreich. Deinen Datenauszug konnte ich zwar nicht in InfluxDB importieren (falsches Format), aber bei manueller Betrachtung ist mir aufgefallen, dass `house_power_grid` zeitweise negativ (!) war, was natürlich nicht sein darf. Zur gleichen Zeit wurde die Batterie beladen.

Somit ist die Ursache klar: Mit den Custom-Sensoren hat es gar nichts zu tun, sondern mit einer kleinen Änderung, die ich vorher (in `develop`) gemacht habe: 3fbfe85ed7550a6e23e8acb6d866493a83111964

Da wollte ich eigentlich nur den Fall abdecken, wenn eine Beladung der Batterie aus dem Netz erfolgt - dann soll nämlich dieser Teil nicht den Verbrauchern zugerechnet werden. Leider hatte ich das fehlerhaft implementiert und nicht geprüft, ob überhaupt ein positiver Rest verbleibt. Dies ist nun korrigiert.

Somit empfehle ich nun Folgendes zu tun:

- Update des PowerSplitters auf den aktuellen Stand von `develop` (`docker-compose pull`) und neu starten (`docker-compose up -d`)
- Neuberechnung auslösen mit `docker-compose kill --signal USR1 power-splitter`
- Warten, bis der PowerSplitter fertig ist
- Tageszusammenfassungen zurücksetzen

Anschließend können wir das Issue hoffentlich schließen :)

unclej84, 2024-12-14:
Es hat ein wenig gedauert alles neu zu berechnen, aber scheint jetzt zu passen. Vielen Dank für die Hilfe. 

Was mir gerade noch aufgefallen ist, dass die Zahlen zwischen dem Tooltip links und der Anzeige rechts nicht zusammenpassen für den Aktuellen Tag:

[image]
[image]

ledermann, 2024-12-15:
Super, dank deiner Unterstützung sind wir hier durch. Die Abweichung beim PV-Anteil ist ein anderes Problem, dafür habe ich ein neues Issue angelegt.

closed 2024-12-15 by ledermann