solectrus/solectrus #3791: Hausverbrauch kleiner als Summe Verbraucher

q2doc, 2024-12-16:
### Beschreibung (auf Deutsch oder Englisch)

Bei der aktuellen Beta is bei mir ein einzelner Verbraucher größer als der Hausverbrauch:

[image]

Die Daten kommen per MQTT aus dem ioBroker, da der Shellie trotz "...#...." nicht direkt gefunden wird.

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-129-g046c221a

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspberry Pi5 docker

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
services:
  dashboard:
    image: ghcr.io/solectrus/solectrus:pr-3747 #next
    depends_on:
      postgresql:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - postgresql
      - influxdb
      - redis
    ports:
      - '3000:3000'
    environment:
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - WEB_CONCURRENCY
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - DB_HOST=postgresql
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_POLL_INTERVAL
      - INFLUX_SENSOR_INVERTER_POWER
      - INFLUX_SENSOR_HOUSE_POWER
      - INFLUX_SENSOR_GRID_IMPORT_POWER
      - INFLUX_SENSOR_GRID_EXPORT_POWER
      - INFLUX_SENSOR_BATTERY_CHARGING_POWER
      - INFLUX_SENSOR_BATTERY_DISCHARGING_POWER
      - INFLUX_SENSOR_BATTERY_SOC
      - INFLUX_SENSOR_WALLBOX_POWER
      - INFLUX_SENSOR_CASE_TEMP
      - INFLUX_SENSOR_INVERTER_POWER_FORECAST
      - INFLUX_SENSOR_SYSTEM_STATUS
      - INFLUX_SENSOR_SYSTEM_STATUS_OK
      - INFLUX_SENSOR_GRID_EXPORT_LIMIT
      - INFLUX_SENSOR_HEATPUMP_POWER
      - INFLUX_SENSOR_CAR_BATTERY_SOC #Auto Ladestand
      - INFLUX_SENSOR_WALLBOX_CAR_CONNECTED #Auto angeschlossen?
      - INFLUX_EXCLUDE_FROM_HOUSE_POWER
      - INFLUX_SENSOR_CUSTOM_POWER_01
[... 623 lines cut ...]
# Set these names before first run, they can't be changed later!
INFLUX_BUCKET=my-solectrus-bucket

# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE= xxx
INFLUX_TOKEN_READ= xxx

# Volume path for storing the InfluxDB data
INFLUX_VOLUME_PATH=xxx/solectrus/influxdb

##################################################################
###                    PostgreSQL database                     ###
##################################################################

# Volume path for storing the PostgreSQL data
DB_VOLUME_PATH=xxx/solectrus/postgresql

##################################################################
###                 Redis in-memory database                   ###
##################################################################

# Volume path for storing the database snapshot
REDIS_VOLUME_PATH=xxx/solectrus/redis
```

q2doc, 2024-12-16:
Bei der Betrachtung der influxDaten ist mir dann gerade folgendes aufgefallen:

[image]

es scheint also ein Problem mit dem Nullwert zu sein, wobei ich das Phänomen nur hier sehe...

ledermann, 2024-12-16:
Ok, die Betrachtung in InfluxDB ist sehr hilfreich. Offenbar kann es bei deinem Herd passieren, dass er nicht bei 0 startet, sondern bei einem hohen Wert. Gibt es beim Herd Besonderheiten gegenüber anderen Sensoren? Womit wird der Herd gemessen? 

Bei der Berechnung der Tagessumme verwendet SOLECTRUS die Integral-Funktion. Hier könnte ich ansetzen, um eine andere Interpolation zu erreichen und solche Artefakte auszugleichen.

Ich habe jetzt mal probeweise die Query für InfluxDB geändert, sodass fehlende Werte explizit auf 0 gesetzt werden. Du könntest ausprobieren, ob das bei dir das Problem löst. 

Vorgehensweise:

1. Update auf `custom-consumers-139`
2. Unter "Einstellungen" die Tageszusammenfassungen zurücksetzen (Neuaufbau könnte ein wenig länger dauern als bisher)

Mich würde interessieren, wie anschließend die Darstellung des 15. Dezembers aussieht. Wenn es nicht hilft, müsste ich tiefer eintauchen. Hilfreich wäre dann, wenn ich deine Messwerte von diesem Tag erhalten könnte (als CSV).

q2doc, 2024-12-16:
Danke! Kommt von einem Shelly 3pm über MQTT zu ioBroker und dann über Javascript an den MQTT Collector. Das Problem mit den Nullwerten hatte ich anfangs auch bei der Temperatur (Lücken, altes Issue), hier hat das Deaktivieren in der MQTT Instanz/ioBroker "senden nur bei Änderungen" geholfen. Die 139 hat leider keine Änderung gebracht:

[image]

Für die CSV  muss ich erst die Influx zurück auf 2.6 setzen (fehlt gerade der ruhige Moment). Melde mich...  einer möglichen Ursache  gehe ich auch noch nach (weil es mir komisch vorkommt das heute alles OK ist), der Pi hat ziemlich genau dazu dem Zeitpunkt wo die Steigung beginnt ein BackUp gemacht und dafür alle DockerContainer geschlossen - ob es das ist sehe ich kommenden Sonntag.

ledermann, 2024-12-16:
Hm, hast du wirklich auch mit der 139 die Tageszusammenfassungen zurückgesetzt und **danach** den 15. Dezember aufgerufen?

Bezüglich CSV: Zurück auf InfluxDB v2.6 ist nicht unbedingt erforderlich, es gibt eine andere Möglichkeit, siehe hier:
https://docs.solectrus.de/wartung/datenkorrektur#2-messwerte-als-csv-exportieren

Docker-Container beendet? InfluxDB sollte nonstop erreichbar sein. Ein Backup, das dazu den Raspi herunterfährt oder Container stoppt, würde ich eher nicht empfehlen. Das hier wäre eine Alternative, wenn es nur um SOLECTRUS geht:
https://docs.solectrus.de/wartung/datensicherung

q2doc, 2024-12-17:
Ja, zurückgesetzt, aber zur Sicherheit noch einmal und mit der 145:
[image]

[image]

und mit folgendem Ergebnis:

[image]

CSV folgt. BackUp ist kompliziert, da läuft einiges auf dem Pi... und ohne Beenden der Container bekomme ich ACL_TYPE_ACCESS Fehler.

q2doc, 2024-12-17:
Die influx Daten für den Herd: 
[2024-12-17_19_37_influxdb_data.csv](https://github.com/user-attachments/files/18169952/2024-12-17_19_37_influxdb_data.csv)

ledermann, 2024-12-18:
Ok, die Sache ist kompliziert. Was mir in der CSV-Datei auffällt:

1) Es sind negative Werte enthalten - zwar in geringer Höhe (bis zu -6W), aber das ist schon etwas seltsam. Das wäre ein Hinweis darauf, die Art der Messung zu überprüfen.

2) Das CSV enthält bereits aggregierte Daten, also nicht die Originalmesswerte. Es wurde beim Exportieren offenbar ein Zeitfenster von 10min gewählt. InfluxDB macht das automatisch, wenn man den QueryBuilder benutzt. Um an die Originalwerte heranzukommen, muss man den ScriptEditor verwenden, z.B. so:
  
    ```
    from(bucket: "my-solectrus-bucket")
    |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
    |> filter(fn: (r) => r["_measurement"] == "Herd")
    |> filter(fn: (r) => r["_field"] == "power")
    ```
    Insbesondere sollte kein `aggregateWindow` enthalten sein. 

3) Es sind große Lücken enthalten (ok, der Herd war dann wohl aus) und nach einer Lücke geht es direkt auf hohem Niveau los. Hier die entscheidende Stelle:

    ```csv
    2024-12-15T02:30:00Z,-0.586
    2024-12-15T02:40:00Z,-0.5045
    2024-12-15T13:30:00Z,3707.2292692307697
    2024-12-15T13:40:00Z,3030.0011304347836
    2024-12-15T13:50:00Z,556.0824528301885
    ```

Das Problem ist also die Zeit zwischen 03:40 und 14:30 (CET) am 15. Dezember. Es stellt sich nun die Frage, wie SOLECTRUS am besten damit umgeht. Momentan wird einfach nur das Integral gebildet. Das führt aber zu einer linearen Interpolation von 03:40 bis 13:30, was natürlich nicht der Realität entspricht.

Mein gestern vorgenommener Korrekturversuch war unsinnig, weil er gar nichts verändert hat, das habe ich wieder rückgängig gemacht. 

Ich bin noch etwas unschlüssig, wie SOLECTRUS am besten mit großen Lücken in den Messwerten umgehen soll. Das betrifft übrigens nicht nur die neuen selbstdefinierten Verbraucher, sondern grundsätzlich die Berechnung der Energie aus den Leistungsdaten. Wir hatten ähnliche Probleme bereits früher mal, durch die Verwendung des Integrals (mit v0.18) ist es deutlich besser geworden, aber offenbar gibt es immer noch Fälle, wo es zu Problemen kommen kann.

Ich habe verschiedene Lösungsansätze ausprobiert, mit keiner bin ich aber wirklich zufrieden. Denkbar ist, die Lücken durch Aggregationen aufzufüllen, was bei deinen Messwerten auch helfen würde. Allerdings hätte das auch Auswirkungen auf Messreihen, die keine Lücken enthalten, es würde also zu anderen Summen führen. Insofern möchte ich vorschlagen, dass du zunächst versucht, deine Messungen anders zu realisieren.

Grundsätzlich sind Lücken an sich kein Problem - wenn diese auf der Null-Linie enden und beginnen. Wenn aber, wie in dem betrachteten Fall deiner Messreihe am 15. Dezember, eine längere Zeit kein Messwert vorhanden ist und dann mit einem hohen Wert eingestiegen wird, dann führt die lineare Interpolation zu einer unrealistisch hohen Fläche unter der Kurve.

q2doc, 2024-12-24:
Hi, Problem ist bisher nicht mehr aufgetreten. Das BackUp lief sauber durch und der Tag sieht in Summe danach dennoch OK aus. Die Nullwerte filtere ich nun erstmal mit ioBroker raus (muss ja eh die Werte "kopieren" da die Shelly mit dem # trotz "" nicht funktionieren. Falls noch es noch einmal auftreten sollte schicke ich die CSV. Ansonsten erstmal Frohes Fest!

ledermann, 2024-12-25:
Ok, prima, dann schließe ich das Issue.

Das Problem mit dem Hashtag (`#`) kann ich weiterhin nicht reproduzieren. Vielleicht magst du diesen Fall in https://github.com/solectrus/mqtt-collector/issues/284 genauer beschreiben, dann prüfe ich das nochmal.

Happy 🎄!

closed 2024-12-25 by ledermann