solectrus/solectrus #3802: Neuer Speicher Zustand wird nur als Nummer angezeigt

patricknitsch, 2024-12-18:
### Beschreibung (auf Deutsch oder Englisch)

Seit Einführung der Aufwachladung gibt es Nummer 99. Diese wird aktuell nur als 99 angezeigt

[image]

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-145

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Proxmox LXC

ledermann, 2024-12-18:
Einfache Lösung: SENEC-Collector aktualisieren (falls nicht bereits aktuell) und ganz wichtig: Neu starten. Die Status-Bezeichnungen werden nur einmal beim Start vom SENEC-Speicher abgerufen.

Also:

```
docker compose down
docker compose pull
docker compose up -d
```

patricknitsch, 2024-12-18:
Auf latest stand er schon. Der Neustart hat aber geholfen. Danke!

closed 2024-12-18 by patricknitsch