solectrus/solectrus #3817: base/16384/pg_filenode.map": No such file or directory

Tobi0484, 2024-12-23:
### Beschreibung (auf Deutsch oder Englisch)

Dashboard ist nicht mehr erreichbar nach Neuinstallation

### Welche Version von SOLECTRUS verwendest du?

?

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspberry PI 5

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
SOLECTRUS Photovoltaic Dashboard
Copyright (C) 2020-2024 Georg Ledermann
License: GNU AGPLv3 - https://www.gnu.org/licenses/agpl-3.0.html
Version v0.18.2, built on 2024-11-23T09:17:52+01:00
Using ruby 3.3.6 (2024-11-05 revision 75015d4c1f) [aarch64-linux-musl]
Based on Alpine Linux 3.20.3
## Waiting for services...
redis (172.18.0.3:6379) open
Redis is up and running!
influxdb (172.18.0.4:8086) open
InfluxDB is up and running!
postgresql (172.18.0.8:5432) open
PostgreSQL is up and running!
## Preparing database...
bin/rails aborted!
ActiveRecord::ConnectionNotEstablished: connection to server at "172.18.0.8", port 5432 failed: FATAL:  could not open file "base/16384/pg_filenode.map": No such file or directory (ActiveRecord::ConnectionNotEstablished)
Caused by:
PG::ConnectionBad: connection to server at "172.18.0.8", port 5432 failed: FATAL:  could not open file "base/16384/pg_filenode.map": No such file or directory (PG::ConnectionBad)
Tasks: TOP => db:prepare
(See full trace by running task with --trace)
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
services:
  dashboard:
    image: ghcr.io/solectrus/solectrus:latest
    depends_on:
      postgresql:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - postgresql
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - TZ
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - WEB_CONCURRENCY
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - CO2_EMISSION_FACTOR
      - DB_HOST=postgresql
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL
[... 467 lines cut ...]
# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token

# Volume path for storing the InfluxDB data
INFLUX_VOLUME_PATH=./influxdb

##################################################################
###                    PostgreSQL database                     ###
##################################################################

# Volume path for storing the PostgreSQL data
DB_VOLUME_PATH=./postgresql

##################################################################
###                 Redis in-memory database                   ###
##################################################################

# Volume path for storing the database snapshot
REDIS_VOLUME_PATH=./redis

# URL of the Redis database
REDIS_URL=redis://redis:6379/1
```

ledermann, 2024-12-23:
> FATAL:  could not open file "base/16384/pg_filenode.map": No such file or directory 

Das sieht nach einem ernsthaften Defekt in PostgreSQL aus, möglicherweise ist die Datenbank kaputt. Für weitere Informationen zur Ursache ist es sinnvoll, das Log des PostgreSQL-Containers anzusehen:

```
docker compose logs postgresql
```

Falls die Datenbank wirklich defekt ist, fragt sich natürlich, wie das geschehen konnte. Bei einem Raspi, der auf einer SSD arbeitet, sollte man immer an die begrenzte Lebensdauer der SSD denken.

Wenn kein Backup besteht, würde ich PostgreSQL zurücksetzen, es gehen dabei nur Kleinigkeiten verloren:

```
docker compose down
mv postgresql postgresql.defekt
mkdir postgresql
rm redis/dump.rdb # Redis-Cache löschen
docker compose up -d
```

SOLECTRUS wird dann mit einer neuen Datenbank starten. Die Registrierung muss dadurch erneut durchgeführt werden (mit gleicher E-Mail-Adresse), die Einstellungen und Strompreise sind neu zu erfassen und die Tageszusammenfassungen werden dann neu berechnet.

Tobi0484, 2024-12-23:
Perfekt! Das zurücksetzen hat geklappt!
Danke...

closed 2024-12-23 by ledermann