solectrus/solectrus #3834: Preparing database... env: can't execute 'ruby': No such file or directory

Tobi0484, 2024-12-30:
### Beschreibung (auf Deutsch oder Englisch)

Nach einem Umzug meiner Installation auf einen neue SSD kann das Dashbord nicht mehr aufgerufen werden und folgende Logs werden verzeichnet:

Log des Dashbords:

SOLECTRUS Photovoltaic Dashboard
Copyright (C) 2020-2024 Georg Ledermann
License: GNU AGPLv3 - https://www.gnu.org/licenses/agpl-3.0.html
Version v0.18.2, built on 2024-11-23T09:17:52+01:00
docker/entrypoint.sh: line 11: ruby: not found
Using 
Based on Alpine Linux 3.20.3
## Waiting for services...
redis (172.18.0.5:6379) open
Redis is up and running!
influxdb (172.18.0.4:8086) open
InfluxDB is up and running!
postgresql (172.18.0.3:5432) open
PostgreSQL is up and running!
## Preparing database...
env: can't execute 'ruby': No such file or directory

Log des seneccollector:

env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory
env: can't execute 'ruby': No such file or directory

### Welche Version von SOLECTRUS verwendest du?

latest

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Pi 5

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
-
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
-
```

### Dein vollständiges `.env`

```text
-
```

ledermann, 2024-12-31:
Das sieht nach einem defekten Docker-Image aus. Versuch mal, das Image zu löschen und neu zu laden, z.B. so:

```
docker compose down
docker rmi ghcr.io/solectrus/solectrus:latest
docker compose pull
docker compose up -d
```

Tobi0484, 2024-12-31:
Perfekt Danke...
Läuft wieder!

closed 2024-12-31 by ledermann