solectrus/solectrus #3854: Unterstützung von Balkonkraftwerken

ledermann, 2025-01-09:
### Beschreibung (auf Deutsch oder Englisch)

Bereits mehrfach gewünscht (siehe #2621, #2862): SOLECTRUS sollte zusätzliche Strom-Erzeuger unterstützen, z.B. ein Balkonkraftwerk, das parallel zur bestehenden PV-Anlage betrieben wird. 

Hierbei ist zu berücksichtigen, dass der bestehende Wechselrichter oder Stromspeicher einen falschen Hausverbrauch meldet, weil der Strom aus dem Balkonkraftwerk direkt im Haus genutzt wird und somit der gemessene Hausverbrauch sich verringert. Dies muss korrigiert werden.

Da die Strombilanz immer ausgeglichen sein muss (`Herkunft == Verwendung`) kann der tatsächliche Hausverbrauch errechnet werden. Für die beiden Seiten der Strombilanz gilt ja bekanntermaßen:

```
Herkunft = PV_Dach + PV_BKW + Netzbezug + Akkuentnahme
Verwendung = Hausverbrauch + Akkubeladung + Einspeisung + E_Auto + Wärmepumpe
```

Geht man davon aus, dass genau ein Wert (z.B. der Hausverbrauch) unbekannt oder falsch ist, kann man diesen rückwärts aus der Strombilanz berechnen:

```
Hausverbrauch =   PV_Dach 
                + PV_BKW  
                + Netzbezug 
                + Akkuentnahme 
                - Akkubeladung
                - Netzeinspeisung
                - E-Auto 
                - Wärmepumpe
```

Es sind somit zwei Dinge zu tun:

- [ ] Zusätzlicher Sensor "Balkonkraftwerk", der die erzeugte Leistung (in Watt) erhält
- [ ] Hausverbrauch über obige Formel **berechnen**, wenn der Sensor `INFLUX_SENSOR_HOUSE_POWER` fehlt

Zur Konfiguration eines BKW muss man also dann den neuen Sensor **ergänzen** und den bestehenden Sensor für den Hausverbrauch **leeren**.

spirit0607, 2025-03-14:
Hallo, wollte nur mal Fragen, ob und wann eine Umsetzung geplant ist. Die diesjährige "Sonnensaison" beginnt ja nun und ich überlege zur Zeit ebenfalls ein Balkonkraftwerk zusätzlich zur PV Anlage zuzulegen. Ich würde das aber nur machen, wenn ich das auch in SOLECTRUS integrieren könnte. Vielen Dank und schöne Grüße!

ledermann, 2025-03-14:
@spirit0607 Der Grund, warum dieses Feature noch nicht implementiert ist, liegt an der für mich aufwendigen Berechnung des Hausverbrauchs - das macht Änderungen an recht vielen Stellen in SOLECTRUS erforderlich. Außerdem ist bislang nur theoretisch überlegt worden, ob das ganze mit obiger Formel überhaupt so funktionieren würde.

Daher folgender Vorschlag: Wärst du in der Lage, die Berechnung des Hausverbrauchs mit obiger Formel selbst vorzunehmen (z.B. in HomeAssistant oder ioBroker) und diesen korrigierten Wert dann per MQTT bereitzustellen? Denkbar wäre vielleicht auch ein "Task" direkt in InfluxDB.

Dann wäre es nämlich recht einfach, SOLECTRUS anzupassen, es müsste von mir dann nur ein zusätzlicher Sensor (z.B. `INFLUX_SENSOR_BALCONY_INVERTER_POWER`) ergänzt und die Darstellung in der Strombilanz erweitert werden. Als `INFLUX_SENSOR_HOUSE_POWER` würdest du dann deinen korrigierten Wert nehmen. Dadurch könnten wir dann praktisch testen, ob die bisher theoretischen Überlegungen auch praktisch funktionieren.

StevieC121176, 2025-03-15:
Also ich hab die Möglichkeit und könnte das testen. Ich übergebe in Solectrus per MQTT den Wert für die Gesamt PV Erzeugung und müsste nur den Wert für das Balkonkraftwerk da herausreichen lassen und den PV Wert für das BKW dann per extra Sensor an Solectrus übergeben.

ledermann, 2025-03-15:
@StevieC121176 Super, das würde helfen! Und da du die beiden Erzeuger bereits als addierten Wert vorliegen hast und an SOLECTRUS als `INFLUX_SENSOR_INVERTER_POWER` sendest, würde ich daran erst mal nichts ändern - denn dann kannst du die Tests bereits mit der vorliegenden Version `v0.19.0` (oder `develop`) machen. Das Entscheidende ist hingegen, einen korrigierten Wert für den **Hausverbrauch** bereitzustellen (gemäß obiger Formel) und für `INFLUX_SENSOR_HOUSE_POWER` zu nehmen - denn den sehe ich als den kritischen Punkt aller Überlegungen hier.

Wenn du das gemacht hast, wäre interessant, ob der korrigierte Hausverbrauch tatsächlich der Realität des Hauses entspricht. Der vom SENEC gemeldete Hausverbrauch dürfte ja, wenn das BKW aktiv ist, immer niedriger werden bis hin zu 0W. Mit dem korrigierten Wert sollte genau das nicht passieren.

StevieC121176, 2025-03-15:
@ledermann Ich sortiere mal meine Gedanken dazu. Grundsätzlich würde ich sagen die Formel stimmt. 
`Hausverbrauch = (PV_Dach + PV_BKW + Netzbezug + Akkuentnahme) - ( Akkubeladung + Einspeisung + E_Auto + Wärmepumpe + aus Hausverbrauch heuausgerechnete Verbraucher)`
In meinem Fall (Ich habe kein Senec) wird der Hausverbrauch per MQTT bereitgestellt (und dieser schon von EVCC berechnet), daher muss ich da nichts korrigieren.

Aktuell habe ich 2 PV Anlagen (2 Dachflächen) und 1 BKW. Diese 3 Werte werden als Gesamterzeugung an Solectrus übergeben.

Wenn ich nun das BKW da herausrechen und extra übergebe als `INFLUX_SENSOR_BALCONY_INVERTER_POWER` bleibt die Gesamterezugung ja gleich in meinem Fall. Es würde sich nur die Darstellung in Solectrus auf der Erzeugerseite ändern (sofern der zusätzliche Sensor dort separat angezeigt werden würde.

Durch das BKW ändert sich entweder die Einspeisung oder der Netzbezug bzw. die Batterieladung / Entladung. mit obiger Formel sollte sich der Hausverbrauch ergeben.

StevieC121176, 2025-03-15:
Ich habe mir jetzt mal ein Blockly Script in ioBroker gebaut, das den Hausverbrauch unabhängig von EVCC berechnet (nach obiger Formel). Diesen Wert übergebe ich jetzt als Hausverbrauch an Selectrus.

Zeppelin-PV, 2025-03-15:
... natürlich würde ich bei der Implementierung eines BKWs gerne mitmachen.
Mein BKW läuft seit 1.9.2023 und liefert seither über einen Shelly pro 3em auf der 3. Phase (power_c) die Werte
über den Shelly Collector direkt zu Solectrus. Der Shelly Collector zeichnet seit 10.9.2024 auf.
Der Hausverbrauch kommt bei mir über den Sensor INFLUX_SENSOR_HOUSE_POWER=SENEC:house_power
Eigentlich müsste das soweit ausreichen.
Zum Hausverbrauch, der ja zwischenzeitlich eine Anzahl X an Shellys und Zigbee Messadapter eine feste rechenbare Größe hat
müsste man, so denke ich, nur den Bereich "Andere Verbraucher" bei der Berechnung anpassen, wenn überhaupt.
Georg, - was meinst Du ???

ledermann, 2025-03-15:
@StevieC121176 Ok, das sieht ja schon mal sehr gut aus. 

Ich hatte irgendwie im Kopf, dass du zur SENEC-Fraktion gehörst. Ok, damit ist das Rückwärts-Berechnen des Hausverbrauchs in deinem Fall wohl gar nicht nötig. Möglicherweise berechnet das bereits evcc, das weiß ich nicht.

Ich vermute, dass dein jetzt mit Blockly-Script berechneter Hausverbrauch identisch mit dem ist, was evcc liefert. Oder zumindest sehr ähnlich. Danke für die Bestätigung, dass der Wechselrichter einen falschen Hausverbrauch liefert, das wäre auch meine Vermutung gewesen.

In deinem Fall würde jetzt also nur ein zusätzlicher Sensor für das BKW fehlen, damit SOLECTRUS die beiden Erzeuger differenziert anzeigen kann. Das würde ich aber gerne erst dann angehen, wenn die Sache mit dem Hausverbrauch wirklich geklärt ist.

---

Insofern die **Frage an die anderen**: Gibt es hier jemanden mit Balkonkraftwerk, der momentan einen **falschen** Hausverbrauch (also zu klein und manchmal auch 0W) in SOLECTRUS sieht und **zusätzlich** in der Lage ist, mit externen Tools einen korrigierten Hausverbrauch über obige Formel per MQTT an SOLECTRUS zu liefern?

---

@Zeppelin-PV: Ja, für dich wäre das Feature zweifellos sinnvoll. Ich würde gerne aber erst mal klären, ob die Formel alltagstauglich ist, also bevor ich das in SOLECTRUS implementiert. Und dazu wäre es gut, jemanden zu finden, der mit seinen Tools eigenständig in der Lage ist, den berechneten Wert zu liefern.

Mit den selbstdefinierten Verbrauchern hat das ganze Thema eher nichts zu tun. Die "Anderen Verbraucher" ergeben sich ja aus korrigiertem Hausverbrauch abzüglich Summe der Kleinverbraucher.

StevieC121176, 2025-03-15:
@ledermann Könnte ich dir die Werte nicht auch mit meiner Konstellation liefern? Wenn ich an Solectrus nur PV1 + PV2 übergebe, ohne BKW und den nach der Formel berechneten Hausverbrauch, wird auch bei mir der Hausverbrauch in Solectrus gegen 0 gehen, da ja bei meiner Berechnung des Hausverbrauchs der BKW Wert mit herangezogen wird. Wäre im Endeffekt nichts anderes als bei dem Senec Gerät, oder?

ledermann, 2025-03-16:
@StevieC121176 Prinzipiell hast du natürlich recht, das könnte so funktionieren. Mir wäre aber eine Bestätigung wichtig, dass sich mit obiger Formel tatsächlich (in einer Situation mit PV + BKW) der falsche Hausverbrauch korrigieren lässt - und zwar in allen denkbaren (Sonnen-)Situationen. 

Für mich ist das ganze Thema sowieso etwas problematisch - denn ich habe kein BKW, kann also weder selbst etwas ausprobieren noch habe ich einen Nutzen davon. Bevor ich also Entwicklungsarbeit im Wert einiger Tausender in die Entwicklung stecke (in der Hoffnung auf weitere Verbreitung von SOLECTRUS und damit weitere Sponsoren), möchte ich zumindest sicherstellen, dass es funktioniert wie gedacht.

spirit0607, 2025-03-16:
Ich würde auch gerne helfen und testen, allerdings habe ich bislang noch kein Balkonkraftwerk um etwas auszuprobieren.

ledermann, 2025-03-19:
Wer mutig ist **und** sich mit InfluxDB besser auskennt (letzteres ist wichtiger), kann mit folgender Anleitung die Formel umsetzen, und zwar **ohne** externe Tools. Der Trick ist, einen InfluxDB-Task einzurichten:
https://github.com/orgs/solectrus/discussions/4094

ledermann, 2025-03-21:
Da es einen Sponsor 💰 gibt, der sich an der Finanzierung beteiligen möchte, habe ich mir genauere Gedanken zur Umsetzung gemacht - weitere Sponsoren sind natürlich herzlich willkommen!

Ich skizziere hier mal die konkreten technischen Herausforderungen, die sich aus der Unterstützung für Balkonkraftwerke ergeben. Und wie ich mir die Implementierung vorstelle.

Das Ergänzen eines neuen Sensors für die Erzeugung des BKW sowie dessen Visualisierung in der Strombilanz ist unproblematisch. Kniffliger ist es, dass der Hausverbrauch korrigiert werden muss, wenn ein BKW installiert ist.

Der Hausverbrauch wird in SOLECTRUS als **Ausgangsgröße** betrachtet und an etlichen Stellen benötigt, z.B.:

- In der Strombilanz für AKTUELL und für jeden ZEITRAUM (Tag, Woche, Monat, etc.)
- Im Diagramm für den Hausverbrauch (Liniendiagramm aus InfluxDB und Säulendiagramm aus PostgreSQL)
- Im PowerSplitter für die Berechnung des Netzanteils für ALLE Verbraucher. Da der PowerSplitter den Netzbezug auf alle Verbraucher verteilt, führt ein falscher Hausverbrauch zu falschen Netzanteilen für alle Verbraucher.
- In der Top-10
- Auf der separaten Seite für den Hausverbrauch mit den selbstdefinierten Verbrauchern

Durch ein BKW wird der Hausverbrauch aber verfälscht und ist keine Ausgangsgröße mehr, sondern muss korrigiert werden, bevor er genutzt werden kann. Und damit wird es kompliziert.

## Bisherige Überlegungen

- Eine Idee ist die Verwendung eines **Tasks** in InfluxDB, der regelmäßig einen korrigierten Hausverbrauch berechnet und in die Datenbank schreibt. Der korrigierte Hausverbrauch wird dann von SOLECTRUS verwendet und der bisherige Code funktioniert unverändert.

  Der Task müsste von SOLECTRUS automatisch angelegt und gepflegt werden. Eine einmalige Berechnung für historische Messwerte müsste erfolgen und auch wiederholbar sein. Machbar, aber nicht trivial. Ein Pferdefuß ist außerdem, dass in [InfluxDB v3 die Task-Engine nicht mehr enthalten](https://www.influxdata.com/blog/resources-tasks-influxdb-v3/) sein wird. Es wäre also nicht zukunftssicher, auf diese Idee zu setzen.

- Ein anderer Ansatz ist eine On-the-fly-Korrektur direkt im Dashboard, ohne den Hausverbrauch in der Datenbank zu korrigieren. Das würde zumindest für einige Bereiche funktionieren, macht den Code aber komplexer. Und der (vom Dashboard unabhängige) PowerSplitter würde weiterhin eine separate Logik benötigen, um seine Berechnungen durchzuführen.

## Aktuell verfolgter Ansatz

Da der korrigierte Hausverbrauch bereits im PowerSplitter benötigt wird, würde es sich anbieten, ihn dort auch zu berechnen. Es würden dann in jedem Durchlauf (also alle paar Minuten) zwei Schritte ablaufen:

1. Zunächst wird der korrigierte Hausverbrauch berechnet und in ein separates Measurement geschrieben. Der PowerSplitter benötigt dafür weitere Input-Werte (siehe obige Formel), die aber einfach via ENV übergeben werden können.
2. Anschließend erfolgt die Berechnung der Netzanteile für alle Verbraucher wie bisher, nur dass dabei der korrigierte Hausverbrauch verwendet wird.

Praktischerweise hat der PowerSplitter bereits einen Mechanismus für die Initiierung einer Neuberechnung aller Daten, dieser könnte weiterverwendet werden.

Ein Nachteil ist aber, dass der PowerSplitter keine Live-Daten produziert, sondern nur alle paar Minuten läuft. Diese Latenz könnte man (unter Inkaufnahme von erhöhter Last) zwar reduzieren, aber LIVE wird es dadurch nicht.

Live-Daten werden im Dashboard aber nur für die Seite AKTUELL benötigt, bei allen anderen Seiten wäre ein Zeitverzug von wenigen Minuten kein Problem. Dort müsste also weiterhin der Hausverbrauch on-the-fly korrigiert werden.

## Szenarien

Die Unterstützung für BKW darf die bisherige Funktionalität nicht beeinträchtigen. Wer kein BKW hat, für den darf sich überhaupt nichts ändern. Es gibt also zwei Einsatzszenarien:

1. PV-Anlage ohne BKW: Wechselrichter liefert korrekten Hausverbrauch, der weiterhin unverändert verwendet wird.
2. PV-Anlage plus BKW: Wechselrichter liefert falschen Hausverbrauch, der korrigiert werden muss.

## Konfiguration

Alles wird über die zusätzliche Variable `INFLUX_SENSOR_BALCONY_INVERTER_POWER` gesteuert. Diese gibt optional den Ort in der InfluxDB an (Measurement und Field), an dem die aktuelle Leistung des BKW zu finden ist.

Wenn diese Variable gesetzt ist, bedeutet das (für das Dashboard sowie für den PowerSplitter) auch, dass nicht mehr der Original-Hausverbrauch verwendet werden darf, sondern eine Korrektur erfolgen muss.

Fehlt die Variable, wird der Hausverbrauch unverändert verwendet und SOLECTRUS verhält sich wie bisher.

## Nächste Schritte

- [ ] Implementierung der (optionalen) Korrektur des Hausverbrauchs im PowerSplitter
- [ ] Implementierung der Korrektur des Hausverbrauchs im Dashboard für AKTUELL
- [ ] Visualisierung der BKW-Leistung in der Strombilanz

ledermann, 2025-03-21:
@StevieC121176 Die Idee ist, dass ein gelieferter Hausverbrauch (also `INFLUX_SENSOR_HOUSE_POWER`) **ignoriert** wird, sobald `INFLUX_SENSOR_BALCONY_INVERTER_POWER` definiert ist. Die Berechnung nach obiger Formel erfolgt dann durch SOLECTRUS. Das oben von Dir oben erwähnte Blockly-Script kann dann entfallen. 

Das Testszenario hier bei mir (ich habe kein BKW) sieht übrigens so aus: Ich setze die Variable für das BKW auf ein nicht existentes Feld, SOLECTRUS erhält dadurch also keinen Wert, fasst diesen als 0 auf und führt aber dennoch die Berechnung des Hausverbrauches durch. Den berechneten Wert vergleiche ich mit dem vom SENEC-Speicher gelieferten Original-Hausverbrauch. Und wenn das nahezu gleich ist, insbesondere auf lange Sicht, funktioniert die Berechnung.

spirit0607, 2025-03-21:
Sehr spannend und ziemlich kompliziert in der Umsetzung. Vielleicht liege ich komplett falsch und verstehe den Zusammenhang noch nicht ganz korrekt, aber könnte nicht eine Berechnung des Hausverbrauchs und der PV Leistung bereits im MQTT Collector, bzw. SENEC Collector erfolgen? Dort also die abgefragten Werte entsprechend mit Formeln im Collector berechnen und dann korrigiert an Influx und SOLECTRUS senden?

ledermann, 2025-03-21:
@spirit0607 Ja, es ist wirklich etwas kompliziert und ich bin dankbar für alles, was die Sache vereinfachen würde. Die Verlagerung der Berechnung in einen Collector klingt erst mal gut. Es würde nämlich das vorherige Abrufen der Werte aus InfluxDB ersparen und würde eine Live-Anzeige ermöglichen. Aber es hätte andere Nachteile:

1. Im SENEC-Collector kann die Berechnung nicht erfolgen, weil dort die Leistung des BKW gar nicht vorliegt. Ähnliches gilt für den Shelly-Collector, auch dem fehlen Werte.

2. Im MQTT-Collector **könnte** es klappen, aber nur, wenn dort auch sämtliche benötigten Werte ankommen (bis zu 8 Stück, siehe obige Formel). Das ist aber nicht zwangsläufig der Fall. Wenn jemand z.B. einen SENEC-Speicher hat, dann wird er hauptsächlich den SENEC-Collector verwenden. Per MQTT kommen dann nur die restlichen Werte an, wie z.B. die Leistung des BKW. Damit wäre eine Berechnung dann nicht möglich.

3. Der MQTT-Collector ist nur ein Durchlauferhitzer, der die Werte vom Broker an InfluxDB weiterreicht. Er könnte also nicht dafür benutzt werden, um historische Werte zu berechnen. Das ist aber wünschenswert.

Daher halte ich eine Integration in den PowerSplitter für eine geeignete Lösung. Der ist immer vorhanden (bzw. kann leicht ergänzt werden). Allerdings muss der die Werte aus InfluxDB erst mal abholen, bevor er rechnen und zurückschreiben kann. Das sorgt zwangsläufig für eine gewisse Latenz. Daher die Idee, die Berechnung ZUSÄTZLICH noch im Dashboard vorzunehmen (dort aber nur für AKTUELL).

Eine ganz andere Idee, die ich auch zwischendurch hatte, wäre ein zusätzlicher Processing-Layer. Also ein neuer Service, ebenfalls als Container, der zwischen die Collectoren und InfluxDB geschaltet wird und daher sämtliche Messwerte sieht, Berechnungen jeder Art durchführen kann und dann erst nach InfluxDB schreibt. Ja, wäre cool, aber das würde die ganze Architektur über den Haufen werfen - das wäre eher SOLECTRUS 2.0, und wir sind noch nicht mal bei 1.0 ;-)

ledermann, 2025-03-29:
Auch wenn ich mich hier länger nicht geäußert habe - es ist etwas in Arbeit :)
Dieses Thema ist wahrlich kein einfaches und ich habe verschiedenste Ansätze evaluiert, wieder verworfen, nochmal aufgegriffen usw.

Möglicherweise gibt es bald etwas zu vermelden. Ich habe da etwas implementiert und führe gerade einen Last-Test durch. Wenn die Live-Demo in den nächsten Stunden und Tagen aussieht wie immer, wäre das ein gutes Zeichen - stay tuned!

ledermann, 2025-03-31:
Alle bisher dargestellten Ansätze habe ich wieder verwerfen müssen:

- Eine Korrekturberechnung direkt im Dashboard ist aufwändig - und müsste doppelt auch im Power-Splitter erfolgen. Das wäre keine schöne Lösung. 
- Eine Berechnung in HomeAssistant oder anderen Tools könnte man machen, erfordert aber weitreichende Kenntnisse für den User - und natürlich auch die Existenz dieser Tools auf dem eigenen Server.
- Eine Berechnung über InfluxDB-Tasks (siehe #4094) ist ebenfalls anspruchsvoll für den User, fehleranfällig und nicht nachhaltig, weil das nächste große Release von InfluxDB voraussichtlich keine Tasks mehr enthalten wird.

Daher habe ich etwas ganz Neues ausgetüftelt, dass sich hoffentlich bewähren wird: Einen **Ingestion-Proxy**, durch den sämtliche relevanten Messwerte durchgeleitet werden, bevor sie an InfluxDB gehen. 

Vor 10 Tagen habe ich das noch als Zukunftsmusik abgetan, nun aber doch realisiert. Ich hoffe, die 60+ Programmierstunden waren eine gute Investition. Ich möchte nun gerne herausfinden, ob es funktioniert. Wer sich traut - bitte hier entlang zum Ausprobieren:
https://github.com/orgs/solectrus/discussions/4138

ledermann, 2025-04-03:
Die Darstellung im Dashboard kann nun ebenfalls getestet werden:
https://github.com/orgs/solectrus/discussions/4152

Damit ist alles vorläufig implementiert. Ich erhoffe mir reichhaltiges Feedback :-)

PR #4195 (2025-05-19): Multiple inverters
Closes #3854 
Closes #1756