solectrus/solectrus #3859: Wärmepumpe vor SENEC EnFluRi - wie richtig Konfigurieren (+Power-Splitter)

unclej84, 2025-01-11:
### Beschreibung (auf Deutsch oder Englisch)

Hallo,

wie muss die Wärmepumpe für Solectrus in der Konfiguration hinterlegt werden damit alles korrekt angezeigt und berechnet werden kann?
Die Wärmepumpe zweigt direkt hinter dem Hausanschlusszähler ab und wird mittels Shelly erfasst (sind somit nicht in `GRID_IMPORT_POWER` oder `HOUSE_POWER` enthalten) - die Daten liegen vor und werden im Solectrus soweit ich feststellen kann auch korrekt angezeigt (Diagramm und Summe links). Jedoch scheint der Power-Splitter mit der Konfiguration dann etwas verwirrt zu sein (siehe Screenshots) - es werden Prozente angezeigt, die überhaupt nicht sein können (gesamte Erzeugung < prozentualer Wert des Verbrauchs der Wärmepumpe) 

[image]
[image]

Funktioniert dieser Aufbau so überhaupt oder müsste ich dann die Werte für `GRID_IMPORT_POWER` vom Hausanschlusszähler verwenden?

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-254-g9ad13c97

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Docker

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
name: solectrus
services:
  dashboard:
    depends_on:
      influxdb:
        condition: service_healthy
        required: true
      postgres:
        condition: service_healthy
        required: true
      redis:
        condition: service_healthy
        required: true
    environment:
      ADMIN_PASSWORD: XXXXXXXXXXX
      APP_HOST: localhost
      CO2_EMISSION_FACTOR: "258"
      DB_HOST: postgres
      DB_PASSWORD: XXXXXXXXXXX
      DB_USER: solectrus
      FORCE_SSL: "false"
      INFLUX_BUCKET: my-solectrus-bucket
      INFLUX_EXCLUDE_FROM_HOUSE_POWER: CUSTOM_POWER_01,CUSTOM_POWER_08
      INFLUX_HOST: influxdb
      INFLUX_MEASUREMENT: SENEC
      INFLUX_ORG: sbk-unclej-de
      INFLUX_POLL_INTERVAL: "10"
      INFLUX_PORT: "8086"
      INFLUX_SCHEMA: http
      INFLUX_SENSOR_BATTERY_CHARGING_POWER: SENEC:bat_power_plus
      INFLUX_SENSOR_BATTERY_DISCHARGING_POWER: SENEC:bat_power_minus
      INFLUX_SENSOR_BATTERY_SOC: SENEC:bat_fuel_charge
      INFLUX_SENSOR_CAR_BATTERY_SOC: ""
      INFLUX_SENSOR_CASE_TEMP: SENEC:case_temp
      INFLUX_SENSOR_CUSTOM_POWER_01: heatpump:power
      INFLUX_SENSOR_CUSTOM_POWER_02: freezer:power
      INFLUX_SENSOR_CUSTOM_POWER_03: ac_bedroom:power
      INFLUX_SENSOR_CUSTOM_POWER_04: ac_veranda:power
      INFLUX_SENSOR_CUSTOM_POWER_05: ac_living_room:power
      INFLUX_SENSOR_CUSTOM_POWER_06: fridge_basement:power
      INFLUX_SENSOR_CUSTOM_POWER_07: microwave:power
      INFLUX_SENSOR_CUSTOM_POWER_08: well_pump:power
      INFLUX_SENSOR_CUSTOM_POWER_09: washing_machine:power
      INFLUX_SENSOR_CUSTOM_POWER_10: dryer:power
      INFLUX_SENSOR_GRID_EXPORT_LIMIT: SENEC:power_ratio
      INFLUX_SENSOR_GRID_EXPORT_POWER: SENEC:grid_power_minus
      INFLUX_SENSOR_GRID_IMPORT_POWER: SENEC:grid_power_plus
      INFLUX_SENSOR_HEATPUMP_POWER: pro3em:power
      INFLUX_SENSOR_HOUSE_POWER: SENEC:house_power
      INFLUX_SENSOR_INVERTER_POWER: SENEC:inverter_power
      INFLUX_SENSOR_INVERTER_POWER_FORECAST: Solcast:watt
      INFLUX_SENSOR_OUTDOOR_TEMP: netatmo:temperature
      INFLUX_SENSOR_SYSTEM_STATUS: SENEC:current_state
      INFLUX_SENSOR_SYSTEM_STATUS_OK: SENEC:current_state_ok
      INFLUX_SENSOR_WALLBOX_CAR_CONNECTED: SENEC:ev_connected
      INFLUX_SENSOR_WALLBOX_POWER: SENEC:wallbox_charge_power
      INFLUX_TOKEN: XXXXXXXXXXX
      INSTALLATION_DATE: "2022-04-28"
      REDIS_URL: redis://redis:6379/1
      SECRET_KEY_BASE: XXXXXXXXXXX
      TZ: Europe/Berlin
      WEB_CONCURRENCY: "0"
    healthcheck:
      test:
        - CMD-SHELL
        - nc -z 127.0.0.1 3000 || exit 1
    image: ghcr.io/solectrus/solectrus:pr-3747
    links:
      - postgres
      - influxdb
      - redis
    logging:
      options:
        max-file: "3"
        max-size: 10m
    networks:
      default: null
    ports:
      - mode: ingress
        target: 3000
        published: "3000"
        protocol: tcp
    restart: unless-stopped
  influxdb:
    command:
      - influxd
      - run
      - --bolt-path
      - /var/lib/influxdb2/influxd.bolt
      - --engine-path
      - /var/lib/influxdb2/engine
      - --store
      - disk
    environment:
      DOCKER_INFLUXDB_INIT_ADMIN_TOKEN: XXXXXXXXXXX
      DOCKER_INFLUXDB_INIT_BUCKET: my-solectrus-bucket
      DOCKER_INFLUXDB_INIT_MODE: setup
      DOCKER_INFLUXDB_INIT_ORG: sbk-unclej-de
      DOCKER_INFLUXDB_INIT_PASSWORD: XXXXXXXXXXX
      DOCKER_INFLUXDB_INIT_USERNAME: XXXXXXXXXXX
      TZ: Europe/Berlin
    healthcheck:
      test:
        - CMD
        - curl
        - -f
        - http://localhost:8086/ping
      timeout: 10s
      interval: 30s
      retries: 5
      start_period: 1m0s
    image: influxdb:2.7
    logging:
      options:
        max-file: "3"
        max-size: 10m
    networks:
      default: null
    ports:
      - name: influx
        mode: ingress
        host_ip: 0.0.0.0
        target: 8086
        published: "8086"
        protocol: tcp
        app_protocol: http
    restart: unless-stopped
    volumes:
      - type: bind
        source: /home/eichhorn/docker/solectrus/influxdb/data
        target: /var/lib/influxdb2
        bind:
          create_host_path: true
      - type: bind
        source: /home/eichhorn/docker/solectrus/influxdb/config
        target: /etc/influxdb2
        bind:
          create_host_path: true
      - type: bind
        source: /home/eichhorn/docker/solectrus/db_dumps
        target: /dump
        bind:
          create_host_path: true
  power_splitter:
    depends_on:
      influxdb:
        condition: service_healthy
        required: true
      redis:
        condition: service_healthy
        required: true
    environment:
      INFLUX_BUCKET: my-solectrus-bucket
      INFLUX_EXCLUDE_FROM_HOUSE_POWER: CUSTOM_POWER_01,CUSTOM_POWER_08
      INFLUX_HOST: influxdb
      INFLUX_ORG: sbk-unclej-de
      INFLUX_PORT: "8086"
      INFLUX_SCHEMA: http
      INFLUX_SENSOR_BATTERY_CHARGING_POWER: SENEC:bat_power_plus
      INFLUX_SENSOR_CUSTOM_POWER_01: heatpump:power
      INFLUX_SENSOR_CUSTOM_POWER_02: freezer:power
      INFLUX_SENSOR_CUSTOM_POWER_03: ac_bedroom:power
      INFLUX_SENSOR_CUSTOM_POWER_04: ac_veranda:power
      INFLUX_SENSOR_CUSTOM_POWER_05: ac_living_room:power
      INFLUX_SENSOR_CUSTOM_POWER_06: fridge_basement:power
      INFLUX_SENSOR_CUSTOM_POWER_07: microwave:power
      INFLUX_SENSOR_CUSTOM_POWER_08: well_pump:power
      INFLUX_SENSOR_CUSTOM_POWER_09: washing_machine:power
      INFLUX_SENSOR_CUSTOM_POWER_10: dryer:power
      INFLUX_SENSOR_GRID_IMPORT_POWER: SENEC:grid_power_plus
      INFLUX_SENSOR_HEATPUMP_POWER: pro3em:power
      INFLUX_SENSOR_HOUSE_POWER: SENEC:house_power
      INFLUX_SENSOR_WALLBOX_POWER: SENEC:wallbox_charge_power
      INFLUX_TOKEN: XXXXXXXXXXX
      POWER_SPLITTER_INTERVAL: "600"
      REDIS_URL: redis://redis:6379/1
      TZ: Europe/Berlin
    image: ghcr.io/solectrus/power-splitter:develop
    links:
      - influxdb
      - redis
    logging:
      options:
        max-file: "3"
        max-size: 10m
    networks:
      default: null
    restart: unless-stopped
  senec_collector:
    depends_on:
      influxdb:
        condition: service_started
        restart: true
        required: true
    deploy:
      resources:
        limits:
          cpus: 0.35
          memory: "104857600"
    environment:
      INFLUX_BUCKET: my-solectrus-bucket
      INFLUX_HOST: influxdb
      INFLUX_MEASUREMENT: SENEC
      INFLUX_ORG: sbk-unclej-de
      INFLUX_PORT: "8086"
      INFLUX_SCHEMA: http
      INFLUX_TOKEN: XXXXXXXXXXX
      SENEC_ADAPTER: local
      SENEC_HOST: 192.168.39.19
      SENEC_INTERVAL: "7"
      SENEC_LANGUAGE: de
      SENEC_PASSWORD: null
      SENEC_SCHEMA: https
      SENEC_USERNAME: null
      TZ: Europe/Berlin
    image: ghcr.io/solectrus/senec-collector:0.16.3
    links:
      - influxdb
    logging:
      options:
        max-file: "3"
        max-size: 10m
    networks:
      default: null
    restart: unless-stopped
networks:
  default:
    name: solectrus_default
```

### Dein vollständiges `.env`

```text
Werte sind über docker compose config direkt im Feld der compose.yml enthalten
```

unclej84, 2025-01-11:
Hausanschluss
|
Hausanschlusszähler
|--Shelly für Wärmpumpe
| |--Wärmpumpe
|--SENEC EnFluRi
  |--SENEC + Hausverbraucher

ledermann, 2025-01-11:
Nicht nur der Power-Splitter ist verwirrt, auch die Strombilanz ist nicht ausgeglichen. Der Grund dürfte sein, dass `GRID_IMPORT_POWER` viel zu gering ist. SOLECTRUS sowie der PowerSplitter erwarten hier den **insgesamt** aus dem Netz bezogenen Strom, also inklusive des Stroms für die Wärmepumpe.

unclej84, 2025-01-11:
Also am Besten die beiden SENEC Werte für `GRID_IMPORT_POWER` und `GRID_EXPORT_POWER` durch die Messung vom Hausanschlusszähler ersetzen? Wie verhält es sich mit dem `GRID_EXPORT_LIMIT`? Der müsste bleiben können, oder?

ledermann, 2025-01-11:
So ganz habe ich deine Hausinstallation nicht verstanden, aber du musst sicherstellen, dass SOLECTRUS den **Gesamt**-Netzbezug (als `GRID_IMPORT_POWER`) sowie die **Gesamt**-Netzeinspeisung (als `GRID_EXPORT_POWER`) kennt. Wenn der SENEC-Speicher diese Werte nicht kennt und du auf den Hausanschlusszähler zugreifen kannst, dann nimm dessen Werte.

Die Einspeisebegrenzung (`GRID_EXPORT_LIMIT`) ist relativ unbedeutend. Der SENEC-Speicher liefert diese, wenn aus rechtlichen Gründen weniger eingespeist wird als möglich. Wenn der Wert kleiner 100 ist (nur dann wird tatsächlich limitiert), wird im UI ein kleines Warn-Icon angezeigt. Wenn der Sensor fehlt, erscheint das Icon nie.