solectrus/solectrus #3901: Liniendiagramm fehlt nach restore

Zeppelin-PV, 2025-01-24:
### Beschreibung (auf Deutsch oder Englisch)

folgendes ist passiert:
mein RP4 mit custom consumer
[image]
läuft instabil. 
Nun habe ich auf einem  Ersatz-RP4 flux alles neu installiert
Solectrus läuft mit einer Datensicherung von gestern (vom "defekten" RP4) auf dem neuen RP4
Einzig - es wird kein Liniendiagramm Tag angezeigt die älter sind als heute.
Die Tage in Woche, Monat etc sind da

[image]

[image]

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-158-ga8c429da

### Auf welchem Gerät läuft bei dir SOLECTRUS?

RP4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
kein Fehler Log vorhanden
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
services:
  dashboard:
    image: ghcr.io/solectrus/solectrus:pr-3747
    depends_on:
      postgresql:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - postgresql
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - TZ
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - WEB_CONCURRENCY
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - DB_HOST=postgresql
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_POLL_INTERVAL
      - INFLUX_SENSOR_INVERTER_POWER
      - INFLUX_SENSOR_HOUSE_POWER
      - INFLUX_SENSOR_GRID_IMPORT_POWER
      - INFLUX_SENSOR_GRID_EXPORT_POWER
[... 656 lines cut ...]
# Set these names before first run, they can't be changed later!
INFLUX_BUCKET=my-solectrus-bucket

# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token

# Volume path for storing the InfluxDB data
INFLUX_VOLUME_PATH=/home/pi/solectrus/influxdb

##################################################################
###                    PostgreSQL database                     ###
##################################################################

# Volume path for storing the PostgreSQL data
DB_VOLUME_PATH=/home/pi/solectrus/postgresql

##################################################################
###                 Redis in-memory database                   ###
##################################################################

# Volume path for storing the database snapshot
REDIS_VOLUME_PATH=/home/pi/solectrus/redis
```

ledermann, 2025-01-24:
> mein RP4 läuft instabil

Rein interessehalber: Wie äußert sich das? SD-Karte defekt?

> Einzig - es wird kein Liniendiagramm Tag angezeigt die älter sind als heute.

Wahrscheinliche Ursache: Du hast auf dem alten und auf dem neuen Raspi unterschiedliche Sensoren definiert. Das darf nicht sein.

Bei einem Raspi-Umzug besser nicht den Konfigurator nutzen, sondern die bestehende `.env` und `compose.yaml` kopieren.

Zeppelin-PV, 2025-01-24:
der RP4 läuft mit einer 120GB SSD auf einer Zusatzplatine die über eine USB Brücke mit dem RP4 verbunden ist.
Das System friert hin und wieder ein. Wenn ich es nicht bemerke fehlen Zeiten.

Ich habe nicht den Konfigurator benutzt. Alle Dateien sind vom alten RP4.
Aber - ich habe die Instanz unter  /home/admin/solectrus  angelegt  (wegen den admin Rechten)
hier habe ich auch die Order  influx, postgreql und redis angelegt
Nach dem restore wurde die alte Struktur angelegt  /home/pi/solectrus
hier sind dann wohl auch die aktiven   influx, postgreql und redis  ordner.

Ist das wohl die Ursache ?  - wo doch alles ordnungsgemäß funktioniert, bis eben dem Liniendiagramm in heute

ledermann, 2025-01-24:
Vielleicht bist du irgendwie mit den Ordnern durcheinander gekommen. Du musst dafür sorgen, dass das Backup an den richtigen Ort wiederhergestellt wird. Das muss der sein, der in der `.env` referenziert wird (als `*_VOLUME_PATH`).

Möglicherweise hast du die Datenbanken jetzt zweimal: Einmal leer und einmal mit den richtigen Daten.

Zeppelin-PV, 2025-01-24:
genau so ist es, 
ich  werde wohl nicht drum-herum kommen alles wieder in den Pfad /home/pi/solectrus zu legen
oder kann ich die Einträge ändern sodass alles in  /home/admin/solectrus läuft 
INFLUX_VOLUME_PATH=/home/pi/solectrus/influxdb  -->  /home/admin/solectrus (auch für postgresql und redis)

so, jetzt habe ich wieder alles in /home/pi/solectrus
docker compose down -> docker compose pull -> docker compose up -d
kein Liniendiagramm in Heute wenn man zurückblättert und die Daten aus der wiederhergestellten Datei 2025-01-23 kommen

der Problem-RP4 läuft gerade und da sind die Liniendiagramme vorhanden

Zeppelin-PV, 2025-01-25:
... und noch eine Feststellung
wenn ich nun am Ersatz RP4 ein Backup mache, macht er dies nur mit der Differenz die zwischen dem Restore und der neuen Laufzeit entstandenen Daten

ledermann, 2025-01-25:
Ich kann dir nicht so recht helfen, vermutlich hast du beim Kopieren oder Wiederherstellen der Datenbanken etwas durcheinander gebracht. Hier nur ein paar allgemeine Hinweise:

- SOLECTRUS speichert alle Messwerte in InfluxDB ab, das ist der "single source of truth", aus der alle Statistiken und Anzeigen gespeist werden. Aus Performance-Gründe werden aber Tageszusammenfassungen berechnet und in PostgreSQL abgelegt. Diese kann man jederzeit zurücksetzen, sodass diese dann erneut (aus InfluxDB) berechnet werden.
- Die Werte in der Strombilanz für **vergangene Zeiträume** werden ausschließlich aus den Tageszusammenfassungen gebildet. Nur die Strombilanz aus **AKTUELL** greift auf InfluxDB zu.
- Das Kurvendiagramm für **AKTUELL** sowie für einen Tag greift direkt auf InfluxDB zu, die Balkendiagramme hingegen auf die Tageszusammenfassungen.
- Die Datenbank Redis speichert nur einen Cache, den muss man nicht wiederherstellen.

Es ist daher wichtig, InfluxDB und PostgreSQL immer nur zusammen zu sichern und zusammen wiederherzustellen. Tut man dies nicht, gibt es Durcheinander. Wenn man beispielsweise auf einem neuen Server nur PostgreSQL wiederherstellt, aber InfluxDB neu anlegt, dann sieht man weiterhin historische Daten in der Strombilanz (aus den Tageszusammenfassungen), aber die Kurvendiagramme sind alle leer. Setzt man die Tageszusammenfassungen dann zurück, ist alles leer. Das ist vermutlich bei dir die Situation.

In der `.env` steht, wo die Datenbanken zu finden sind. Folgende Angabe legt beispielsweise fest, dass die InfluxDB im Unterordner `influxdb` des aktuellen Ordners zu finden ist:

```env
INFLUX_VOLUME_PATH=./influxdb
```

Analog für PostgreSQL und Redis.

Die Pfadangaben kann man natürlich ändern, wenn man die Datenbanken ganz woanders ablegen möchte. Jede Änderung in `compose.yml` erfordert natürlich einen Neustart des Compose-Stacks: `docker compose down && docker compose up -d`

Eine weitere Quelle für Probleme, die bei einem Server-Umzug entstehen können, sind geänderte Sensor-Variablen. Man muss darauf achten, dass die aktuellen Messwerte an den gleichen Ort in der InfluxDB geschrieben wird (angegeben durch Measurment/Field), wo auch die historischen Daten liegen.

Ein ganz einfacher Weg, um SOLECTRUS auf einen neuen Server umzuziehen, ist folgender:

1) Auf dem alten Server `docker compose down` (wichtig, es muss für den Folgeschritt zunächst alles gestoppt werden!)
2) Den **gesamten** SOLECTRUS-Ordner mit allen Unterordnern auf den neuen Server an einen beliebigen Ort kopieren, also `.env`, `compose.yaml` und die drei Datenbank-Ordner
3) Auf dem neuen Server `docker compose up -d`

Das Backup/Restore-Tool benötigt man dafür also gar nicht.

closed 2025-01-28 by ledermann

Zeppelin-PV, 2025-01-28:
... abschließend
die Ursache für das zeitweise Einfrieren des RP4 habe ich gefunden
Der RP4 wurde zu nahe an den HomeAssistant mit dessen ZigBee Transmitter platziert
einen halben Meter weiter weg funktioniert er wieder - ohne einfrieren

zu Deinem einfachen Weg, kompletter Ordner solectrus kopieren - trotz compose down wird das kopieren verweigert
wie kann ich dem Pfad /home/pi/solectrus einfache admin Rechte vergeben ?

ledermann, 2025-01-28:
Die Rechteverwaltung in einem Linux-Dateisystem kann komplex sein, und ohne dein bestehendes System genau zu kennen, möchte ich da ungern zu tief eingreifen. 

Der genannte Ordner gehört dem User `pi`. Wenn du dich mit diesem einloggst, sollte auch das Kopieren klappen. Es scheint bei dir auch den User `admin` zu geben, der hat dann sein eigenes Home-Directory (unter `/home/admin`). Es ist wichtig, darauf zu achten, mit welchem User du dich einloggst.

Ansonsten kannst du versuchen, mit einem vorangestellten `sudo` zu kopieren. Oder mach eine neue Session auf mit `sudo -i`. In beiden Fällen bist du damit `root` und hast Zugriff auf alles.