solectrus/solectrus #3921: Anteil Photovoltaik negativ (custom_power)

unclej84, 2025-02-01:
### Beschreibung (auf Deutsch oder Englisch)

Ich habe gerade gesehen, dass die Prozentuale Verteilung für die Custom power Sensoren eingefügt wurde und bin dabei auf eine Sache gestoßen.
Scheinbar wird bei der Summenbildung nicht geprüft, ob der Netzbezug höher als der Verbrauch ist (vielleicht kommt der Unterschied durch die Mittelwertbildung oder so zustande) und zeigt dann einen negativen Photovoltaik-Anteil an:

[image]

In der Datenbank sieht es für den Zeitraum so aus:

[image]

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-196-ga9698bd5

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Docker

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

unclej84, 2025-02-01:
Es sind beim betreffenden Verbraucher meist nur sehr kurze Verbrauchszeiträume dafür mit entsprechend höherem Verbrauch (Mikrowelle) und die Werte werden in größeren Fenstern erfasst als der Hauszähler (Grid)

ledermann, 2025-02-01:
Interessant, danke für dein Feedback! Ich denke, die Ursache ist hier der PowerSplitter, der muss sicherstellen, dass der Netzanteil niemals höher ist als der Gesamtverbrauch. Eigentlich wird das auch geprüft. 

Daher zunächst ein paar Rückfragen: 

1) Welche Version hast du? Es sollte sein (gemäß Log): `Version develop, built at 2025-02-01 06:06:27 +0100`

2) Woher kommen die Messwerte für die Mikrowelle? Vom Shelly-Collector? Hast du da die Variable [INFLUX_MODE](https://docs.solectrus.de/referenz/shelly-collector/konfiguration.html#influx_mode-ab-version-050) auf `essential` gesetzt?

3) Bitte einmal die Tageszusammenfassungen zurücksetzen. Ändert das bei dir die Summenbildung?

unclej84, 2025-02-01:
1. Ich hatte den `develop` Power-Splitter vorhin zusammen mit dem Dashdoard gepullt - ist also genau die Version, die du geschrieben hast (habe ich geprüft).
2. Die Daten kommen von einem Tasmota Zwischenstecker (Powerdelta 105 = 5W Änderung für publish) per MQTT über den MQTT-Collector. (Zur Info: die Grid-Werte kommen auch von Tasmota [anderes Device] mit einem IR-Lesekopf am Stromzähler und werden auch bei 5W Änderung per MQTT gepublished - dies ist aber deutlich häufiger als beim Zwischenstecker)
3. Den Gedanken hatte ich auch als ich es gesehen hatte. Hatte den gesamten Zeitraum zurückgesetzt und neu berechnen lassen. aber das Ergebnis ist gleich.

Ich hoffe die Infos helfen weiter.

ledermann, 2025-02-01:
Super, besten Dank. Damit gehe ich jetzt mal auf Ursachenforschung.

unclej84, 2025-02-01:
Wenn du mir verrätst wie ich die Influx werde für z.B. heute im korrekten Format für dich exportieren müsste, kann ich die auch gerne zur Verfügung stellen, falls das hilft.

Kurze Frage nebenbei - inwiefern sind schon Rückmeldungen zum commit "[Use PowerSplitter for custom sensors](https://github.com/solectrus/solectrus/commit/bef660ca8735b4bcb8d611bbc9e1ca89ac02d2f7)" erwünscht bzgl. der "fehlenden" Darstellungen? Oder lieber noch warten, weil noch nicht komplett?

ledermann, 2025-02-01:
Influx-Auszug benötige ich nicht. Sehe gerade, dass ich das gleiche Problem auch in meinen Daten habe. Bin dran und kläre, woran das liegt.

Weiteres Feedback zum PowerSplitter erst mal nicht nötig, möglicherweise gibt es Folgefehler. Melde mich wieder :)

ledermann, 2025-02-01:
@unclej84 Wie es scheint, ist die Ursache wie folgt: Wenn ein Stromzähler seine Messwerte nur bei aktivem Verbrauch sendet (wie dein Tasmota im PowerDelta-Modus oder mein Shelly-Collector im Essential-Modus), dann hat der Power-Splitter Probleme, nach Abschaltung den Verbrauch richtig zu berechnen. Es wird dann falsch gemittelt. Das habe ich nun [korrigiert](https://github.com/solectrus/power-splitter/commit/3187175679efa7109eb99e4a1d27df784f8196ab).

Es sind nun folgende Schritte zu tun (leider etwas umständlich):

1) Update des Power-Splitters auf `Version develop, built at 2025-02-01 13:06:11 +0100`, also wieder `docker compose pull && docker compose up -d`
2) Im Power-Splitter eine Neuberechnung veranlassen: `docker compose kill --signal USR1 power-splitter` und warten, bis der Vorgang abgeschlossen ist (im Log verfolgen). Erst danach (!) mit Schritt 3 fortfahren.
3) Im Dashboard die Tageszusammenfassungen zurücksetzen und neu aufbauen lassen.

Bei mir ist das Problem damit behoben, d.h. es gibt keine negativen PV-Anteile mehr. Falls nicht, bitte hier Rückmeldungen.

Bei weiteren Beobachtungen/Fehlern zum PowerSplitter am besten ein neues Issue aufmachen.

unclej84, 2025-02-01:
Kein Thema - so umständlich ist es ja nicht. Schritt 1 ist durch (Version passt) und Schritt 2 läuft gerade (dauert wegen der konventionellen SSD leider ein wenig). Werde dann Feedback geben, wenn durch und danach Schritt 3 erfolgt ist.

gereons, 2025-02-01:
> Im Power-Splitter eine Neuberechnung veranlassen: docker compose kill --signal USR1 power-splitter und warten, bis der Vorgang abgeschlossen ist (im Log verfolgen). Erst danach (!) mit Schritt 3 fortfahren.
Im Dashboard die Tageszusammenfassungen zurücksetzen und neu aufbauen lassen.

Frage in dem Zusammenhang: gibt es eine Möglichkeit, das Löschen und Neu-Erzeugen der Summaries per zB curl-Request zu triggern, oder geht das nur über's Web-UI? Via curl wäre super weil man dann diesen "generiere _alles_ neu"-Prozess scripten könnte.

unclej84, 2025-02-01:
> Frage in dem Zusammenhang: gibt es eine Möglichkeit, das Löschen und Neu-Erzeugen der Summaries per zB curl-Request zu triggern, oder geht das nur über's Web-UI? Via curl wäre super weil man dann [...]

Du könntest einen Post-Request mittels Curl abschicken, der das macht - URL müsste <solectrus url>/summaries sein und als values müsstest du `_method` mit dem Wert `delete` schicken soweit ich den Quellcode der Seite verstehe.

ledermann, 2025-02-01:
@gereons Es wäre denkbar, so etwas zu ermöglichen, aber eine ganz automatische Lösung würde mir lieber. So wie derzeitig der Power-Splitter auf Redis zugreift und den Cache löscht, könnte der Power-Splitter zusätzlich nach einer Neuberechnung ein `DELETE from summaries` auf PostgreSQL ausführen. Das probiere ich mal zu implementieren.

@unclej84 Prinzipiell richtig, aber da müsste auch noch ein Cookie gesetzt werden, denn das Zurücksetzen der Tageszusammenfassungen über das UI darf nur der Admin.

gereons, 2025-02-01:
Prima, danke. Super wäre, wenn auch das Neuberechnen der Summaries direkt angestossen werden könnte, weil das aktuell (zumindestens in meinem Safari) nur dann komplett durchläuft, wenn man den entsprechenden Browsertab die komplette Zeit im Vordergrund behält, und das nervt irgendwie.

unclej84, 2025-02-01:
Das Neuberechnen selbst war soweit ich verstanden hatte nicht gemeint sondern das zurücksetzen - das Neuberechnen wird ja immer dann - auch häppchenweise - ausgeführt, wenn du eine Ansicht aufrufst deren Werte noch nicht vorliegen.

unclej84, 2025-02-01:
Der Fehler scheint behoben zu sein

ledermann, 2025-02-01:
Bezüglich Neuberechnen der Tageszusammenfassungen: Es ist tatsächlich so, dass momentan nur ein **Löschen** möglich ist, das Neuaufbauen findet dann sukzessive bei Bedarf statt. Ja, das kann dann etwas doof sein, wenn man schnellstmöglich alle Tage automatisch neu haben will. Wenn ein Tab im Hintergrund liegt, stoppt leider die Bearbeitung. Ich habe mich ursprünglich bewusst gegen eine Hintergrundverarbeitung im Backend entschieden, weil das nicht sauber im gleichen Docker-Container möglich gewesen wäre, ich bin da immer wieder auf Performance-Probleme und Konflikte mit den http-Requests gestoßen.

Grundsätzlich sollte das Neuberechnen die totale Ausnahme bleiben. In den letzten Tagen habe ich ein paar Dinge geändert, die eine Neuberechnung erzwungen haben, das wird (voraussichtlich) aber nicht der Normalfall werden.

Danke für die Bestätigung, dass der Fehler behoben ist. Somit schließe ich das hier und bearbeite alles Weitere an einem anderen Ort.

closed 2025-02-01 by ledermann