solectrus/solectrus #3922: Offene Punkte Power-Splitter-Berücksichtigung Custom-Consumers

unclej84, 2025-02-01:
### Beschreibung (auf Deutsch oder Englisch)

Folgende Punkte sind mir seit dem Commit [Use PowerSplitter for custom sensors](https://github.com/solectrus/solectrus/commit/bef660ca8735b4bcb8d611bbc9e1ca89ac02d2f7) noch aufgefallen:

- Keine Darstellung des Photovoltaik-Anteils bei den `INFLUX_EXCLUDE_FROM_HOUSE_POWER` Custom-Verbrauchern:
[image]

- Keine Darstellung des Photovoltaik-Anteils für `ANDERE` im Hausverbrauch:
[image]

- Chart zeigt im Balken bei `INFLUX_EXCLUDE_FROM_HOUSE_POWER` Custom-Verbrauchern keine Anteile:
[image]

- Hausverbrauch zeigt außer beim gesamten Hausverbrauch auch im Chart keine Anteile in den Balken (Custom + Andere):
[image]

- Akku-Beladung zeigt Photovoltaik-Anteil, aber im Diagramm wird bei der Beladung kein Anteil gezeigt:
[image]

Ich weiß nicht, ob die Punkte noch nicht implementiert oder durchgerutscht sind oder im Fall des Akku vielleicht gar nicht vorgesehen sind. Wollte es nur mal zusammentragen, da der Punkt `Berücksichtigung durch den Power-Splitter` im PR #3747 als letzter Task jetzt abgehakt wurde.

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-196-ga9698bd5

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Docker

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-02-01:
Danke für die Auflistung, da ist tatsächlich für mich noch etwas zu tun. Alle genannten Punkte sind berechtigt. Die Implementierung fehlt noch, also kein Bug.

ledermann, 2025-02-01:
Zur besseren Erledigung habe ich ein paar separate Issues angelegt, alles Weitere dort:

#3923, #3924, #3925, #3926

closed 2025-02-01 by ledermann