solectrus/solectrus #3926: Chart: Bei Akkubeladung fehlt PV/Netz-Anteil

ledermann, 2025-02-01:
### Beschreibung (auf Deutsch oder Englisch)

Extrahiert aus #3922: 

- Akku-Beladung zeigt Photovoltaik-Anteil, aber im Diagramm wird bei der Beladung kein Anteil gezeigt:
[image]

ledermann, 2025-02-09:
Ist implementiert. Bei mir (mit nur wenigen Notbeladungen) sieht das z.B. so aus:
https://demo.solectrus.de/battery_power/2024

[image]

closed 2025-02-09 by ledermann

unclej84, 2025-02-09:
Ja, wird angezeigt - mag eine optische Täuschung sein, aber irgendwie sieht der Netzbezug breiter als die PV aus?

[image]

[image]

ledermann, 2025-02-09:
Das halte ich tatsächlich für eine optische Täuschung - oder für ein Browser/OS-spezifisches Problem. Bei mir auf dem Mac sieht es auch in extremer Vergrößerung korrekt aus:

[image]

Momentan habe ich keine Idee, das anders zu machen.