solectrus/solectrus #3927: PV-Anteil in Monats-/Jahres-Anzeige nicht korrekt

unclej84, 2025-02-01:
### Beschreibung (auf Deutsch oder Englisch)

Irgendwie scheint die Berechnung des Anteils für die Monats-/Jahresanzeige (zumindest bei der Wärmepumpe) nicht korrekt zu sein.

Die einzelnen Tage haben 0% - ~16% Anteil (z.B. der 22.01. hat ~24% von 43,035kWh -> je nach Rundung 10,5 - 11kWh PV-Anteil) und trotzdem haben der Monat bzw. das Jahr nur ~2% (~372kWh -> je nach Rundung 5,6 - 9,3kWh). Also hat ein Tag schon mehr Anteil als der Monat.
Habe ich da ein falsches Verständnis? Für die Tage scheint es zu passen.

Monats-Anzeige:
[image]

Beispiel 22.01.:
[image]

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-197-gcc3a90e4

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Docker

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

unclej84, 2025-02-01:
Für den Februar mit bisher nur 1 Tag scheint es zu passen - da zeigen Tag und Monat jeweils den gleichen Prozent-Wert

ledermann, 2025-02-02:
Danke für deine ausführliche Analyse, das hilft mir sehr. Ursache ist hier der Umgang mit lückenhaften Daten, das ist nämlich alles andere als einfach. Es gibt z.B. folgende Szenarien:

- Bei einem Stromzähler, der nur dann etwas meldet, wenn auch ein Verbrauch stattfindet (wie beim Shelly-Collector im Essential-Modus) soll ein fehlender Wert als 0 interpretiert werden
- Bei einem Stromzähler, der nur dann etwas meldet, wenn sich ein Verbrauch signifikant ändert (wie bei Tasmota im PowerDelta-Modus), soll der letzte Wert fortgeschrieben werden.
- Bei einem Stromzähler, der nur alle 5min einen Wert liefert (wie beim SENEC V3 im Cloud-Modus), sollen fehlende Werte dazwischen fortgeschrieben werden
- Bei einem Stromzähler, der plötzlich ausfällt und nichts mehr meldet (wie bei deiner Wärmepumpe am 13. Januar), soll etwas Sinnvolles passieren. Nicht ganz klar, was. Vermutlich: Auf 0 setzen, wenn der Zustand anhält.

Mal sehen, ob ich da eine Lösung findet, die für alle Szenarien passt oder zumindest keinen groben Unfug ausrechnet.

In deinem Fall des Wärmepumpen-Ausfalls wird es vermutlich gar keine perfekte Lösung geben können. Denn es sind hier ja nicht alle Messwerte ausgefallen, sondern nur der Verbrauch der Wärmepumpe. Der Netzbezug wurde ja vermutlich weiterhin korrekt gemessen. Was soll der PowerSplitter da machen? Die Fortschreibung des letzten Wertes ist falsch. Ein Annehmen von 0 ist aber auch falsch, da durch den weiterhin hohen Netzbezug dieser zu Unrecht auf die anderen Verbraucher verteilt wird. Knifflig.

unclej84, 2025-02-02:
Ja, das mit den lückenhaften Daten hat mir auch im Grafana schon Kopfzerbrechen bereitet (auch Flux-Queries). 
Beim Tasmota gibt es für den Fall zum Glück noch die `Teleperiod` (Default 300s) sodass z.B. auch im Standby regelmäßig ein Wert übertragen wird (unabhängig vom PowerDelta). Aber leider auch nur, wenn der Tasmota übertragen kann (Netzwerkstörung oder Rausgezogen).

Bei mir war es jetzt wirklich so, dass an dem Tag zu dem Zeitpunkt alle Werte weg sind - es fiel nur nicht auf, weil die Werte im Vergleich zum Verbrauch der Wärmepumpe recht klein waren.
Mir würde jetzt eine spontane Lösung für mein Problem einfallen - manuell einen Wert "0" für alle Werte direkt nach dem Ausfall in die Influx schreiben - aber das würde natürlich das zugrundeliegende Problem nicht lösen.
Es kann ja durchaus immer mal vorkommen, dass auch für einen längeren Zeitraum, keine Daten kommen (Stromausfall, Messwertgeber defekt, Netzwerkproblem etc.)

Bin gerne bereit die Werte erstmal so zu lassen und dann eine eventuelle Lösung auszuprobieren, wenn du die Muße hast da nochmal zu tüfteln in der nächsten Zeit.

ledermann, 2025-02-02:
Trotz vieler Versuche habe ich hier leider keine Lösung anzubieten. Wenn ein Stromzähler ausfällt, führt die Berechnung des Power-Splitters zu einem falschen Ergebnis. Ich sehe nicht, wie das gerettet werden könnte. 

Ich werde versuchen, eine Art Validierung der Tageszusammenfassungen zu ergänzen. Es muss ja an jedem Tag gelten: `Summe der Netzanteile der Verbrauche = Gesamtnetzbezug`. Wenn das nicht gegeben ist, müssten die Netzanteile verworfen werden. Mal sehen.

unclej84, 2025-02-02:
Habe temporär erstmal im Postgres die `_grid` Werte auf die anderen Werte gesetzt um dann 0% zu erzeugen - um 1 Uhr Nachts wird wohl nichts aus Photovoltaik gespeist worden sein. Denke sowas würde auch passieren, wenn die Validierung hinzu kommt. So könnte ich für Tests den Zeitraum einfach noch mal neu rechnen lassen.

ledermann, 2025-02-03:
Ich habe nun eine Auto-Korrektur-Funktion für die PowerSplitter-Werte [implementiert](https://github.com/solectrus/solectrus/commit/ef30eb7d877882b73ded5fac6e876d2f7b51fff5). Das war erstaunlich aufwändig. Zu sehen in `custom-consumers-207`.

Diese Funktion verändert die Summen, die sich aus dem PowerSplitter ergeben. Üblicherweise sind das nur Rundungsfehler, in deinem Fall aber auch grob falsche Berechnungen durch Ausfall eines Stromzählers.

Es wird jetzt sichergestellt, dass:

- ... der Netzanteil eines Verbrauchers höchsten dem Verbrauch entspricht (also 100% Netzanteil)
- ... dass die Summe der Netzanteile aller Verbraucher exakt dem Netzbezug entspricht

Es wird ein Näherungsverfahren angewandt. Es geht also nicht um absolute Korrektheit (die bei einem Zählerausfall sowie nicht möglich ist), sondern um untereinander konsistente Werte.

Die Änderung bringt es leider mit sich, dass wieder die Tageszusammenfassungen zurückgesetzt werden (passiert automatisch) und neu berechnet werden müssen.

@unclej84 Mit dieser Änderung sollte das Problem der widersprüchlichen Daten behoben sein.

unclej84, 2025-02-03:
Da ist leider was schief gelaufen. Bekomme sobald er bei dem betroffenen Tag ankommt Fehler 500.

Log:

```
[192.168.30.20] method=GET path=/house_power/2025-W03 format=html controller=Balance::HomeController action=index status=200 allocations=18860 duration=129.24 view=119.33 db=3.69
[192.168.30.20] [ActiveJob] [SummarizerJob] [e3c3962d-fbe3-4142-8958-66594c5779d5] Performing SummarizerJob (Job ID: e3c3962d-fbe3-4142-8958-66594c5779d5) from Async(default) with arguments: Tue, 14 Jan 2025
[192.168.30.20] [ActiveJob] [SummarizerJob] [e3c3962d-fbe3-4142-8958-66594c5779d5] Error performing SummarizerJob (Job ID: e3c3962d-fbe3-4142-8958-66594c5779d5) from Async(default) in 17.27ms: TypeError (nil can't be coerced into Integer):
/app/app/services/summary_corrector.rb:69:in 'Integer#+'
/app/app/services/summary_corrector.rb:69:in 'Array#sum'
/app/app/services/summary_corrector.rb:69:in 'SummaryCorrector#distribute_evenly'
/app/app/services/summary_corrector.rb:62:in 'SummaryCorrector#fix_grid_values'
/app/app/services/summary_corrector.rb:24:in 'SummaryCorrector#adjusted'
/app/app/jobs/summarizer_job.rb:54:in 'SummarizerJob#correct_values'
/app/app/jobs/summarizer_job.rb:27:in 'block in SummarizerJob#perform_calculations'
/usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_adapters/abstract/transaction.rb:626:in 'block in ActiveRecord::ConnectionAdapters::TransactionManager#within_new_transaction'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/concurrency/null_lock.rb:9:in 'ActiveSupport::Concurrency::NullLock#synchronize'
/usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_adapters/abstract/transaction.rb:623:in 'ActiveRecord::ConnectionAdapters::TransactionManager#within_new_transaction'
/usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_adapters/abstract/database_statements.rb:367:in 'ActiveRecord::ConnectionAdapters::DatabaseStatements#within_new_transaction'
/usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_adapters/abstract/database_statements.rb:359:in 'ActiveRecord::ConnectionAdapters::DatabaseStatements#transaction'
/usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/transactions.rb:234:in 'block in ActiveRecord::Transactions::ClassMethods#transaction'
/usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_adapters/abstract/connection_pool.rb:418:in 'ActiveRecord::ConnectionAdapters::ConnectionPool#with_connection'
/usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_handling.rb:310:in 'ActiveRecord::ConnectionHandling#with_connection'
/usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/transactions.rb:233:in 'ActiveRecord::Transactions::ClassMethods#transaction'
/app/app/jobs/summarizer_job.rb:20:in 'SummarizerJob#perform_calculations'
/app/app/jobs/summarizer_job.rb:7:in 'SummarizerJob#perform'
/usr/local/bundle/gems/activejob-8.0.1/lib/active_job/execution.rb:68:in 'block in ActiveJob::Execution#_perform_job'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/callbacks.rb:120:in 'block in ActiveSupport::Callbacks#run_callbacks'
/usr/local/bundle/gems/i18n-1.14.7/lib/i18n.rb:348:in 'I18n::Base#with_locale'
/usr/local/bundle/gems/activejob-8.0.1/lib/active_job/translation.rb:9:in 'block (2 levels) in <module:Translation>'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/callbacks.rb:129:in 'BasicObject#instance_exec'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/callbacks.rb:129:in 'block in ActiveSupport::Callbacks#run_callbacks'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/core_ext/time/zones.rb:65:in 'Time.use_zone'
/usr/local/bundle/gems/activejob-8.0.1/lib/active_job/timezones.rb:9:in 'block (2 levels) in <module:Timezones>'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/callbacks.rb:129:in 'BasicObject#instance_exec'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/callbacks.rb:129:in 'block in ActiveSupport::Callbacks#run_callbacks'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/callbacks.rb:140:in 'ActiveSupport::Callbacks#run_callbacks'
/usr/local/bundle/gems/activejob-8.0.1/lib/active_job/execution.rb:67:in 'ActiveJob::Execution#_perform_job'
/usr/local/bundle/gems/activejob-8.0.1/lib/active_job/instrumentation.rb:32:in 'ActiveJob::Instrumentation#_perform_job'
/usr/local/bundle/gems/activejob-8.0.1/lib/active_job/execution.rb:51:in 'ActiveJob::Execution#perform_now'
/usr/local/bundle/gems/activejob-8.0.1/lib/active_job/instrumentation.rb:26:in 'block in ActiveJob::Instrumentation#perform_now'
/usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/railties/job_runtime.rb:13:in 'block in ActiveRecord::Railties::JobRuntime#instrument'
/usr/local/bundle/gems/activejob-8.0.1/lib/active_job/instrumentation.rb:40:in 'block in ActiveJob::Instrumentation#instrument'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/notifications.rb:210:in 'block in ActiveSupport::Notifications.instrument'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/notifications/instrumenter.rb:58:in 'ActiveSupport::Notifications::Instrumenter#instrument'
/usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/notifications.rb:210:in 'ActiveSupport::Notifications.instrument'
/usr/local/bundle/gems/activejob-8.0.1/lib/active_job/instrumentation.rb:39:in 'ActiveJob::Instrumentation#instrument'
[... cut ...]
[192.168.30.20] app/jobs/summarizer_job.rb:20:in 'SummarizerJob#perform_calculations'
[192.168.30.20] app/jobs/summarizer_job.rb:7:in 'SummarizerJob#perform'
[192.168.30.20] app/controllers/summaries_controller.rb:8:in 'SummariesController#show'
[192.168.30.20] app/controllers/concerns/auto_locale.rb:14:in 'ApplicationController#switch_locale'
[192.168.30.20] method=GET path=/500 format=html controller=ErrorsController action=show status=500 allocations=9774 duration=84.03 view=79.85 db=0.00
```

unclej84, 2025-02-04:
Mit `custom-consumers-208-g98379df9` läuft die Berechnung der Tageszusammenfassung erstmal wieder durch und es sieht auch plausibel aus.

ledermann, 2025-02-04:
Ja, da war noch ein Fehler enthalten, der auftrat, wenn ein Zähler einen ganz Tag lang oder länger nicht geliefert hat. Das ist jetzt behoben.

closed 2025-02-04 by ledermann