solectrus/solectrus #3931: Icons für Custom-Consumers

unclej84, 2025-02-02:
### Beschreibung (auf Deutsch oder Englisch)

Wenn die Idee aus #3396 noch aktuell ist:

> Hierfür könnten zusätzliche Sensoren definiert werden, idealerweise mit folgenden Angaben:
> 
> - Bezeichnung
> - **Icon**
> - Darstellung als separates Segment in der Strombilanz ja/nein

Würden aktuell noch die Icons fehlen bzw. die Möglichkeit diese zu definieren.

ledermann, 2025-02-03:
Bei den Icons muss noch geklärt werden:

- Auswahl mitliefern? Führt zu häufigen Fragen, ob dieses oder jenes Icon noch ergänzt werden könnte.
- Selber hochladen? Führt zu Problemen mit Dateiformaten und stört möglicherweise die visuelle Darstellung, wenn ein Icon nicht gut zu den bisherigen passt

Daher habe ich das erst mal zurückgestellt.

unclej84, 2025-02-03:
So wie ich gesehen habe, verwendest du schon Font Awsome Icons. Wie wäre es den Icon-Namen zu verwenden - dann ist klar was es gibt und was nicht und die Auswahl ist recht groß

ledermann, 2025-02-03:
Ja, FontAwesome. Derzeitig liefere ich aber nur eine [ganz kleine Auswahl](https://github.com/solectrus/solectrus/blob/develop/app/javascript/controllers/icons_controller.ts) mit, nämlich die Icons, die auch wirklich benötigt werden. Die liegen in `vendor-xxx.js` als SVG. Da alle Icons reinzupacken, würde diese Datei erheblich vergrößern.

Da müsste ich mal sehen, ob sich das anders regeln lässt. Aber momentan sehe ich das eher als ein Randthema.