solectrus/solectrus #3945: Fehler 500 bei custom consumer - Hausverbrauch

Zeppelin-PV, 2025-02-06:
### Beschreibung (auf Deutsch oder Englisch)

hallo Georg,
war neugierig und habe gerade die Version 232 geholt. (mit 230 lief noch alles)
Jetzt kommt bei den custom consumer der Fehler 500
Beim Hausverbrauch geht Aktuell, aber bei Tag usw. kommt der Fehler 500
das habe ich versucht:
docker compose down
docker compose pull
docker compose up -d
und die Tageszusammenfassungen habe ich gelöscht und neu laufen lassen

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-232-ge4d26256

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspi4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
dashboard-1  | [192.168.1.110] Checked for update availability, valid for 720 minutes
dashboard-1  | [192.168.1.110] [ActiveJob] [SummarizerJob] [a33b3dc3-d12e-43b9-ba3c-8ea463085ab4] Performing SummarizerJob (Job ID: a33b3dc3-d12e-43b9-ba3c-8ea463085ab4) from Async(default) with arguments: Thu, 06 Feb 2025
dashboard-1  | [192.168.1.110] [ActiveJob] [SummarizerJob] [a33b3dc3-d12e-43b9-ba3c-8ea463085ab4] Performed SummarizerJob (Job ID: a33b3dc3-d12e-43b9-ba3c-8ea463085ab4) from Async(default) in 405.11ms
dashboard-1  | [192.168.1.110] method=GET path=/house/stats/house_power/2025-02-06 format=html controller=House::StatsController action=index status=500 allocations=20609 duration=831.21 view=0.00 db=33.28
dashboard-1  | [192.168.1.110]   
dashboard-1  | [192.168.1.110] ActionView::Template::Error (nil can't be coerced into Integer)
dashboard-1  | [192.168.1.110] Caused by: TypeError (nil can't be coerced into Integer)
dashboard-1  | [192.168.1.110] 
dashboard-1  | Information for: ActionView::Template::Error (nil can't be coerced into Integer):
dashboard-1  | [192.168.1.110]     53: 
dashboard-1  | [192.168.1.110]     54:           // Inner content
dashboard-1  | [192.168.1.110]     55:           .flex-1.flex
dashboard-1  | [192.168.1.110]     56:             = render SegmentContainer::Component.new tippy_placement: 'bottom',
dashboard-1  | [192.168.1.110]     57:                                                      calculator:,
dashboard-1  | [192.168.1.110]     58:                                                      timeframe: do |balance|
dashboard-1  | [192.168.1.110]     59:               ruby:
dashboard-1  | [192.168.1.110]   
dashboard-1  | [192.168.1.110] app/services/calculator/range.rb:352:in 'Integer#+'
dashboard-1  | [192.168.1.110] app/services/calculator/range.rb:352:in 'Array#sum'
dashboard-1  | [192.168.1.110] app/services/calculator/range.rb:352:in 'Calculator::Range#house_power_without_custom_grid_ratio'
dashboard-1  | [192.168.1.110] app/components/segment/component.rb:84:in 'Kernel#public_send'
dashboard-1  | [192.168.1.110] app/components/segment/component.rb:84:in 'Segment::Component#power_grid_ratio'
dashboard-1  | [192.168.1.110] app/components/segment/component.html.slim:66:in 'block in Segment::Component#call'
dashboard-1  | [192.168.1.110] app/components/segment/component.rb:30:in 'Segment::Component#link_to_or_div'
dashboard-1  | [192.168.1.110] app/components/segment/component.html.slim:15:in 'Segment::Component#call'
dashboard-1  | [192.168.1.110] app/components/segment_container/component.html.slim:5:in 'block in SegmentContainer::Component#call'
dashboard-1  | [192.168.1.110] app/components/segment_container/component.html.slim:4:in 'Array#each'
dashboard-1  | [192.168.1.110] app/components/segment_container/component.html.slim:4:in 'SegmentContainer::Component#call'
dashboard-1  | [192.168.1.110] app/views/house/stats/index.turbo_stream.slim:56
dashboard-1  | [192.168.1.110] app/controllers/house/stats_controller.rb:8:in 'House::StatsController#index'
dashboard-1  | [192.168.1.110] app/controllers/concerns/auto_locale.rb:14:in 'ApplicationController#switch_locale'
dashboard-1  | [192.168.1.110] 
dashboard-1  | Information for cause: TypeError (nil can't be coerced into Integer):
dashboard-1  | [192.168.1.110]   
dashboard-1  | [192.168.1.110] app/services/calculator/range.rb:352:in 'Integer#+'
dashboard-1  | [192.168.1.110] app/services/calculator/range.rb:352:in 'Array#sum'
dashboard-1  | [192.168.1.110] app/services/calculator/range.rb:352:in 'Calculator::Range#house_power_without_custom_grid_ratio'
dashboard-1  | [192.168.1.110] app/components/segment/component.rb:84:in 'Kernel#public_send'
dashboard-1  | [192.168.1.110] app/components/segment/component.rb:84:in 'Segment::Component#power_grid_ratio'
dashboard-1  | [192.168.1.110] app/components/segment/component.html.slim:66:in 'block in Segment::Component#call'
dashboard-1  | [192.168.1.110] app/components/segment/component.rb:30:in 'Segment::Component#link_to_or_div'
dashboard-1  | [192.168.1.110] app/components/segment/component.html.slim:15:in 'Segment::Component#call'
dashboard-1  | [192.168.1.110] app/components/segment_container/component.html.slim:5:in 'block in SegmentContainer::Component#call'
dashboard-1  | [192.168.1.110] app/components/segment_container/component.html.slim:4:in 'Array#each'
dashboard-1  | [192.168.1.110] app/components/segment_container/component.html.slim:4:in 'SegmentContainer::Component#call'
dashboard-1  | [192.168.1.110] app/views/house/stats/index.turbo_stream.slim:56
dashboard-1  | [192.168.1.110] app/controllers/house/stats_controller.rb:8:in 'House::StatsController#index'
dashboard-1  | [192.168.1.110] app/controllers/concerns/auto_locale.rb:14:in 'ApplicationController#switch_locale'
dashboard-1  | [192.168.1.110] method=GET path=/500 format=html controller=ErrorsController action=show status=500 allocations=9704 duration=64.24 view=63.09 db=0.00
dashboard-1  | [192.168.1.110] method=GET path=/inverter_power/now format=html controller=Balance::HomeController action=index status=200 allocations=16986 duration=333.85 view=332.81 db=0.00
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-02-06:
Ist behoben, danke für den Hinweis!

ledermann, 2025-02-07:
Dieser Fehler ein anderes Problem, das bei fehlenden PowerSplitter-Werten auftreten konnte. Auch das ist jetzt behoben.

closed 2025-02-07 by ledermann