solectrus/solectrus #3963: Erweiterung auf 20 selbstdefinierte Verbraucher

ledermann, 2025-02-10:
Auf vielfachen Wunsch eines einzelnen Herrn (hallo @Zeppelin-PV) soll es statt bislang maximal 10 selbstdefinierter Verbraucher zukünftig 20 geben.

Aufgrund der kürzlich vorgenommen [internen Änderung der Tageszusammenfassungen](https://github.com/orgs/solectrus/discussions/3749#discussioncomment-11971391) ist das nun machbar, ohne dass es negative Auswirkungen für diejenigen hätte, die deutlich weniger Verbraucher im Einsatz haben. Speicherplatzbedarf und Rechenaufwand von PowerSplitter und Tageszusammenfassungen steigt proportional zur Anzahl der **tatsächlich definierten** Verbraucher.

Ob die Behandlung von 20 Verbrauchern so sinnvoll ist, steht auf einem anderen Blatt. Ich sehe das etwas kritisch, weil die Darstellung möglicherweise unübersichtlich wird und auch der Mehrwert gering ist, wenn es sich um Kleinverbraucher handelt. Hat man ein altes Schloss mit 20 Zimmern und in jedem steht ein Elektroradiator, dann mag das anders sein ;-)

ledermann, 2025-02-10:
Die Erweiterung ist nun verfügbar, Dashboard und PowerSplitter können also jetzt mit bis zu 20 selbstdefinierten Verbrauchern umgehen.

closed 2025-02-10 by ledermann

Zeppelin-PV, 2025-02-10:
herzlichen Dank, Georg
das ging ja ratz-fatz
zwei weitere Verbraucher sind schon eingebunden und werden angezeigt
da sie schon "ewig" lang mitgeloggt wurden sind auch alle Daten da - super
zwei weitere muss ich noch über HA holen - dann ist auch bei mal Schluss mit Einzelverbraucher
LG Ernst