solectrus/solectrus #3974: Skalierung Hausverbrauch Aktuell

Zeppelin-PV, 2025-02-12:
### Beschreibung (auf Deutsch oder Englisch)

hallo Georg
ich habe beobachtet, dass sich die Skalierung im Hausverbrauch Aktuell nicht automatisch anpasst.
Im Diagram war bei 2kW Schluss obwohl der momentane Verbrauch höher war

[image]

[image]

bei einem Diagramwechsel zurück passt sich die Skalierung dann an

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-261-g40452dbc

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspi4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-02-12:
Das kann ich bei mir leider nicht reproduzieren. Ein Diagramm wird zunächst so skaliert, dass die gesamte Kurve sichtbar ist. Wenn dann ein neuer Messwert hereinkommt, wird der rechts ergänzt. Falls erforderlich (also wenn der neue Wert größer ist als das bisherige Maximum der Y-Achse), wird dabei das Diagramm neu skaliert.

### Kleines Video zur Veranschaulichung

[image]

Zeppelin-PV, 2025-02-12:
habe jetzt festgestellt, dass dieses Phänomen nur auf dem Tablet ist, nicht am PC bzw. Laptop - dort wird sofort richtig skaliert.
Am Tablet reicht ein von oben nach unten ziehen um einen Refresh zu machen, dann stimmt auch hier die Skalierung

ledermann, 2025-02-13:
Danke, du hast recht, auf kleineren Displays stimmt die Anpassung der Skalierung nicht. Ich habe das jetzt korrigiert. Zu sehen in `custom-consumer-273`.