solectrus/solectrus #3978: Diagramm Fehler Anzeige Y Achse

Zeppelin-PV, 2025-02-13:
### Beschreibung (auf Deutsch oder Englisch)

sorry Georg,
schon wieder ich
in der Y Achse steht zweimal 1kW, sowohl bei Mobilgeräten als auch auf dem Laptop

[image]

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-273-g92ccd32a

### Auf welchem Gerät läuft bei dir SOLECTRUS?

PI4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-02-14:
Danke für das genaue Hinschauen und die Fehlerreports, das hilft sehr!

Ursache war in diesem Fall die fehlenden Dezimalstellen in der Achsenbeschriftung. Das habe ich nun korrigiert, zu sehen in `custom-consumer-282`.

closed 2025-02-14 by ledermann

gereons, 2025-02-14:
Ich sehe auch in `custom-consumer-282` noch diesen Chart (an dem Tag war das Auto nie an die WB angeschlossen):

[image]

Aber auch an den Tagen an denen ich (mit knapp unter 11 kW) geladen habe geht jetzt der Chart oben bis 15, wo vorher immer auf 11 gekappt wurde:

[image]

Höchster Einzelwert in diesem Chart ist 10,677

ledermann, 2025-02-14:
@gereons Ok, ich habe zwei Dinge geändert:

1) In allen Diagrammen erfolgt nun eine Mindestskalierung auf 50W. Das vermeidet die unschöne Skalierung von -1W bis +1W bei fehlender Kurve. Außerdem wird dadurch eine Kurve mit sehr geringen Messwerten (deutlich unter 50 W) nicht mehr auf volle Höhe hochskaliert.

2) Ich verwende für die Skalierung nur noch die Bordmittel von Chart.js, alles andere hat zu unerwünschten Seiteneffekten geführt. Das Ermitteln des Maximums für y-Achse läuft daher wieder automatisch durch Charts.js und etwas anders als vorher. Die Skaleneinteilung kann jetzt weitere Zwischenpunkte enthalten, wodurch leerer Raum im oberen Bereich vermieden wird.

Zu sehen in `custom-consumers-285`.

gereons, 2025-02-14:
Danke, das sieht jetzt gut aus!