solectrus/solectrus #3985: Prozentanzeige im Minusbereich

Zeppelin-PV, 2025-02-16:
### Beschreibung (auf Deutsch oder Englisch)

hallo Georg,
habe gerade etwas merkwürdiges entdeckt - in der Tagesanzeige von heute (Akku leer, noch keine Erzeugung) wird bei den sonstigen Verbrauchern ein negativer PV-Anteil ausgegeben

[image]

auf der anderen Seite lief die Kaffeemaschine mit 100% aus PV-Strom

[image]

wobei die Kostenberechnung schlüssig zur Situation ist

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-285-gcd0a652b

### Auf welchem Gerät läuft bei dir SOLECTRUS?

PI4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

Zeppelin-PV, 2025-02-16:
auch mit der gerade gezogenen 293
jetzt wird aber bereits erzeugt

[image]

ledermann, 2025-02-16:
Hier habe ich nun einen Fehler korrigiert, der möglicherweise damit zusammenhängt (siehe `custom-consumer-297`). Ganz sicher bin ich nicht, weil mir etwas Kontext fehlt. Ein negativer Wert wird aber jetzt verhindert. 

Es wäre hilfreich, wenn du zukünftig nicht nur einzelne Bildschirm-Schnippsel einstellst, sondern auch die Gesamtsicht. Es ist dann manchmal schwierig, einen Fehler nachzuvollziehen. 

Zur Kaffeemaschine: 

- Eine Energie von 55 Wh ist so gering, dass sie leicht in der Messungenauigkeit verschwinden kann. Insofern wirkt ein PV-Anteil von 100 % nicht unbedingt falsch. 
- Bei meinem SENEC-Speicher bemerke ich, dass selbst ein leerer Akku immer noch kontinuierlich 13W liefert.
- Bei einem Verbraucher, der nur kurz läuft, kann es zu Ungenauigkeiten kommen. Der PowerSplitter arbeitet einer Auflösung von 1min, d.h. es wird immer mit einem Mittelwert über diese eine Minute gerechnet.
- Der PowerSplitter läuft alle 5min, dessen Berechnungen liegen somit verzögert vor. Anders ausgedrückt: Das, was in den letzten bis zu 5 Minuten passiert ist, wurde noch gar nicht aufgesplittet. Das kann sich dann (vorübergehend!) in einem verfälschten PV-Anteil niederschlagen. Das korrigiert sich dann aber im weiteren Verlauf von selbst.

ledermann, 2025-02-16:
@gereons Ja, das sieht mir auch danach aus, als hätte das mit dem Tageswechsel zu tun. Der PV-Anteil am ersten Tag ist zu hoch, am zweiten Tag zu gering (sogar negativ). Da wird offnenbar nicht richtig zwischen den Tagen getrennt. Das werde ich mal nachstellen.

gereons, 2025-02-16:
Wenn ich mit InfluxDB-Dumps helfen kann, sag Bescheid.

ledermann, 2025-02-16:
@gereons Ja, das wäre sehr hilfreich. Am besten genau die zwei Tage, also 14. und 15. Februar 2025, jeweils von 00:00 bis 24:00 CET. Und ganz wichtig: Bitte kein `aggregateWindow` verwenden, sondern die Messwerte in voller Auflösung exportieren.

Meine bisherige Vermutung, dass es wieder ein Zeitzonenproblem ist, scheint sich nicht zu bestätigen. Insofern stochere ich noch im Nebel, die Daten wäre daher nützlich.

gereons, 2025-02-16:
Kein Problem: [query.csv.gz](https://github.com/user-attachments/files/18815529/query.csv.gz)

Meine Verbraucher sind in `shellies:xyz_power` zu finden, der Trockner entsprechend in `shellies:dryer_power`

ledermann, 2025-02-17:
Dank der Daten von @gereons habe ich das Problem nachvollziehen können. Der PowerSplitter rechnet zwar korrekt, der Verlauf der erzeugten Netzbezugs-Kurve ist jedoch leicht versetzt im Vergleich zum Verbrauch. Das liegt daran, dass der PowerSplitter 5min-Abschnitte betrachtet und seinen Datenpunkt bislang am Ende des jeweiligen Abschnitts platziert hat. Bei der Berechnung des Integrals (für die Tageszusammenfassung) kann es dann in seltenen Fällen passieren, dass Netzbezug größer als Verbrauch war - nämlich genau dann, wenn ein Verbraucher kurz vor Mitternacht eingeschaltet wurde und nach Mitternacht wieder ausgeschaltet wurde.

Ich habe nun zwei Dinge geändert:

1) Der PowerSplitter platziert seine Datenpunkte jetzt **in die Mitte** des 5min-Intervalls. Dadurch sind die Kurven nahezu deckungsgleich. 
2) Bei der Berechnung der Tageszusammenfassungen erfolgt eine zusätzliche Prüfung für jeden Verbraucher, ob weiterhin ein Tages-Netzbezug höher ist als der Tages-Verbrauch (nun sehr unwahrscheinlich, aber nicht ausgeschlossen). In diesem Fall wird der Netzbezug gedeckelt.

Somit kommt es jetzt nicht mehr zu einem negativen PV-Anteil.

Um meine Änderungen zu übernehmen, ist Folgendes zu tun:

1) Aktuelle Images holen: `docker compose pull` 
2) Alles neu starten: `docker compose up -d`
3) PowerSplitter neu berechnen: `docker compose kill --signal USR1 power-splitter`. Warten, bis die Berechnung abgeschlossen ist (im Log verfolgen)
4) Tageszusammenfassungen zurücksetzen

Wem dies zu lästig ist, der muss das nicht tun. Die PowerSplitter-Berechnungen ändern sich nur in geringem Maße und auch nur dann, wenn man häufiger kurz vor Mitternacht etwas eingeschaltet hat.

gereons, 2025-02-17:
Prima, danke für den Fix. In meinem Beispiel von oben werden mir jetzt 7 bzw 5% PV-Anteil angezeigt, was zwar nicht 100% korrekt ist, aber doch **deutlich** näher an der Wahrheit als vorher, und mit dieser Ungenauigkeit kann ich problemlos leben.