solectrus/solectrus #4015: Umstellung von custom consumer PR #3747 auf develop

Zeppelin-PV, 2025-02-22:
### Beschreibung (auf Deutsch oder Englisch)

hallo Georg
seit der Umstellung bekomme ich den Fehler 500 wenn ich bei den Verbrauchern Woche und höher anwähle
Aktuell geht, Tag kreisen die drei Punkte schon sichtbar, ab Woche kommt der Fehler 500
kann aber zurück zur Startseit

### Welche Version von SOLECTRUS verwendest du?

v0.18.3-359-gd8562cb7

### Auf welchem Gerät läuft bei dir SOLECTRUS?

PI4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
dashboard-1  | Information for: ActionView::Template::Error (nil can't be coerced into Float):
dashboard-1  | [192.168.1.110]     13:   .flex.flex-col.flex-auto.sm:gap-2
dashboard-1  | [192.168.1.110]     14:     - unless timeframe.out_of_range?
dashboard-1  | [192.168.1.110]     15:       = render ChartSelector::Component.new(sensor:, timeframe:, sensors: chart_sensors)
dashboard-1  | [192.168.1.110]     16:       = render ChartLoader::Component.new(sensor:, timeframe:)
dashboard-1  | [192.168.1.110]   
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:104:in 'Float#-'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:104:in 'ChartData::HousePowerWithoutCustom#other_power_pv'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:58:in 'block in ChartData::HousePowerWithoutCustom#splitted_chart'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:74:in 'block in ChartData::HousePowerWithoutCustom#map_house_power'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:73:in 'Array#map'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:73:in 'Enumerator#with_index'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:73:in 'ChartData::HousePowerWithoutCustom#map_house_power'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:57:in 'ChartData::HousePowerWithoutCustom#splitted_chart'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:19:in 'ChartData::HousePowerWithoutCustom#chart'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:6:in 'ChartData::HousePowerWithoutCustom#data'
dashboard-1  | [192.168.1.110] app/services/chart_data/base.rb:14:in 'ChartData::Base#blank?'
dashboard-1  | [192.168.1.110] app/components/chart_loader/component.html.slim:1:in 'ChartLoader::Component#call'
dashboard-1  | [192.168.1.110] app/views/house/charts/index.turbo_stream.slim:16
dashboard-1  | [192.168.1.110] app/views/house/charts/index.turbo_stream.slim:12
dashboard-1  | [192.168.1.110] app/controllers/house/charts_controller.rb:6:in 'House::ChartsController#index'
dashboard-1  | [192.168.1.110] app/controllers/concerns/auto_locale.rb:14:in 'ApplicationController#switch_locale'
dashboard-1  | [192.168.1.110] 
dashboard-1  | Information for cause: TypeError (nil can't be coerced into Float):
dashboard-1  | [192.168.1.110]   
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:104:in 'Float#-'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:104:in 'ChartData::HousePowerWithoutCustom#other_power_pv'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:58:in 'block in ChartData::HousePowerWithoutCustom#splitted_chart'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:74:in 'block in ChartData::HousePowerWithoutCustom#map_house_power'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:73:in 'Array#map'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:73:in 'Enumerator#with_index'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:73:in 'ChartData::HousePowerWithoutCustom#map_house_power'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:57:in 'ChartData::HousePowerWithoutCustom#splitted_chart'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:19:in 'ChartData::HousePowerWithoutCustom#chart'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:6:in 'ChartData::HousePowerWithoutCustom#data'
dashboard-1  | [192.168.1.110] app/services/chart_data/base.rb:14:in 'ChartData::Base#blank?'
dashboard-1  | [192.168.1.110] app/components/chart_loader/component.html.slim:1:in 'ChartLoader::Component#call'
dashboard-1  | [192.168.1.110] app/views/house/charts/index.turbo_stream.slim:16
dashboard-1  | [192.168.1.110] app/views/house/charts/index.turbo_stream.slim:12
dashboard-1  | [192.168.1.110] app/controllers/house/charts_controller.rb:6:in 'House::ChartsController#index'
dashboard-1  | [192.168.1.110] app/controllers/concerns/auto_locale.rb:14:in 'ApplicationController#switch_locale'
dashboard-1  | [192.168.1.110] method=GET path=/500 format=html controller=ErrorsController action=show status=500 allocations=670 duration=3.32 view=2.08 db=0.00
dashboard-1  | [192.168.1.110] method=GET path=/house/stats/house_power_without_custom/2025-W08 format=html controller=House::StatsController action=index status=200 allocations=60640 duration=226.59 view=132.43 db=91.84
dashboard-1  | [192.168.1.110] method=GET path=/house/charts/house_power_without_custom/2025-W08 format=html controller=House::ChartsController action=index status=500 allocations=29796 duration=127.46 view=0.00 db=5.74
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

closed 2025-02-22 by ledermann

Zeppelin-PV, 2025-02-22:
leider noch derselbe Fehler
der komplette Log

dashboard-1  | SOLECTRUS Photovoltaic Dashboard
dashboard-1  | Copyright (C) 2020-2025 Georg Ledermann
dashboard-1  | License: GNU AGPLv3 - https://www.gnu.org/licenses/agpl-3.0.html
dashboard-1  | Version v0.18.3-359-gd8562cb7 (develop), built on 2025-02-22T14:53:37+01:00
dashboard-1  | Using ruby 3.4.2 (2025-02-15 revision d2930f8e7a) +PRISM [aarch64-linux-musl]
dashboard-1  | Based on Alpine Linux 3.21.3
dashboard-1  | 
dashboard-1  | ## Waiting for services...
dashboard-1  | Redis is up and running!
dashboard-1  | redis (172.18.0.2:6379) open
dashboard-1  | influxdb (172.18.0.5:8086) open
dashboard-1  | InfluxDB is up and running!
dashboard-1  | postgresql (172.18.0.4:5432) open
dashboard-1  | PostgreSQL is up and running!
dashboard-1  | 
dashboard-1  | ## Preparing database...
dashboard-1  | Database is ready!
dashboard-1  | 
dashboard-1  | ## Starting Rails application...
dashboard-1  | => Booting Puma
dashboard-1  | => Rails 8.0.1 application starting in production 
dashboard-1  | => Run `bin/rails server --help` for more startup options
dashboard-1  | Sensor initialization started
dashboard-1  |   - Sensor INVERTER_POWER mapped to SENEC:inverter_power
dashboard-1  |   - Sensor HOUSE_POWER mapped to SENEC:house_power
dashboard-1  |   - Sensor GRID_IMPORT_POWER mapped to SENEC:grid_power_plus
dashboard-1  |   - Sensor GRID_EXPORT_POWER mapped to SENEC:grid_power_minus
dashboard-1  |   - Sensor BATTERY_CHARGING_POWER mapped to SENEC:bat_power_plus
dashboard-1  |   - Sensor BATTERY_DISCHARGING_POWER mapped to SENEC:bat_power_minus
dashboard-1  |   - Sensor CUSTOM_POWER_01 mapped to Gefrierschrank:power
dashboard-1  |   - Sensor CUSTOM_POWER_02 mapped to Waschmaschine:power
dashboard-1  |   - Sensor CUSTOM_POWER_03 mapped to Trockner:power
dashboard-1  |   - Sensor CUSTOM_POWER_04 mapped to Kuehlschrank:power
dashboard-1  |   - Sensor CUSTOM_POWER_05 mapped to Heissluftfritteuse:power
dashboard-1  |   - Sensor CUSTOM_POWER_06 mapped to Spuelmaschine:power
dashboard-1  |   - Sensor CUSTOM_POWER_07 mapped to Wasserkocher:power
dashboard-1  |   - Sensor CUSTOM_POWER_08 mapped to Kaffeemaschine:power
dashboard-1  |   - Sensor CUSTOM_POWER_09 mapped to Backofen:power
dashboard-1  |   - Sensor CUSTOM_POWER_10 mapped to Buegeleisen:power
dashboard-1  |   - Sensor CUSTOM_POWER_11 mapped to AUFZUG:power
dashboard-1  |   - Sensor CUSTOM_POWER_12 mapped to THERME:power
dashboard-1  |   - Sensor CUSTOM_POWER_13 mapped to IT:power
dashboard-1  |   - Sensor CUSTOM_POWER_14 mapped to TERRASSE:power_a
dashboard-1  |   - Sensor CUSTOM_POWER_15 mapped to HERD:power
dashboard-1  |   - Sensor HOUSE_POWER_GRID mapped to power_splitter:house_power_grid
dashboard-1  |   - Sensor BATTERY_CHARGING_POWER_GRID mapped to power_splitter:battery_charging_power_grid
dashboard-1  |   - Sensor CUSTOM_POWER_01_GRID mapped to power_splitter:custom_power_01_grid
dashboard-1  |   - Sensor CUSTOM_POWER_02_GRID mapped to power_splitter:custom_power_02_grid
dashboard-1  |   - Sensor CUSTOM_POWER_03_GRID mapped to power_splitter:custom_power_03_grid
dashboard-1  |   - Sensor CUSTOM_POWER_04_GRID mapped to power_splitter:custom_power_04_grid
dashboard-1  |   - Sensor CUSTOM_POWER_05_GRID mapped to power_splitter:custom_power_05_grid
dashboard-1  |   - Sensor CUSTOM_POWER_06_GRID mapped to power_splitter:custom_power_06_grid
dashboard-1  |   - Sensor CUSTOM_POWER_07_GRID mapped to power_splitter:custom_power_07_grid
dashboard-1  |   - Sensor CUSTOM_POWER_08_GRID mapped to power_splitter:custom_power_08_grid
dashboard-1  |   - Sensor CUSTOM_POWER_09_GRID mapped to power_splitter:custom_power_09_grid
dashboard-1  |   - Sensor CUSTOM_POWER_10_GRID mapped to power_splitter:custom_power_10_grid
dashboard-1  |   - Sensor CUSTOM_POWER_11_GRID mapped to power_splitter:custom_power_11_grid
dashboard-1  |   - Sensor CUSTOM_POWER_12_GRID mapped to power_splitter:custom_power_12_grid
dashboard-1  |   - Sensor CUSTOM_POWER_13_GRID mapped to power_splitter:custom_power_13_grid
dashboard-1  |   - Sensor CUSTOM_POWER_14_GRID mapped to power_splitter:custom_power_14_grid
dashboard-1  |   - Sensor CUSTOM_POWER_15_GRID mapped to power_splitter:custom_power_15_grid
dashboard-1  |   - Sensor INVERTER_POWER_FORECAST mapped to Forecast:watt
dashboard-1  |   - Sensor GRID_EXPORT_LIMIT mapped to SENEC:power_ratio
dashboard-1  |   - Sensor BATTERY_SOC mapped to SENEC:bat_fuel_charge
dashboard-1  |   - Sensor CASE_TEMP mapped to SENEC:case_temp
dashboard-1  |   - Sensor SYSTEM_STATUS mapped to SENEC:current_state
dashboard-1  |   - Sensor SYSTEM_STATUS_OK mapped to SENEC:current_state_ok
[... 242 lines cut ...]
dashboard-1  | [192.168.1.110] app/views/house/charts/index.turbo_stream.slim:16
dashboard-1  | [192.168.1.110] app/views/house/charts/index.turbo_stream.slim:12
dashboard-1  | [192.168.1.110] app/controllers/house/charts_controller.rb:6:in 'House::ChartsController#index'
dashboard-1  | [192.168.1.110] app/controllers/concerns/auto_locale.rb:14:in 'ApplicationController#switch_locale'
dashboard-1  | [192.168.1.110] 
dashboard-1  | Information for cause: TypeError (nil can't be coerced into Float):
dashboard-1  | [192.168.1.110]   
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:104:in 'Float#-'
dashboard-1  | [192.168.1.110] app/services/chart_data/house_power_without_custom.rb:104:in 'ChartData::HousePowerWithoutCustom#other_power_pv'
[... cut ...]
dashboard-1  | [192.168.1.110] app/views/house/charts/index.turbo_stream.slim:12
dashboard-1  | [192.168.1.110] app/controllers/house/charts_controller.rb:6:in 'House::ChartsController#index'
dashboard-1  | [192.168.1.110] app/controllers/concerns/auto_locale.rb:14:in 'ApplicationController#switch_locale'
dashboard-1  | [192.168.1.110] method=GET path=/500 format=html controller=ErrorsController action=show status=500 allocations=9945 duration=44.53 view=43.28 db=0.00
root@raspberrypi:/home/pi/solectrus#

Zeppelin-PV, 2025-02-23:
guten Morgen Georg

ich habe eine etwas feiner Fehlerbeschreibung
Der Absturz auf Fehler passiert nur, wenn ich auf das Feld "Andere" klicke
d.h. ich kann mich sowohl im Zeitbereich Tag, Woche, etc bewegen als auch innerhalb der einzelnen Verbrauchern
hier passiert nichts, - ein Klick auf "Andere" löst den Fehler aus
... vielleicht ist das hilfreich
LG Ernst

ledermann, 2025-02-23:
Danke für deine Mithilfe! Den Fehler habe ich gestern Abend mit `v0.18.3-360` behoben. Laut deinem Log bist du noch auf Version `v0.18.3-359`. 

Es dauert nach einem Commit auf GitHub immer etwa 15min, bis das neue Docker-Image zusammengebaut ist und zur Verfügung steht. Ein einfaches Update sollte jetzt das Problem beheben.

Zeppelin-PV, 2025-02-23:
v0.18.3-363-gd9f4f67f

SUPER - DANKE

commit e0285118 (2025-02-22): Charts: Fix nil handling

Fixes #4015