solectrus/solectrus #4020: Unterschiedlicher Geräte-Status in Strombilanz vs Hausverbrauch

Zeppelin-PV, 2025-02-25:
### Beschreibung (auf Deutsch oder Englisch)

hallo Georg
ich habe mal wieder eine Kleinigkeit ...
Ich hatte heute morgen Smartphone und Tablet mit Solectrus laufen.
Auf dem Smartphone die Strombilanz, auf dem Tablet den Hausverbrauch.
In der Strombilanz wird momentan sehr oft "keine Verbindung" angezeigt (der nächste NPU Reset wird fällig)
bei den Verbrauchern wird konstant "entladen" angezeigt

[image]

[image]

Eine Frage dazu - wieviel Endgeräte können eigentlich zeitgleich auf eine Solectrus-Instanz zugreifen ?
LG Ernst

### Welche Version von SOLECTRUS verwendest du?

v0.18.3-364-gf77d3bf7

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Pi4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-02-25:
Ja, das ist nicht optimal so. Dass auf den beiden Seiten ein unterschiedlicher Status angezeigt wird, hat folgenden Grund: 

In der Strombilanz werden (bei Verwendung eines SENEC-Stromspeichers) nur Werte aus dieser einen Quelle angezeigt. Bei einem Ausfall (z.B. durch NPU-Fehler) ist der jüngste vorliegende Messwert zu alt und es wird ein (roter) Fehlerstatus angezeigt ("Keine Verbindung"). Das ist korrekt so.

Auf der Hausverbrauchs-Seite werden Werte aus **anderen** Quellen angezeigt (in deinem Fall Shelly-Stromzähler). Diese liefern weiter aktuelle Messwerte, weil sie unabhängig vom SENEC-Stromspeicher sind. Daher wird ein (grüner) Ok-Status angezeigt. Der Text ("Entladen") ist aber falsch bzw. irritierend, weil das der Status vom SENEC ist, der auf dieser Seite keine Rolle spielt. Na ja, **fast** keine Rolle, weil der Gesamtstromverbrauch ja auch angezeigt wird - links in der Klammer.

Ich habe das nun so geändert: Auf der Hausverbrauchs-Seite wird nur noch zwischen "Verbunden" und "Keine Verbindung" unterschieden, je nachdem, ob aktuelle Verbrauchswerte vorliegen oder nicht. Der SENEC-Status wird also **nicht mehr** angezeigt. In der Strombilanz bleibt alles unverändert.

Nach Update auf `v0.18.3-368` wird in deinem Fall (SENEC ist ausgefallen, Shelly sind weiter in Betrieb) nicht mehr "Entladen" angezeigt, sondern "Verbunden".

> Eine Frage dazu - wieviel Endgeräte können eigentlich zeitgleich auf eine Solectrus-Instanz zugreifen ?

Wenn du mit "Endgeräte" deine Tablets, Smartphones und Desktop-Browser meinst: Da gibt es keine Begrenzung, das hängt nur von der Leistungsfähigkeit des Servers (Raspberry) ab. Bei Überlastung dauern die Antworten länger. Über die Variable `RAILS_MAX_THREADS` lässt sich unter Inkaufnahme von mehr Speicherverbrauch zusätzliche Power herauskitzeln (um **gleichzeitige** Requests schneller zu bedienen). Bei einer nicht-öffentlichen SOLECTRUS-Instanz ist das aber wirklich nicht relevant. Und mit der Status-Anzeige hat das nichts zu tun.

Zeppelin-PV, 2025-02-25:
DANKE - optimal erklärt
das war mir nicht bekannt, die differenzierte Anzeige des Status