solectrus/solectrus #4021: min/max error

frizzle0815, 2025-02-25:
Hallo Georg, 
ich weiß nicht, ob es ein Problem des Power Splitter oder des Dashboard ist.

Ich habe auf die aktuelle `develop` des power-splitter umgestellt, so wie von dir [hier ](https://github.com/orgs/solectrus/discussions/3749#discussioncomment-12107973) erläutert. Dashboard läuft seitdem auch auf `develop`, vorher `pr-3747`.

Ich habe `kill --signal USR1` ausgeführt und danach war der Splitter erst einmal gut eine halbe Stunde lang beschäftigt neue Berechnungen nach InfluxDB zu schieben. Danach habe ich im Dashboard die Tageszusammenfassungen gelöscht.

Wenn ich mir dann den aktuellen Monat (Februar) anschauen will, klappt die Berechnung der Tageszusammenfassung problemlos.
Möchte ich dies für den Januar oder das gesamte Jahr machen, dann bricht er mit einem Fehler 500 ab.

Im Dashboard log kommt das hier als zentraler Fehler `ArgumentError (min argument must be less than or equal to max argument)`.
Sagt dir das etwas? Ich benutze bei einigen Mappings das _MIN und _MAX, aber die sind alle richtig eingestellt und liefen bislang problemlos.

```
2025-02-25T10:42:29.681062665Z [192.168.7.7] method=GET path=/inverter_power/2025 format=html controller=Balance::HomeController action=index status=200 allocations=20627 duration=213.31 view=209.80 db=1.37

2025-02-25T10:42:29.766663898Z [192.168.7.7] [ActiveJob] [SummarizerJob] [633f03e0-045d-4008-b50a-b39bf323c3ad] Performing SummarizerJob (Job ID: 633f03e0-045d-4008-b50a-b39bf323c3ad) from Async(default) with arguments: Mon, 06 Jan 2025

2025-02-25T10:42:29.777781803Z [192.168.7.7] [ActiveJob] [SummarizerJob] [633f03e0-045d-4008-b50a-b39bf323c3ad] Error performing SummarizerJob (Job ID: 633f03e0-045d-4008-b50a-b39bf323c3ad) from Async(default) in 10.56ms: ArgumentError (min argument must be less than or equal to max argument):

2025-02-25T10:42:29.777991018Z /app/app/services/summary_corrector.rb:87:in 'Comparable#clamp'

2025-02-25T10:42:29.778068018Z /app/app/services/summary_corrector.rb:87:in 'block in SummaryCorrector#scale_and_adjust'

2025-02-25T10:42:29.778131614Z /app/app/services/summary_corrector.rb:85:in 'Hash#to_h'

2025-02-25T10:42:29.778191205Z /app/app/services/summary_corrector.rb:85:in 'SummaryCorrector#scale_and_adjust'

2025-02-25T10:42:29.778251709Z /app/app/services/summary_corrector.rb:66:in 'SummaryCorrector#fix_grid_values'

2025-02-25T10:42:29.778316601Z /app/app/services/summary_corrector.rb:25:in 'SummaryCorrector#adjusted'

2025-02-25T10:42:29.778382934Z /app/app/jobs/summarizer_job.rb:54:in 'SummarizerJob#correct_values'

2025-02-25T10:42:29.778432708Z /app/app/jobs/summarizer_job.rb:31:in 'block in SummarizerJob#perform_calculations'

2025-02-25T10:42:29.778500117Z /usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_adapters/abstract/transaction.rb:626:in 'block in ActiveRecord::ConnectionAdapters::TransactionManager#within_new_transaction'

2025-02-25T10:42:29.778843246Z /usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/concurrency/null_lock.rb:9:in 'ActiveSupport::Concurrency::NullLock#synchronize'

2025-02-25T10:42:29.778921500Z /usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_adapters/abstract/transaction.rb:623:in 'ActiveRecord::ConnectionAdapters::TransactionManager#within_new_transaction'

2025-02-25T10:42:29.778979243Z /usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_adapters/abstract/database_statements.rb:367:in 'ActiveRecord::ConnectionAdapters::DatabaseStatements#within_new_transaction'

2025-02-25T10:42:29.779046568Z /usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_adapters/abstract/database_statements.rb:359:in 'ActiveRecord::ConnectionAdapters::DatabaseStatements#transaction'

2025-02-25T10:42:29.779110860Z /usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/transactions.rb:234:in 'block in ActiveRecord::Transactions::ClassMethods#transaction'

2025-02-25T10:42:29.779177188Z /usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_adapters/abstract/connection_pool.rb:418:in 'ActiveRecord::ConnectionAdapters::ConnectionPool#with_connection'

2025-02-25T10:42:29.779243395Z /usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/connection_handling.rb:310:in 'ActiveRecord::ConnectionHandling#with_connection'

2025-02-25T10:42:29.779306710Z /usr/local/bundle/gems/activerecord-8.0.1/lib/active_record/transactions.rb:233:in 'ActiveRecord::Transactions::ClassMethods#transaction'

2025-02-25T10:42:29.779371615Z /app/app/jobs/summarizer_job.rb:20:in 'SummarizerJob#perform_calculations'

2025-02-25T10:42:29.779420204Z /app/app/jobs/summarizer_job.rb:7:in 'SummarizerJob#perform'

2025-02-25T10:42:29.779477712Z /usr/local/bundle/gems/activejob-8.0.1/lib/active_job/execution.rb:68:in 'block in ActiveJob::Execution#_perform_job'

2025-02-25T10:42:29.779543946Z /usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/callbacks.rb:120:in 'block in ActiveSupport::Callbacks#run_callbacks'

2025-02-25T10:42:29.779605955Z /usr/local/bundle/gems/i18n-1.14.7/lib/i18n.rb:348:in 'I18n::Base#with_locale'

2025-02-25T10:42:29.779666633Z /usr/local/bundle/gems/activejob-8.0.1/lib/active_job/translation.rb:9:in 'block (2 levels) in <module:Translation>'

2025-02-25T10:42:29.779733615Z /usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/callbacks.rb:129:in 'BasicObject#instance_exec'

2025-02-25T10:42:29.779798638Z /usr/local/bundle/gems/activesupport-8.0.1/lib/active_support/callbacks.rb:129:in 'block in ActiveSupport::Callbacks#run_callbacks'

[... 482 lines cut ...]
2025-02-25T10:42:29.856013328Z [192.168.7.7] app/services/summary_corrector.rb:87:in 'block in SummaryCorrector#scale_and_adjust'

2025-02-25T10:42:29.856063177Z [192.168.7.7] app/services/summary_corrector.rb:85:in 'Hash#to_h'

2025-02-25T10:42:29.856108781Z [192.168.7.7] app/services/summary_corrector.rb:85:in 'SummaryCorrector#scale_and_adjust'

2025-02-25T10:42:29.856154672Z [192.168.7.7] app/services/summary_corrector.rb:66:in 'SummaryCorrector#fix_grid_values'

2025-02-25T10:42:29.856203182Z [192.168.7.7] app/services/summary_corrector.rb:25:in 'SummaryCorrector#adjusted'

2025-02-25T10:42:29.856249600Z [192.168.7.7] app/jobs/summarizer_job.rb:54:in 'SummarizerJob#correct_values'

2025-02-25T10:42:29.856295592Z [192.168.7.7] app/jobs/summarizer_job.rb:31:in 'block in SummarizerJob#perform_calculations'

2025-02-25T10:42:29.856341952Z [192.168.7.7] app/jobs/summarizer_job.rb:20:in 'SummarizerJob#perform_calculations'

2025-02-25T10:42:29.856396279Z [192.168.7.7] app/jobs/summarizer_job.rb:7:in 'SummarizerJob#perform'

2025-02-25T10:42:29.856471609Z [192.168.7.7] app/controllers/summaries_controller.rb:8:in 'SummariesController#show'

2025-02-25T10:42:29.856518938Z [192.168.7.7] app/controllers/concerns/auto_locale.rb:14:in 'ApplicationController#switch_locale'

2025-02-25T10:42:29.907266248Z [192.168.7.7] method=GET path=/500 format=html controller=ErrorsController action=show status=500 allocations=10051 duration=51.37 view=41.79 db=0.00

```

ledermann, 2025-02-25:
@frizzle0815 Danke für den Fehlerreport. Das ist kein Problem im PowerSplitter, sondern im Dashboard. Ob es mit [Min/Max bei den MQTT-Mappings](https://docs.solectrus.de/referenz/mqtt-collector/topics#mapping_x_min-optional-ab-v030) etwas zu tun hat, kann ich noch nicht sagen.

Auslöser für den Fehler ist jedenfalls, dass bei der Berechnung der Tageszusammenfassung für den **06.01.2025** irgendwo ein **negativer** Stromverbrauch vorkommt. 

Ich habe gerade im Dashboard eine Änderung vorgenommen, die negative Werte ignoriert, aber das ist nur eine Symptombehandlung, damit kein Fehler 500 auftritt. Möglicherweise gibt es dadurch Probleme an anderer Stelle.

Schau dir nach Update auf `v0.18.3-369` mal den Tag **06.01.2025** an, ob dir da etwas auffällt. Prüfe ggfs. über das InfluxDB-UI (falls du damit zurechtkommst), ob sich weitere Hinweise finden, wie das zustande gekommen ist.

frizzle0815, 2025-02-25:
Hallo Georg, 
herzlichen Dank für das schnelle einschreiten. Mit der `v0.18.3-369` wird die Tageszusammenfassung problemlos erstellt.

Ein Blick in InfluxDB offenbart, dass beim Feld `house_power` tatsächlich immer mal wieder kurzzeitig negative Werte auftauchen. Das liegt dann aber an meinem persönlichen MQTT-Setup. Ich mache hier einige Nebenrechnungen, um sicherzustellen, dass im Dashboard immer Herkunft = Verwendung ist. Vermutlich fehlen ihm hier kurzzeitig Messwerte und dann wird falsch gerechnet.

Ich habe jetzt mal ein `MAPPING_1_MIN = 0` für die env des mqtt-collectors hinzugefügt. Damit sollten negative Werte dann ja zukünftig ignoriert werden.

ledermann, 2025-02-25:
Top, das klingt plausibel. Ich denke auch, dass mit `MAPPING_1_MIN = 0` das Problem behoben ist.

closed 2025-02-25 by ledermann