solectrus/solectrus #4060: keine Anzeige "erwartet wurden"

Zeppelin-PV, 2025-03-08:
### Beschreibung (auf Deutsch oder Englisch)

hallo Georg,
heute habe ich festgestellt, dass in der develop Version (v0.18.3-431-g52745463) auf Mobilgeräten die Anzeige
"erwartet wurden" nicht erscheint - bei Desktopgeräten geht es

Bei der latest Version 18.3. geht es bei Mobilgeräten und Desktopgeräten

LG Ernst

### Welche Version von SOLECTRUS verwendest du?

v0.18.3-431-g52745463  vs  18.3

### Auf welchem Gerät läuft bei dir SOLECTRUS?

PI4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-03-09:
Danke für den Hinweis, ist behoben in `v0.18.3-434`

closed 2025-03-09 by ledermann

Zeppelin-PV, 2025-03-09:
[image]

DANKE !!!