solectrus/solectrus #4068: Seit 0.19.0 wir auf einmal der Browser nicht mehr unterstützt.

StevieC121176, 2025-03-12:
### Beschreibung (auf Deutsch oder Englisch)

Seit der Version 0.19.0 läuft solectrus nicht mehr auf meinem Kühlschrank . Ich nutze die develop Version von Solectrus und bis vorgestern ging es noch ohne Probleme.

[image]

### Welche Version von SOLECTRUS verwendest du?

0.19.0

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspberry pi5

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-03-12:
Danke für den Hinweis, das soll natürlich nicht so sein.

[Vor zwei Wochen (mit v0.18.3-375)](https://github.com/solectrus/solectrus/commit/2ac58b59b4592e2b6be78610893796982b459589) habe ich auf TailwindCSS 4 aktualisiert, was leider höhere Browser-Anforderungen mit sich bringt ([Chrome 111+, Safari 16.4+, Firefox 128+](https://tailwindcss.com/docs/compatibility#browser-support)). 

Die Browser-Version wurde aber zunächst nicht geprüft, das habe ich erst [vor vier Tagen (mit v0.18.3-430)](https://github.com/solectrus/solectrus/commit/ef69d1424a41faaf2a88d0768d095935c88b38cc) ergänzt. Das habe ich gemacht, weil SOLECTRUS z.B. auf einem sehr alten Safari wirklich nicht mehr funktioniert und völlig kaputt aussieht. Insofern wundert es mich, dass das Problem nicht bereits vor vier Tagen bei dir aufgetreten ist.

Welcher Browser ist denn auf dem Kühlschrank installiert? Prüfen lässt sich das z.B. mit https://www.whatismybrowser.com/detect/what-is-my-user-agent/

Vermutlich wird der Kühlschrank kein Update erlauben. Ich habe daher jetzt erstmal die Browser-Prüfung wieder rausgenommen (`develop` in `v0.19.0-3`). Folgende Bitte nun an dich:

1. Stelle wieder auf `develop`, die Browser-Version wird dann nicht mehr geprüft, d.h. SOLECTRUS wird dann wieder erscheinen
2. Prüfe bitte genauer, ob Darstellung und Bedienung wirklich auf dem Kühlschrank funktionieren. 

Je nach Ergebnis würde ich den Browser-Check dann wieder einbauen, aber mit weniger restriktiven Grenzen. Möglicherweise verwende ich in SOLECTRUS gar nicht die neueren CSS-Features, die bei TailwindCSS zu dieser strengen Limitierung führen.

StevieC121176, 2025-03-12:
Ich war bis vorgestern auf Dienstreise, daher ist es mir nicht eher aufgefallen.

ich habe auf Version 0.19.0-3 aktualisiert (auf dem `develop` Zweig bin ich sowieso) und damit läuft es wieder. Probleme kann ich nicht feststellen. Es lässt sich alles wunderbar bedienen und grafische Fehler gibt es auch nicht.

Den Browsercheck auf dem Kühlschrank habe ich gemacht, mit diesem Ergebnis:

[image]

ledermann, 2025-03-12:
[Chrome 59](https://chromereleases.googleblog.com/2017/06/stable-channel-update-for-desktop.html) stammt aus 2017 und ist acht Jahre alt. Damit funktioniert SOLECTRUS auf gar keinen Fall. Ich habe hier vorhin auf einem alten Gerät einen Test mit Chrome 109 gemacht, auch damit funktioniert es nicht. SOLECTRUS verwendet (wegen TailwindCSS 4) Farben nach dem OKLCH-Modell, die [erst mit Chrome 111 eingeführt](https://caniuse.com/mdn-css_types_color_oklch) wurden.

Somit kann es eigentlich nur sein, dass sich der Browser deines Kühlschranks mit einem falschen UserAgent meldet, tatsächlich aber etwas Moderneres läuft. Hast du vielleicht den UserAgent manuell angepasst? Denkbar ist auch, dass der Hersteller das gemacht hat und man keinen Einfluss darauf hat.

Ich werde mir überlegen, wie ich damit am besten umgehe.

StevieC121176, 2025-03-12:
manuell habe ich nichts angepasst. man kann da auch nicht wirklich etwas einstellen. Ich weiß nur wo wir den Kühlschrank Ende 2023 gekauft haben, lief Solectrus nicht. Danach kamen einige Updates und damit ging es dann. Das letzte Update ist von August 2024.
Aktuell läuft es ohne Probleme, selbst auf Vollbild kann man umschalten. Farben werden auch korrekt dargestellt, auch die Geschwindigkeit beim bedienen ist OK. 

[image]
[image]

ledermann, 2025-03-12:
Ok, das scheint wirklich eine etwas exotische Situation bei deinem Kühlschrank zu sein. Das Menü im Vollbild-Modus erscheint auch nicht korrekt, aber das ist ein anderes Thema und nicht ganz so wichtig.

Ich habe nun eine Lösung gefunden (in `develop` ab `v0.19.0-6`).

Die Browser-Warnung erscheint jetzt wieder, sofern der Browser gemäß User-Agent zu alt ist - also auch bei dir. Allerdings gibt es nun die Möglichkeit, mit einem Button dennoch fortzufahren. Der User-Agent wird außerdem angezeigt.

Der Button setzt ein permanentes Cookie, sodass zukünftige Prüfungen dann unterbleiben. Damit sind jetzt beide Fälle abgedeckt: Wirklich veraltete Browser werden abgewiesen, in begründeten Ausnahmefällen kommt man aber dennoch weiter.

[image]

closed 2025-03-12 by ledermann

StevieC121176, 2025-03-12:
ja guter Kompromiss. Funktioniert. 

bezüglich Menü im Vollbildmodus. Das sieht nur beim Wechsel von normal in Vollbildansicht so aus. Wenn man das Menü dann wieder öffnet wird es als Seitenmenü korrekt angezeigt.