solectrus/solectrus #4087: Liniendiagramm Hausverbrauch bei custom consumer schaltet um auf Andere Verbraucher

Zeppelin-PV, 2025-03-16:
### Beschreibung (auf Deutsch oder Englisch)

folgendes Phänomen habe ich gerade beobachtet. Die Seite mit den selbstdefinierten Verbrauchern ist angewählt.
Als Diagramm wird der Hausverbrauch angezeigt. Lässt man nun dies so stehen kommen von rechts die aktualisierten Werte herein.
Das Diagramm springt nun vom Wert Hausverbrauch runter und zeigt den Wert der anderen Verbrauchern.
Ein klick irgendwohin zeigt den Hausverbrauch wieder normal, als ob nichts gewesen wären

[image]

### Welche Version von SOLECTRUS verwendest du?

v0.19.0-25-g5c4ccc22

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Pi 5

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-03-17:
Kann ich derzeitig nicht reproduzieren (auf einem iPad). Ich werde das mal weiter beobachten...

Zeppelin-PV, 2025-03-17:
Hallo Georg
Nachdem ich das Problem heute mit anderen Browsern untersucht habe
bei denen kein einziger diese seltsame Darstellung zeigte
Ist nun auch der Kiwi Browser ohne Fehler - seltsam
Problem erledigt
LG Ernst

ledermann, 2025-03-18:
Ok, dann schließe ich das erst mal.

closed 2025-03-18 by ledermann