solectrus/solectrus #4102: Kapazitätsauslastung der Batterie visualisieren

ledermann, 2025-03-21:
### Beschreibung (auf Deutsch oder Englisch)

Die Kapazitätsauslastung der Batterie an einem Tag lässt sich als Differenz zwischen dem höchsten und niedrigsten Ladestand (SOC) auffassen:

- Eine Batterie, die am selben Tag sowohl 0 % als auch 100 % erreicht, hat 100 % ihrer Kapazität genutzt
- Eine Batterie, die sich an einem Tag nur zwischen 40 % und 100 % bewegt, nutzt an diesem Tag 60 % ihrer Kapazität

Die Auslastung in einem längeren Zeitraum wäre das Mittel der Tagesauslastungen.

Min(SOC) und Max(SOC) sind bereits in den Tageszusammenfassungen enthalten, die Berechnungsgrundlage ist also vorhanden. SOLECTRUS könnte darauf basierend die Auslastung im Zeitverlauf visualisieren (z.B. Säulen nach Monaten). Langfristige Auswertungen (über mehrere Jahre) dürften interessanter sein als kurzfristige.

Daraus ließe sich erkennen, ob die Batterie eher über- oder unterdimensioniert ist und wie sich Änderungen wie der Einbau zusätzlicher Akkupacks oder die Inbetriebnahme großer Stromverbraucher (z. B. Wärmepumpe) auswirken.

Hier ein Beispiel, wie das grob aussehen könnte:
[image]

(Basierend auf den Messwerten meines SENEC.Home V3 hybrid/duo mit 10 kWh, erstellt mit ChatGPT - erkennbar ist die höhere Auslastung seit Installation der Wärmepumpe in 11/2023)

gereons, 2025-03-21:
Sehr coole Idee! Was ich in ähnlichem Zusammenhang auch schonmal überlegt habe ist eine Angabe, an wievielen Tagen pro Woche/Monat/Jahr der Akku ausgereicht hat um in der Nacht keinen Strom nachkaufen zu müssen.

StevieC121176, 2025-03-24:
im Bezug auf den SOC der Batterie der nächsten Tage könnte man auch den Forecast hinzunehmen, um morgen/übermorgen den zur jeweiligen Uhrzeit möglichen SOC vorauszusagen um entsprechend planen zu können.

WolfImBusch, 2025-03-31:
Hallo zusammen!!

Nur ein Gedanke/eigentlich ne Frage 😅 ... wäre die Berechnung/Grafik auch dann aussagekräftig, wenn der Hausspeicher über das Netz (zB bei niedrigen Strompreisen) und nicht nur über PV geladen werden würde?

Viele Grüße!!

ledermann, 2025-03-31:
@WolfImBusch Ja, davon würde ich ausgehen. Egal, woher der Strom für die Batterie kommt - die Auslastung des Speichers wäre interessant. Also im Sinne von: "Welcher Teil der Kapazität (in kWh) wird überhaupt genutzt".