solectrus/solectrus #4114: Diagramm in Aktuell fehlt - Daten sind durchgängig da in Heute

Zeppelin-PV, 2025-03-25:
### Beschreibung (auf Deutsch oder Englisch)

hallo Georg,
ich habe heute schon folgendes festgestellt
Ein Sensorwert (Radio & TV) liegt kontiunierlich an und wird bei Heute korrekt dargestellt. In Aktuell wird "keine Daten" angezeigt.

[image]

[image]

In einem ähnlichen Sensor (Herd) funktionieren beide Anzeigen Aktuell und Tag

[image]

[image]

### Welche Version von SOLECTRUS verwendest du?

latest & develop

### Auf welchem Gerät läuft bei dir SOLECTRUS?

PI 5

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-03-25:
Die exakt geraden, durchgezogenen Linien im Diagramm sind verdächtig, das deutet auf fehlende Messwerte hin. Da SOLECTRUS nicht weiß, wie oft Daten kommen **müssten**, wird der letzte Wert erst mal fortgeführt. Das heißt nicht unbedingt, dass Daten auch ankommen.

Die erste Anlaufstelle für Probleme dieser Art sind die Logs. In diesem Fall das Log des MQTT-Collector, also die Stelle, wie die Messwerte für "Radio & TV" aufschlagen. Kommt wirklich etwas an für `MAPPING_11`?

Meine Vermutung ist: Dein Sensor für Radio&TV liefert nur, wenn ein Verbrauch auch stattfindet. Gegen 02:00 heute Morgen endete der Verbrauch, der letzte Messwert war 5W. SOLECTRUS kann aber nicht eindeutig erkennen, ob das Gerät ausgeschaltet wurde und geht von einem anhaltenden Verbrauch von 5W aus.

Beim Herd ist es anders, weil der bei dir über einen Shelly gemessen wird. Dieser sendet standardmäßig kontinuierlich 0W, auch wenn kein Verbrauch stattfindet.

ledermann, 2025-03-25:
Du könntest `grep` nutzen, um das Log zu filtern:

```
docker compose logs mqtt-collector --since=5m --timestamps -f | grep Multimedia_Geraete:power
```

Zeppelin-PV, 2025-03-27:
... soweit mal DANKE Georg
es ist eindeutig ein Problem von HA, 
ist dort ein Wert NULL, dann wird solange nicht aktualisiert bis eine Sensoränderung kommt.
für Solectrus ist dieser Punkt erledigt, das dann gar keiner war
LG Ernst

spirit0607, 2025-03-27:
> ... soweit mal DANKE Georg es ist eindeutig ein Problem von HA, ist dort ein Wert NULL, dann wird solange nicht aktualisiert bis eine Sensoränderung kommt. für Solectrus ist dieser Punkt erledigt, das dann gar keiner war LG Ernst

Dieses „Problem“ hatte ich auch. Ich habe nun einfach einen Zeitauslöser für den MQTT Push genommen. Es wird nun alle 5 Sekunden der aktuelle Verbrauchswert per MQTT veröffentlicht, ganz unabhängig vom tatsächlichen Verbrauch. So entstehen zwar deutlich mehr Daten, jedoch ist die Darstellung nun korrekt.

Zeppelin-PV, 2025-03-27:
oh, interessant - ... und wie sieht das Ganze aus ?
für einen Wink mit Zaunpfahl wäre ich dankbar :)

spirit0607, 2025-03-27:
Ja klar, gerne. Ich habe folgende Automation in HomeAssistant, welcher den Stromverbrauch meiner Waschmaschine alle 10 Sekunden (habe mich mit den 5 Sekunden versehen) per MQTT veröffentlich und an den MQTT Broker sendet.

Du kannst in HomeAssistant einfach in die Automationen gehen und oben rechts in den .yaml Modus wechseln, dann kannst du den Code unten einfügen und zurück in den UI Modus gehen und an deine Sensoren anpassen.

```
alias: MQTT Leistung Waschmaschine
description: Sendet die aktuelle Leistung an den MQTT Broker auf der NAS
triggers:
  - trigger: time_pattern
    seconds: "10"
conditions: []
actions:
  - data:
      qos: "0"
      retain: false
      topic: hassio/waschmaschine/watt
      payload: "{{ states('sensor.waschmaschine_power') }}"
    action: mqtt.publish
mode: single
```

Zeppelin-PV, 2025-03-28:
danke Dir, @spirit0607 
bei mir ist es so, dass der Sensor "Multimedia Geräte" ein Summensensor aus HA ist. Bestehend aus drei Geräten die ein Multiroom System sind, und zwei TVs. Einer davon läuft bis tief in die Nacht - ist so. Irgendwann um 2Uhr ist dann der Summensensor NULL.
In HA ist morgens auch zu sehen, zuletzt aktualisiert vor 2 Stunden, z.B.
Seltsamerweise zeigt Solectrus im "TAG" dann durchgehend eine NULL, wenn alles aus ist, im "AKTUELL" jedoch steht keine Daten.
Ein zwangsweises Aktualisieren hat diese Nacht nichts gebracht. Es ist auch so, dass der Verbund Summensensor aus verschiedenen Messsteckdosen besteht. Shelly Plug und Zigbee NOUS A1Z. Als nächstes werde ich den Zigbee durch einen Shelly ersetzten und das beobachten. Meine Grundidee Zigbee zu nutzen war, nicht so viele Aktoren im WLAN, Zigbee baut auch ein schönes Mesh auf.
Im Nachhinein aber stelle ich fest, dass die Zigbee Adapter im unteren Strombereich recht ungenau sind. Habe da Lehrgeld bezahlt und werde wichtige Messungen doch durch Shelly ersetzen.

spirit0607, 2025-03-28:
Okay, aber das müsste an sich trotzdem funktionieren. Du müsstest dann nur deinen "Sammelsensor" in die o.g Automation einfügen. Dann hast du alles paar Sekunden diesen Wert (ob 0 oder höherer Verbrauch) in SOLECTRUS drin

Zeppelin-PV, 2025-03-28:
... bin mir nun auch nicht ganz sicher ob ich alles richtig gemcht habe - es hat in der Automation auf jeden Fall aufgeblinkt, ryhtmisch
hier mal sicherheitshalber das Topic, und der Sensor
sensor.multimedia_geraete
welches Topic muss ich nehmen ?
homeassistant/sensor/multimedia_gerate
homeassistant/sensor/multimedia_gerate/state

spirit0607, 2025-03-28:
Du nimmst das Topic, welches du im MQTT Collector für den Sensor definiert hast. Das kannst du frei gestalten und dir aussuchen nur muss beides gleich sein, also in deiner Automation und im MQTT Collector

Zeppelin-PV, 2025-03-29:
Die Unkenntnisse die man zuweilen hat, holen einen dann doch wieder ein. 
Ich hatte beim Anlegen des Summensensors einen Umlaut benutzt "Geräte". Dies wurde dann vom HA in "Gerate" umgesetzt.
Beim Hinzufügen des Sensors in Solectrus hatte ich dann den Topic aus HA kopiert und übernommen. Das funktionierte auch so.
Jetzt bei dem Erzeugen einer Automation in HA für den MQTT Refresh hatte ich das Wort "Gerate" getippt, und nicht kopiert.
Die automation lief, hat aber keinen Refresh erzeugt. Heut früh habe ich nun den Wert in die Automation auch kopiert, und siehe da,
der Refresh funktioniert und Solectrus empfängt jetzt den Sensor, alle 5 Sek. Ist dieser Wert ok, oder kann man da auch etwas höher gehen ohne dass Solectrus meldet "keine Daten vorhanden" ?
DANKE an Euch fürs mithelfen

ledermann, 2025-03-29:
@Zeppelin-PV Prima, dass es nun klappt. Beim Intervall wäre 5s gut, wenn das Gerät in Betrieb ist, dann hat man eine gute Live-Darstellung. Wenn das Gerät nicht in Betrieb ist, würde eine 0 einmal pro Minute genügen.

In diesem Zusammenhang noch ein Hinweis: Beim Shelly-Collector (und nur dort) gibt es den [esssential mode](https://docs.solectrus.de/referenz/shelly-collector/konfiguration.html#influx_mode-ab-version-050). Dieser vermeidet das endlose Schreiben von 0-Werten. Für ein Gerät, das sich ausschaltet, wird dann genau einmal eine 0 gesendet und dann nichts mehr. Das nutze ich für selten laufende Gerät (Waschmaschine etc.)

closed 2025-03-29 by ledermann