solectrus/solectrus #4131: Steckerstatus wird nicht mehr abgebildet

bennysweb, 2025-03-29:
### Beschreibung (auf Deutsch oder Englisch)

Seit einiger Zeit (ich kann leider nicht die genaue Version sagen) wird der Steckerstaus nicht mehr korrekt im Dashbord angezeigt (Auto eingesteckt: weißes Auto-Icon; Auto nicht eingesteckt: graues Auto-Icon). Der Status wird wie bisher per MQTT geliefert, in die Datenbank geschrieben und auch der Sensor wird vom Dashbord initialisiert - bloß halt nicht korrekt angezeigt. Das Icon bleibt weiß (=Stecker eingesteckt). An der .env und docker-compose habe ich nichts geändert.

### Welche Version von SOLECTRUS verwendest du?

v19.0.70

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Synology

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-03-29:
Kommt der Wert für `INFLUX_SENSOR_WALLBOX_CAR_CONNECTED` wirklich per MQTT an? Steht er also in den Logs vom MQTT-Collector?

Wenn ja, würde ich weiterhin prüfen, ob das Feld wirklich in InfluxDB landet. Findest du es mit dem Admin-UI von InfluxDB wirklich an der Stelle, wo es laut Sensor-Konfiguration auch sein sollte? Hat es den richtigen Wert? 

In meinen Fall sieht es so aus:
[image]

Der Wert steht also auf `false`, d.h. das Auto ist **nicht** angeschlossen. Entsprechend sieht das Icon korrekt so aus:
[image]

bennysweb, 2025-03-30:
Ja, es kommt definitv in InfluxDB an, aber als integer (0 und 1). Bei dir ist es ja true/false. Eventuell liegt es daran? Wie gesagt, es hat bisher immer funktioniert und ich habe in letzter Zeit nichts geändert.

ledermann, 2025-03-30:
Ah, jetzt ist mir klar, was da passiert: Die Boolean-Behandlung ist in `v0.19.0` durch diverse Umbauten kaputtgegangen. Derzeitig wird eine `0` behandelt wie `true`. Das habe ich gerade repariert, in `develop` ab `v0.19.0-76`. Damit sollte es auch bei dir wieder funktionieren.

closed 2025-03-30 by ledermann

bennysweb, 2025-03-31:
Ja, funktioniert bei mir auch wieder, danke ☺️