solectrus/solectrus #4132: Klick auf Solectrus-Logo führt nicht mehr zu "Aktuell"-Ansicht

bennysweb, 2025-03-30:
### Beschreibung (auf Deutsch oder Englisch)

Der Klick auf das Solectrus-Logo oben links im Dashbord führt nicht mehr zur Aktuell-Ansicht, sondern aktualisiert nur die gerade aufgrufene Seite (Woche, Monat, Jahr usw. mit Ansicht "Erzeugung"). Bisher war es eigentlich so, dass man mit Klick auf das Symbol auf die Aktuell-Ansicht (quasi die Startseite des Dashbords) zurück kommt. Bug oder Feature ;-) ? Getestet auf meheren Endgeräten.

### Welche Version von SOLECTRUS verwendest du?

v19.0.71

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Synology

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-03-30:
Das hat sich tatsächlich geändert (mit Version 0.19.0), weil die Seite Hausverbrauch dazu gekommen ist. Das Logo führt jetzt zur Strombilanz (unter Beibehaltung des gewählten Zeitraums).

Mal sehen, vielleicht ändere ich das so, dass ein weiterer Klick dann doch zur "Aktuell" führt. Das wäre dann ein kontextabhängiger Home-Button.

Ich lasse das mal offen und experimentiere damit.

ledermann, 2025-03-30:
Ich habe das jetzt so geregelt:

- Befindet man sich gerade auf der neuen Seite "Hausverbrauch", führt ein Klick auf das Logo zur Strombilanz bei gleichem Zeitfenster (wie bislang bei v0.19.0)
- Befindet man sich bereits in der Strombilanz (oder ganz woanders, z.B. in den Einstellungen), führt ein Klick auf das Logo zur "Startseite", also Erzeugung/Aktuell (so wie es bis zur v0.18 war)

Von der Seite "Hausverbrauch" aus führt also ein **zweimaliges** Klicken auf das Logo zur Startseite.

Zu sehen in `develop` ab `v0.19.0-79`.

closed 2025-03-30 by ledermann