solectrus/solectrus #4201: Beta-Test für Balkonkraftwerke: Historische Daten teilweise verschwunden

WolfImBusch, 2025-04-16:
### Beschreibung (auf Deutsch oder Englisch)

Hallo @ledermann 

Wie immer konnte ich nicht warten und habe die "pr-4195" installiert wie von Dir beschrieben (Szenario 2). Ich denke, ich habe das richtig gemacht - die Anzeige "aktuell" funktioniert und zeigt mir nun die PV Leistung getrennt für meine Ost- und Westseite an. 

Leider fehlen mir nun historische Daten - auch nach Zurücksetzen und Neuberechnung der Tageszusammemfassung .. :-( Diese sind
- Ansicht "Woche": Die Tage Mo und Di 
- Ansicht Monat: Alle Tage dieses Monats außer heute
- Ansicht Jahr: Alle bisherigen Monate Jan, Feb, März

Hilfe!! :-)

Gerne stelle ich auch meine compose/.env zur Verfügung - aber ich vermute, das wird nicht das Problem sein ..

[image]

[image]

Ich danke Dir und viele Grüße!!!

Alex

### Welche Version von SOLECTRUS verwendest du?

pr-4195

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Synology DS220+ (Docker9

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

ledermann, 2025-04-16:
Bitte `.env` und `compose.yaml` hier posten.

WolfImBusch, 2025-04-16:
Hi @ledermann 

Gerne! Ich habe es oben angehängt ...

Mir scheint jemand anderes scheint ein sehr ähnliches Problem zu haben .. (https://github.com/solectrus/solectrus/issues/4202).

Falls ich etwas tun kann (testen, weitere Infos) gib mir bitte Bescheid.

Ich danke Dir, drücke die Daumen und viele Grüße

ledermann, 2025-04-16:
Auch hier die Frage: Sind in `pv:power_east` und `pv:power_west` erst seit heute Messwerte drin oder schon immer?

WolfImBusch, 2025-04-16:
Diese habe ich erst heute eingefügt ...

ledermann, 2025-04-16:
Meine Vermutung ist, dass nur in genau den Zeiträumen die Balken für "Erzeugung" fehlen, in denen einige Einzelwerte da sind, aber nicht alle. Also:

- Aktuelle Woche: Montag und Dienstag haben keine Werte, Mittwoch aber schon => es fehlen die Balken für Mo und Di. Die Vorwoche müsste korrekt sein.
- Aktueller Monat: Der 16. hat Werte, alle davor aber nicht => es fehlen die Balken für 1-15. Die Vormonate müssten korrekt sein.
- Aktuelles Jahr: Der April hat Werte, die davor aber nicht => es fehlen die Balken für Jan/Feb/März. Frühere Jahre müssten korrekt sein.

Kannst du das bestätigen? Wenn das so ist, hätte ich einen Ansatzpunkt: Dann läuft etwas mit dem Fallback schief.

WolfImBusch, 2025-04-16:
Hallo @ledermann 

kann ich genau so zu 100% bestätigen. 

Die Daten sind ja "da" .. anbei ein Beispiel ...

Ansicht "Tag" von gestern, 15. April --> Alles super

[image]

Ansicht "Woche". Hier wird dann der 15. April nicht dargestellt

[image]

Das Gleiche gilt dann natürlich entsprechend für Ansicht "Monate", "Jahr" etc ...

Viele Grüße

Alex

WolfImBusch, 2025-04-16:
> Ich kann das Problem nun reproduzieren: Es fehlen genau dann in einem Diagramm einzelne Balken, wenn der gewählte Zeitraum sowohl Perioden **mit** Teilerzeugung als auch Perioden **ohne** Teilerzeugung enthält. Ich arbeite daran.
> 
> Verloren geht übrigens nichts, es ist "nur" ein Darstellungsproblem.

Super! Kein Problem - dafür sind Beta-Tests ja da ... 

Viel Erfolg bei der Fehlerbehebung!!!

WolfImBusch, 2025-04-16:
Hi @ledermann 

sorry - noch eine Anmerkung. "Speziell" könnte genau der (heutige) Tag sein, wo es aufgrund der Umstellung alle 3 Messwerte gibt. Aktuell scheinen für die Darstellung "Tag" nur die beiden neuen Messwerte verwendet zu werden. 

Bedeutet: laut Dashboard dabe ich 8.9kWh erzeugt - aber bereits 12.9 kWh eingespeist. 
Ist aber vermutlich nicht tragisch und ein Corner Case - sofern das Dashboard keine Plausibilität-Checks macht oder die Werte irgendwie für eine Berechnung herangezogen werden ... :-)

ledermann, 2025-04-16:
Habe jetzt den Fallback-Mechanismus verbessert, sodass jetzt keine Balken mehr fehlen werden. Zu sehen in `pr-4195` ab `multi-inverter-178`.

Eine ähnliche Sache gibt es noch in der Strombilanz - auch dort muss auf der linken Seite die Erzeugung anders berechnet werden, wenn die Zeitraum nicht vollständig durch Teile abgedeckt wird. Kommt noch...

WolfImBusch, 2025-04-16:
Hallo @ledermann 

Ich danke Dir, dass Du Dich der Sache so schnell angenommen hast. 

Ich habe mir natürlich gleich die neue Version gepulled. Dazu folgende Anmerkungen (s. Screenshot Anhang)
-- Anzeige Messwert "Erzeugung": Die Balken sind jetzt alle wieder da! Super!!! Die schicke Teilung des "Mittwoch-Balkens" (Vergleich s.o. mein letzter Screenshot), wo man sieht, wie viel in meinen Fall auf der Ost- und Westseite erzeugt worden ist,  habe ich noch sehen können, aber das hängt vielleicht mit Deinem letzen Kommentar (Bug in Strombilanz) zusammen
-- Auch bestätigen kann ich, dass die Anzeige auf der linken Seite noch nicht mit der rechten Seite übereinstimmt. 
Erzeugung linke Seite: 11.3 kWh. Balken rechts für heute. can 24 kWh
-- Was leider noch **nicht** funktioniert ist die Anzeige des Messwertes "Eigenverbrauch". Dort gibt es immer noch das gleiche Verhalten wie zuvor bei "Erzeugung". Wird das gleiche Problem sein.

Ich danke Dir und viele Grüße!!

[image]

ledermann, 2025-04-16:
Ja, ich denke, es handelt sich alles mehr oder weniger um das gleiche Problem: Wenn man einen Teil-Erzeuger nachträglich hinzufügt (während die Gesamterzeugung bereits vorhanden ist), kann das für den aktuellen Tag/Woche/Monat/Jahr nicht sinnvoll genutzt werden, weil die Gesamterzeugung in dieser Periode eben nicht die Summe der Teile ist.

Ein solcher Fall muss vernünftig abgefangen werden. Ich habe jetzt weiteres ergänzt (`multi-inverter-179`), werde aber noch genauer prüfen, ob das noch an anderen Stellen nötig ist.

WolfImBusch, 2025-04-16:
Hi @ledermann 

super! Habe mir die 179er gezogen - sieht schon viel besser aus .. allerdings .. :-)

[image]

Dazu bitte "Tag", "Woche", "Monat", "Jahr". auswählen und - am Beispiel vom "Tag" bitte auf das Kleiner-Zeichen tippen - also den Tag von gestern (versuchen) anzuschauen .,.

Anbei die Log Infos .. wenn ich irgendwie helfen kann, sag bitte Bescheid.

Viel Grüße! 
Alex

`solectrus-dashboard-1  | Information for cause: FloatDomainError (Infinity):
solectrus-dashboard-1  | [172.18.0.1]   
solectrus-dashboard-1  | [172.18.0.1] app/services/calculator/base.rb:68:in 'Float#round'
solectrus-dashboard-1  | [172.18.0.1] app/services/calculator/base.rb:68:in 'Calculator::Base#valid_multi_inverter?'
solectrus-dashboard-1  | [172.18.0.1] app/components/segment/component.rb:36:in 'Segment::Component#outer_multi_inverter?'
solectrus-dashboard-1  | [172.18.0.1] app/components/segment/component.html.slim:43:in 'Segment::Component#call'
solectrus-dashboard-1  | [172.18.0.1] app/components/segment_container/component.html.slim:5:in 'block in SegmentContainer::Component#call'
solectrus-dashboard-1  | [172.18.0.1] app/components/segment_container/component.html.slim:4:in 'Array#each'
solectrus-dashboard-1  | [172.18.0.1] app/components/segment_container/component.html.slim:4:in 'SegmentContainer::Component#call'
solectrus-dashboard-1  | [172.18.0.1] app/components/balance/component.html.slim:2:in 'Balance::Component#call'
solectrus-dashboard-1  | [172.18.0.1] app/components/stats_range/component.html.slim:1:in 'StatsRange::Component#call'
solectrus-dashboard-1  | [172.18.0.1] app/views/balance/stats/index.turbo_stream.slim:14
solectrus-dashboard-1  | [172.18.0.1] app/views/balance/stats/index.turbo_stream.slim:10
solectrus-dashboard-1  | [172.18.0.1] app/controllers/balance/stats_controller.rb:8:in 'Balance::StatsController#index'
solectrus-dashboard-1  | [172.18.0.1] app/controllers/concerns/auto_locale.rb:14:in 'ApplicationController#switch_locale'
solectrus-dashboard-1  | [172.18.0.1] method=GET path=/500 format=html controller=ErrorsController action=show status=500 allocations=10141 duration=38.59 view=34.38 db=0.00
`

StevieC121176, 2025-04-16:
mit der 179 bekomme ich auch bei Aktuell Erzeugung den Server Fehler 500. Zusätzlich fehlt die Unterteilung der Balken bei Tag, Monat usw.

WolfImBusch, 2025-04-16:
Hi @ledermann 

die 181´iger sieht gut as - der 500er Fehler ist weg. Morgen wird es spannend, wenn meine neuen Sensoren (Ost- und West-Seite) erstmalig komplett mit Werten befüllt sind. Und "Eigenverbrauch" (s.o.) wird sicher auch bald funktionieren ...

Ich hoffe, ich gehe Dir mit meinen Kommentaren/Rückmeldungen nicht auf die Nerven. Falls doch, sag bitte Bescheid - ich komme damit klar .. :-)

Viele Grüße (morgen ist auch noch ein Tag!)

Alex

ledermann, 2025-04-17:
Die Rückmeldungen sind sehr wichtig und hilfreich für mich, danke dafür! 

Nach der Korrektur des Fehler 500 (`multi-inverter-180`) habe ich jetzt noch mit `multi-inverter-186` das Problem mit dem Eigenverbrauch behoben.

WolfImBusch, 2025-04-17:
Danke Dir Georg!!

"Eigenverbrauch" wird jetzt auch richtig angezeigt!!!! Klasse!!!

Schwierigkeiten habe ich momentan noch mit dem Tool-Tipp auf einem mobilen Device. Manchmal wird hier nur die Gesamtleistung angezeigt - manchmal aber auch (in meinem Fall/Konfiguration) die Aufschlüsselung nach Ost und West. Vermutlich abhängig davon wie groß die Erzeugung und damit die Grafik ist. Ich kann mir das aktuell nicht ansehen und der Punkt hat auch in diesem Issue nichts verloren. Optik machst Du ja erst später. Aus meiner Sicht können wir das Issue hier schließen.

Danke Dir und viele Grüße!!

ledermann, 2025-04-17:
Sehr schön :) Ja, mit der Darstellung bin ich auch noch nicht glücklich, da hakt es noch an verschiedenen Stellen. Das kümmere ich mich auch noch drum. Wichtig ist mir zunächst aber, dass die Zahlen alle korrekt sind, keine Balken fehlen und die Aufteilung der Erzeugung auch so funktioniert, wie es konfiguriert ist.

Insofern, ja, ich schließe das Issue. Alles Weitere dann in separaten Issues.

closed 2025-04-17 by ledermann