solectrus/solectrus #4202: Beta-Test für Balkonkraftwerke und andere Erzeuger

StevieC121176, 2025-04-16:
### Beschreibung (auf Deutsch oder Englisch)

Ich habe mich für Variante 2 erstmal entschieden. Nach der Installation habe ich einige Anzeigeprobleme (sh. screenshots) unter IOS 18.4
- desweiteren sind bei mir alle Teile der Erzeugung in der Ansicht gleich gross, obwohl sie alle 3 verschieden sind
- Anzeigereihenfolge : in der Grafik wird es genau umgekehrt zum Tooltip angezeigt. Heisst in der Grafik ist InverterPower3 oben und 1 unten. Tooltip dementsprechend umgekehrt.
- fehlende historische Daten. Bei mir fehlen leider alle vergangenen Erzeugerdaten (SensorWert für Gesamterzeugung ist nicht geändert worden)

env und yaml kann ich gern zur Verfügung stellen (obwohl die etwas verwirrend bei mir ist, da noch eine sehr alte Installation)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Welche Version von SOLECTRUS verwendest du?

pr-4195

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspberry 4 8gb

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
services:
  app:
#    image: ghcr.io/solectrus/solectrus:latest
    image: ghcr.io/solectrus/solectrus:pr-4195
#    image: ghcr.io/solectrus/solectrus:pr-4151
#    image: ghcr.io/solectrus/solectrus:next
    labels:
      - 'com.centurylinklabs.watchtower.scope=solectrus'
    depends_on:
      db:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - db
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - DB_HOST=db
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
[... 470 lines cut ...]
MAPPING_20_TOPIC=Solectrus/Kueche
MAPPING_20_FIELD=power
MAPPING_20_MEASUREMENT=Kueche
MAPPING_20_TYPE=integer
MAPPING_21_TOPIC=Solectrus/mqtt/1/evcc/site/pv/3/power
MAPPING_21_FIELD=inverter_power
MAPPING_21_MEASUREMENT=balcony
MAPPING_21_TYPE=integer
MAPPING_22_TOPIC=evcc/site/pv/2/power
MAPPING_22_FIELD=pvhaupt
MAPPING_22_MEASUREMENT=my_pv_measurement
MAPPING_22_TYPE=integer
MAPPING_23_TOPIC=evcc/site/pv/1/power
MAPPING_23_FIELD=pvneben
MAPPING_23_MEASUREMENT=my_pv_measurement
MAPPING_23_TYPE=integer
MAPPING_24_TOPIC=evcc/site/pv/3/power
MAPPING_24_FIELD=pvgarten
MAPPING_24_MEASUREMENT=my_pv_measurement
MAPPING_24_TYPE=integer
#MAPPING_25_TOPIC=evcc/site/pv/3/power
#MAPPING_25_FIELD=pvgarten
#MAPPING_25_MEASUREMENT=my_pv_measurement
#MAPPING_25_TYPE=integer
```

ledermann, 2025-04-17:
Nach Behebung verschiedener Probleme sollten mit `multi-inverter-186` keine Balken mehr fehlen. Kannst du das nochmal prüfen?

StevieC121176, 2025-04-17:
Ja mit der 186 werden die Balken korrekt dargestellt. Auch die Balkenaufteilung funktioniert. Wird ab dem heutigen Tag dargestellt. Gestern fehlt die, aber liegt vermutlich an der vorigen Version oder die Daten erst ab Mitte des Tages gesammelt wurden.

ledermann, 2025-04-17:
Prima! Dass keine Aufteilung für die vergangenen Tage erfolgt, liegt an den unvollständigen oder fehlenden Messwerten bei diesen Tagen. Bei Montag/Dienstag fehlen sie ganz, da kann logischerweise nichts aufgeteilt werden. Bei Mittwoch wurden diese Messwerte erst im Laufe des Tages ergänzt, daher ist die Summe der Teilerzeuger nicht identisch mit der separat gemessenen Gesamterzeugung - also wäre eine Aufteilung falsch und wird nicht dargestellt.

Es zeigt sich aber, dass der Umgang mit fehlenden Messwerten noch nicht optimal ist. Insbesondere in der Anfangszeit, nachdem man die Sensoren eingerichtet hat. Da überlege ich mir noch etwas zu.

Auch die Darstellung in der Strombilanz (Segment "Erzeugung") gefällt mir noch nicht, speziell bei AKTUELL, wenn die Werte schwanken. Das werde ich noch verbessern.

Das Issue schließe ich erst mal, da das Problem der fehlenden Balken beseitigt ist.

closed 2025-04-17 by ledermann