solectrus/solectrus #4203: Beta-Test multi inverter

Zeppelin-PV, 2025-04-16:
### Beschreibung (auf Deutsch oder Englisch)

hallo Georg
zunächst mal "RESPEKT" für diese tolle Arbeit
Ich vermisse meine Erzeugung im Zeitraum der Inbetriebnahme 1.9.22 bis März 2023 wo Solectrus in Betrieb ging.
Dieser Zeitraum ist gefüllt mit csv Daten von SENEC, diese sind nun weg.
Idealerweise wäre eine Darstellung für diesen Zeitraum eine Gesamterzeugung, wie bisher.
Das BKW ist seit 2024 in Betrieb, somit erfolgt in Gesamt keine Anzeige links in Erzeugung
Generell wünschte ich mir, dass beim Definieren der Sensoren Groß- Kleinschreibung dann im Dashboard übernommen wird - machbar?
Wäre es außerdem auch noch möglich in den einzelnen Inverter-Feldern einen Platzhalter zu schaffen um in jedem Feld den aktuellen Wert anzuzeigen ?
Solectrus fehlt nun etwas die Dynamik, da sich in der Erzeugung nichts mehr bewegt :)
Gruß Ernst
[image]

### Welche Version von SOLECTRUS verwendest du?

multi-inverter-175-g44002ce7

### Auf welchem Gerät läuft bei dir SOLECTRUS?

RP5

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml

```

### Dein vollständiges `.env`

```text

```

Zeppelin-PV, 2025-04-16:
wenn man die Inbetriebnahme von Solectrus berücksichtigen würde, könnte man vermutlich den Zeitraum der csv Daten beibehalten

ledermann, 2025-04-16:
Welches Szenario hast du gewählt und wie sind die Variablen belegt?

Zeppelin-PV, 2025-04-16:
komme gerade zurück, nun sieht es so aus
sofort docker pull und up -d gemacht

in der latest version 19.0 ist die Verbindung da

# INFLUX_SENSOR_INVERTER_POWER=SENEC:inverter_power
INFLUX_SENSOR_INVERTER_POWER_1=SENEC:mpp1_power
INFLUX_SENSOR_INVERTER_POWER_2=SENEC:mpp2_power
INFLUX_SENSOR_INVERTER_POWER_3=SENEC:mpp3_power
INFLUX_SENSOR_INVERTER_POWER_4=TERRASSE:power_c
INFLUX_SENSOR_INVERTER_POWER_5=

[image]

[image]

Zeppelin-PV, 2025-04-17:
heute morgen sieht es so aus
die Werte in Erzeugung sind wieder vollständig da
die Einzelwerte mpp1, mpp2, mpp3 sieht man ab IBN Solectrus
das BKW ab dessen Inbetriebnahme
hatte übrigens zunächst die Version 3 der Sensoren Config eingesetzt, aber seit gestern spät Abends auf Verion 2
heute morgen dann die 186 gezogen
... ich sehe aber keine aufgeteilte Erzeugung mehr - habe ich da was verpasst ?

[image]

[image]

ledermann, 2025-04-17:
Mit Update auf `multi-inverter-186` wurden verschiedene Probleme behoben, die zu fehlenden Balken führten.

Ob jetzt eine Aufteilung erfolgt oder nicht, hängt vom gewählten Zeitraum ab, also ob in dem Zeitraum vollständig die Messwerte vorhanden sind, die zu einem Aufsplitten erforderlich sind.

Um Probleme zu klären, bitte unbedingt angeben, wie genau denn jetzt deine Konfiguration aussieht, also die 6 Sensoren zu den Invertern.

Zeppelin-PV, 2025-04-17:
hallo Georg,
anbei meine aktuelle env und docker Dateien
mein Erscheinungsbild momentan ist quasi so wie vor Einführung multi inverter - allerdings im Diagramm sind die einzelnen Inverter anwählbar,
eine gestaffelte Dartstellung in den Balkendiagrammen ist auch nicht mehr sichtbar

[docker-compose.txt](https://github.com/user-attachments/files/19790164/docker-compose.txt)
[env.txt](https://github.com/user-attachments/files/19790163/env.txt)

[image]

[image]

ledermann, 2025-04-17:
Du verwendest also diese Einstellung:

```env
INFLUX_SENSOR_INVERTER_POWER=SENEC:inverter_power
INFLUX_SENSOR_INVERTER_POWER_1=SENEC:mpp1_power
INFLUX_SENSOR_INVERTER_POWER_2=SENEC:mpp2_power
INFLUX_SENSOR_INVERTER_POWER_3=SENEC:mpp3_power
INFLUX_SENSOR_INVERTER_POWER_4=TERRASSE:power_c
INFLUX_SENSOR_INVERTER_POWER_5=
```

(Das sind die Zeilen, die ich bei allen Fragen rund um dieses Thema immer sehen will)

Diese Konfiguration ist **nicht richtig**, denn `SENEC:inverter` enthält bei dir nicht die Gesamterzeugung, sondern nur die Erzeugung vom Dach. SOLECTRUS merkt, dass die Summe nicht stimmt und stellt daher nirgendwo eine Aufteilung dar.

Wenn du erst kürzlich mit SOLECTRUS angefangen hättest, würde ich dir folgende Konfiguration vorschlagen:

```env
INFLUX_SENSOR_INVERTER_POWER=
INFLUX_SENSOR_INVERTER_POWER_1=SENEC:mpp1_power
INFLUX_SENSOR_INVERTER_POWER_2=SENEC:mpp2_power
INFLUX_SENSOR_INVERTER_POWER_3=SENEC:mpp3_power
INFLUX_SENSOR_INVERTER_POWER_4=TERRASSE:power_c
INFLUX_SENSOR_INVERTER_POWER_5=
```

Dann würde sich die Gesamterzeugung rechnerisch als Summe aus den 3 Strings sowie dem BKW ergeben. Die vom SENEC gemeldete Dach-Erzeugung (in `SENEC:inverter_power`) würde gar nicht verwendet werden.

Nun bist du aber schon sehr lange dabei, der SENEC-Collector schreibt aber erst seit [v0.7.0](https://github.com/solectrus/senec-collector/releases/tag/v0.7.0) vom 23.03.2023 die Messwerte für MPP1/2/3 nach InfluxDB. Das heißt, für frühere Zeiträume gibt es diese Aufteilung nicht in deiner InfluxDB. Somit würde diese Konfiguration für Zeiträume VOR März 2023 nicht funktionieren. Dadurch wären nicht nur die ersten Jahre, sondern auch die Ansicht "GESAMT" auf Dauer falsch.

Es bleibt daher in deinem Fall nur diese Konfiguration, also ein Verzicht auf die Aufteilung nach Strings:

```env
INFLUX_SENSOR_INVERTER_POWER=
INFLUX_SENSOR_INVERTER_POWER_1=SENEC:inverter_power
INFLUX_SENSOR_INVERTER_POWER_2=TERRASSE:power_c
INFLUX_SENSOR_INVERTER_POWER_3=
INFLUX_SENSOR_INVERTER_POWER_4=
INFLUX_SENSOR_INVERTER_POWER_5=
```

Um dennoch eine Aufteilung nach Strings in SOLECTRUS zu erreichen, könntest du theoretisch die Datenbank mit fiktiven Werten auffüllen (also mpp1/2/3 für die Zeiten vor März 2023 "schätzen"). Da muss man dann aber ganz genau wissen, was man tut.

ledermann, 2025-04-17:
Die Darstellung des Segments "Erzeugung" habe ich jetzt verbessert:

- Die Höhe der einzelnen Teilsegmente entsprechen jetzt ihrem relativen Anteil an der Erzeugung
- Bei wenig Platz vereinfacht sich die Darstellung (Wegfall der Beschriftung, Wegfall der ganzen Aufteilung)
- In AKTUELL erfolgt jetzt eine sanftere Überblendung, wenn sich die Erzeugung verändert.

Zu sehen in `multi-inverter-187`. Ich schließe nun das Issue. Bitte bei Bedarf neue anlegen - jeweils mit nur **einem** Thema.

closed 2025-04-17 by ledermann