solectrus/solectrus #4208: Fehler 500 nach Wechsel auf Multi-Inverter Beta

unclej84, 2025-04-17:
### Beschreibung

Nach dem laden des Dashboards erhalte ich Fehler 500
Die Anpassungen habe ich gemäß Discussion durchgeführt (Image, neue Variablen)

### Welche Version von SOLECTRUS verwendest du?

multi-inverter-190-gf60e276e

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Docker

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
SOLECTRUS Photovoltaic Dashboard
Copyright (C) 2020-2025 Georg Ledermann
License: GNU AGPLv3 - https://www.gnu.org/licenses/agpl-3.0.html
Version v0.19.0-190-gf60e276e (multi-inverter), built on 2025-04-17T12:46:54Z
Using ruby 3.4.3 (2025-04-14 revision d0b7e5b6a0) +PRISM [x86_64-linux-musl]
Based on Alpine Linux 3.21.3

## Waiting for services...
Redis is up and running!
redis (172.19.0.5:6379) open
influxdb (172.19.0.16:8086) open
InfluxDB is up and running!
postgres (172.19.0.3:5432) open
PostgreSQL is up and running!

## Preparing database...
Migrating to AddMultipleInvertPowerToSummaryValues (20250407104842)
== 20250407104842 AddMultipleInvertPowerToSummaryValues: migrating ============
-- execute("ALTER TYPE field_enum RENAME VALUE 'balcony_inverter_power' TO 'inverter_power_1';")
   -> 0.0337s
-- add_enum_value(:field_enum, :inverter_power_2, {if_not_exists: true})
   -> 0.0359s
-- add_enum_value(:field_enum, :inverter_power_3, {if_not_exists: true})
   -> 0.0318s
-- add_enum_value(:field_enum, :inverter_power_4, {if_not_exists: true})
   -> 0.0738s
-- add_enum_value(:field_enum, :inverter_power_5, {if_not_exists: true})
   -> 0.0296s
== 20250407104842 AddMultipleInvertPowerToSummaryValues: migrated (0.2066s) ===

Database is ready!

## Starting Rails application...
=> Booting Puma
=> Rails 8.0.2 application starting in production
=> Run `bin/rails server --help` for more startup options
Sensor initialization started
  DEPRECATION WARNING: Missing environment variable INFLUX_SENSOR_INVERTER_POWER.
  To remove this warning, add the following to your environment:
    INFLUX_SENSOR_INVERTER_POWER=SENEC:inverter_power
  or, when you want to ignore this sensor:
    INFLUX_SENSOR_INVERTER_POWER=

  - Sensor inverter_power                 → SENEC:inverter_power
  - Sensor house_power                    → Calculated:house_power
  - Sensor heatpump_power                 → sdm72d-m:power
  - Sensor grid_import_power              → powermeter:power_in
  - Sensor grid_export_power              → powermeter:power_out
  - Sensor battery_charging_power         → batteries:charging_power
  - Sensor battery_discharging_power      → batteries:discharging_power
  - Sensor wallbox_power                  → SENEC:wallbox_charge_power
  - Sensor inverter_power_1               → SENEC:inverter_power
  - Sensor inverter_power_2               → balcony:power
  - Sensor custom_power_01                → heatpump:power
[... 335 lines cut ...]
[192.168.30.20] app/components/segment/component.rb:30:in 'Segment::Component#outer_link'
[192.168.30.20] app/components/segment/component.html.slim:41:in 'Segment::Component#call'
[192.168.30.20] app/components/segment_container/component.html.slim:5:in 'block in SegmentContainer::Component#call'
[192.168.30.20] app/components/segment_container/component.html.slim:4:in 'Array#each'
[192.168.30.20] app/components/segment_container/component.html.slim:4:in 'SegmentContainer::Component#call'
[192.168.30.20] app/components/balance/component.html.slim:2:in 'Balance::Component#call'
[192.168.30.20] app/components/stats_now/component.html.slim:1:in 'StatsNow::Component#call'
[192.168.30.20] app/views/balance/stats/index.turbo_stream.slim:12
[192.168.30.20] app/views/balance/stats/index.turbo_stream.slim:10
[192.168.30.20] app/controllers/balance/stats_controller.rb:8:in 'Balance::StatsController#index'
[192.168.30.20] app/controllers/concerns/auto_locale.rb:14:in 'ApplicationController#switch_locale'
[192.168.30.20] method=GET path=/500 format=html controller=ErrorsController action=show status=500 allocations=10238 duration=179.12 view=154.19 db=0.00
```

### Deine vollständige `docker-compose.yml` oder `compose.yml`

```yaml

```

### Deine vollständige `.env`

```text

```

unclej84, 2025-04-17:
Hat sich erledigt, aber sollte vielleicht erwähnt oder abgefangen werden:

Ich habe folgende Einträge nicht gehabt - nachdem sie ergänzt wurden in der `.env` hat es funktioniert:

`
INFLUX_SENSOR_INVERTER_POWER=
INFLUX_SENSOR_INVERTER_POWER_3=
INFLUX_SENSOR_INVERTER_POWER_4=
INFLUX_SENSOR_INVERTER_POWER_5=
`

ledermann, 2025-04-18:
Ursache war hier die fehlende Variable `INFLUX_SENSOR_INVERTER_POWER`, was zu einem [Fallback](https://docs.solectrus.de/wartung/sensor-konfiguration.html#kompatibilit%C3%A4t-mit-%C3%A4lteren-versionen) zu `SENEC:inverter_power` führte (der bei dir aber keinen Inhalt hat).

Ich habe in `multi-inverter-195` eine zusätzliche Prüfung ergänzt, sodass es nicht mehr zu einem Fehler 500 führt.

closed 2025-04-18 by ledermann