solectrus/solectrus #4214: Anzeigereihenfolge Erzeuger

StevieC121176, 2025-04-18:
### Beschreibung

Die Anzeigereihenfolge der Erzeuger ist nicht überall gleich.
Angelegt habe ich:

Inverter_power 1 Nebengebäude
Inverter_power 2 Haupthaus
Inverter_power 3 Garten
angezeigt wird das auf der Erzeugerseite aber in umgekehrter Reihenfolge, in der Balkenanzeige wiederum in richtiger Reihenfolge.

Besser wäre hier sicher alles in der gleichen Reihenfolge entsprechend den angelegten Erzeugern anzuzeigen.

[image]
[image]
[image]

### Welche Version von SOLECTRUS verwendest du?

multi-inverter-190-gf60e276e

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspberry 4 + 5

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Deine vollständige `docker-compose.yml` oder `compose.yml`

```yaml

```

### Deine vollständige `.env`

```text

```

ledermann, 2025-04-18:
Ja, die Reihenfolge ist derzeitig nicht konsistent. Ich hab es so vor:

- Die Reihenfolge der Sensoren sollte in aufsteigender Wichtigkeit vorgenommen werden. Also `inverter_power_1` sollte der stärkste Erzeuger sein.
- Überall (Strombilanz, Tooltip, Balken) wird dann von unten nach oben sortiert, also der stärkste steht unten, die kleineren obendrauf.

ledermann, 2025-04-18:
Implementiert in `multi-inverter-195`

closed 2025-04-18 by ledermann

StevieC121176, 2025-04-18:
> Implementiert in `multi-inverter-195`

passt alles außer Balken, der ist genau verkehrt herum. (schwächste Erzeuger unten)

[image]

ledermann, 2025-04-18:
Es wird nicht nach der tatsächlichen Zahl sortiert, sondern von `inverter_power_1` (unten) nach `inverter_power_5` (oben). Das heißt, man sollte seine Konfiguration so anlegen, dass der stärkste Erzeuger die 1 bekommt usw.

Der Grund ist, dass dadurch die Position immer fix ist. Wenn nach dem Wert sortiert würde, dann könnte sich die Position ändern, was vermutlich irritierend wäre.

StevieC121176, 2025-04-18:
> Es wird nicht nach der tatsächlichen Zahl sortiert, sondern von `inverter_power_1` (unten) nach `inverter_power_5` (oben). Das heißt, man sollte seine Konfiguration so anlegen, dass der stärkste Erzeuger die 1 bekommt usw.
> 

Genau das irritiert mich etwas, da ich das genauso wie du schreibst angelegt habe. Also inverser_power_1 ist bei mir Nebengebäude und das ist der stärkste Erzeuger und müsste demnach unten angezeigt werden im Balken. Unten im Balken wird bei mir aber der schwächste Erzeuger (Garten) angezeigt. Das war in der Version davor noch anders. Da wurde unten im Balken der stärkste Erzeuger angezeigt.

Screenshot von Version 191:

[image]

Screenshot von Version 195:

[image]

StevieC121176, 2025-04-18:
Mit der 196 schaut das gut aus und passt jetzt mit den Balken. Auch die Farbenanpassung gefällt mir.

ledermann, 2025-04-18:
Ok, jetzt verstehe ich. Die Sortierung im Tooltip ist korrekt, aber die Balken selbst sind genau andersherum gestapelt. Ist nun korrigiert, siehe `multi-inverter-196`.
Außerdem habe ich noch eine kleine Farbabstufung der Balken ergänzt.