solectrus/solectrus #4230: kleinere Anzeigefehler in Multi-inverter-223-gae7a4037

StevieC121176, 2025-04-22:
leider funktioniert das Anlegen eines Issue bei mir gerade nicht, daher mal hier:

- Icons werden teilweise überdimensioniert angezeigt
- zu lange Nahmen werden teilweise nicht gekürzt und somit abgeschnitten angezeigt (bei mir Nebengebäude)
- Namen werden teilweise gekürzt (Ansicht Woche) oder aber komplett angezeigt (Ansicht Aktuell) - in meinem Fall "Haupthaus)

[image]
[image]
[image]

_Originally posted by @StevieC121176 in https://github.com/solectrus/solectrus/discussions/4229_

ledermann, 2025-04-22:
Das Abkürzen des Textes habe ich nun verbessert (in `multi-inverter-224`) - indem grundsätzlich keine zweite Zeile aufgemacht wird (was nur klappen würde, wenn der Text aus mehr als einem Wort bestehen würde), sondern bei Überlänge immer mit "..." abgekürzt wird. Meine Empfehlung ist, die Sensoren so kurz wie möglich zu benennen.

Ob ein Text angezeigt wird, hängt übrigens von der Höhe des Segments ab - hat also nichts mit Woche vs. Aktuell zu tun.

Das übergroße Icon konnte ich aber bislang nicht reproduzieren. Ist es bei dir tatsächlich so, dass es bei einem bestimmtem Zeitraum **dauerhaft** erscheint? Also jederzeit reproduzierbar? Oder ist es nur ein kurzes Aufblitzen? Welcher Browser? In KW17 ist bei dir der Wallbox-Anteil so klein, dass gar keine Beschriftung (und somit kein Icon) erscheinen sollte. Ich frage mich, wo das Icon herkommt. Bist du in der Lage, das mit den Browser-DevTools zu debuggen? Bitte nur versuchen, wenn das sowieso dein tägliches Business ist ;-)

StevieC121176, 2025-04-22:
Das übergroße Icon ist weg nachdem ich auf 224 aktualisiert habe, allerdings ist jetzt der Anteil vom E-Auto auch größer und somit wird in dem Bereich das Icon angezeigt (normal). Das es dauerhaft da war hatte ich bisher auch erst 1x, ansonsten war es wie von dir beschrieben immer nur so ein aufblitzen und war dann weg. 
Texte werden jetzt wie von dir beschrieben dargestellt.
Browser war Safari (iOS 18.4).

ledermann, 2025-04-22:
Ich habe jetzt eine Änderung vorgenommen (`multi-inverter-226`), die möglicherweise das Problem behebt. Wenn es dennoch erneut auftreten sollte, schaue ich mir die Sache nochmal genauer an.

ledermann, 2025-04-23:
Das Problem ist noch **nicht** behoben. Auf dem Desktop mit Safari 18.4 tritt es manchmal weiterhin auf, wenn ein Segment so klein ist, dass der Inhalt (Schrift + Icon) unsichtbar ist:

[image]

(Hier im Bild ein Teil des Flügels vom Icon der Wärmepumpe)

StevieC121176, 2025-04-23:
Ich habe heute mal die iOS 18.5 installiert, da tritt das auch noch auf, allerdings nur kurz und dann verschwindet es.

closed 2025-04-23 by ledermann