solectrus/solectrus #4238: FR: Umwandlung in GROSSBUCHSTABEN abwählbar machen.

gereons, 2025-04-23:
### Beschreibung (auf Deutsch oder Englisch)

Mit den neuen Multi-Invertern kommen wieder neue Text-Elemente zum UI hinzu, die automatisch in GROSSBUCHSTABEN angezeigt werden, und das sind IMHO jetzt schon viel zu viele.

Ich wünsche mir eine Einstellung, die die entsprechenden `text-transform: uppercase` im CSS entfernt oder durch `text-transform: none` ersetzt.

Zeppelin-PV, 2025-04-23:
... ja, das hatte ich in einem meiner vielen Beiträge auch mal erwähnt
So wie man in den Einstellungen die Sensoren beschreibt, so sollten sie im UI erscheinen

ledermann, 2025-04-23:
Ja, ein eigenes Issue sorgt dafür, dass auch nix verloren geht, danke :-)

Zum Inhalt: Ein Webdesigner für SOLECTRUS mit dem richtigen Blick auf das UI/UX wäre schön, gibt es aber nicht, daher muss ich als Aushilfs-Designer diesen Job übernehmen. Ich mache mir aber reichlich Gedanken dazu, die Oberfläche ansehnlich und funktional zu machen. Und ich neige zu [All Caps im UX Design](https://thestory.is/de/journal/all-caps-im-ux-design/), zumindest bei Navigationselementen, die natürlich möglichst kurz beschriftet sein sollten. 

Durch die Möglichkeit, die Sensoren selbst beschriften zu können, treten nun Probleme auf, die es vorher nicht gab. Die Lösung über ein Setting ("UpperCase ja/nein") gefällt mir aber nicht, das wäre zu kleinteilig.

Ich habe jetzt mal probeweise die Beschriftung in den Segmenten auf NormalCase umgestellt. Ob das so ideal ist und jedem gefällt, bleibt abzuwarten. Zu sehen in `multi-inverter-77`.

gereons, 2025-04-23:
Ist auf jeden Fall schon deutlich gefälliger, weil weniger "gebrüllt", danke!

Für meinen Geschmack könnte man in sämtlichen Labels in der "Haupt-Box" auf die Normalschreibweise wechseln, aber wenn's so bleibt wie jetzt ist das auch OK.