solectrus/solectrus #4239: Autakie liefert in den Ansichten "Aktuell" und "Tag" keinen Wert, wenn die Daten aus unterschiedlichen Measurements stammen

Oessi87, 2025-04-23:
### Beschreibung

Hallo,
durch eine Erweiterung der Anlage um einen Speicher mit Victron-Wechselrichtern muss ich den Hausverbrauch nun selbst berechnen. Dieser wird in der DB unter einem anderen Measurement gespeichert.
Dadurch funktioniert die pivot-Funktion in der Abfrage für die Autakie nicht mehr richtig.
Bei mir habe ich es aktuell dadurch gelöst, dass die folgende Zeile in die **app/services/autarky_chart.rb** hinzugefügt wurde:
[47]    q << "|> #{range(start: start - 1.hour, stop:)}"
[48]
[49]    q << "|> #{filter}"

 ##diese Zeile hinzufügen, damit alle Felder aus dem gleichen Measurement kommen
**[50]    q << '|> set(key: "_measurement", value: "temp")'**   

[51]    q << '|> aggregateWindow(every: 5s, fn: last)'
[52]    q << '|> fill(usePrevious: true)'

Ich würde mich freuen, wenn diese Änderung in der nächsten Version übernommen wird.

### Welche Version von SOLECTRUS verwendest du?

v0.19.1

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Ubuntu 24.04 V-Server

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Deine vollständige `docker-compose.yml` oder `compose.yml`

```yaml

```

### Deine vollständige `.env`

```text
# Sensor mapping
INFLUX_SENSOR_INVERTER_POWER=SENEC:inverter_power
INFLUX_SENSOR_INVERTER_POWER_FORECAST=Forecast:watt
####INFLUX_SENSOR_HOUSE_POWER=SENEC:house_power
INFLUX_SENSOR_HOUSE_POWER=Victron:power_house

INFLUX_SENSOR_GRID_IMPORT_POWER=SENEC:grid_power_plus
INFLUX_SENSOR_GRID_EXPORT_POWER=SENEC:grid_power_minus
INFLUX_SENSOR_GRID_EXPORT_LIMIT=SENEC:power_ratio
####INFLUX_SENSOR_BATTERY_CHARGING_POWER=SENEC:bat_power_plus
####INFLUX_SENSOR_BATTERY_DISCHARGING_POWER=SENEC:bat_power_minus

INFLUX_SENSOR_BATTERY_CHARGING_POWER=Victron:combined_battery_charging_power
INFLUX_SENSOR_BATTERY_DISCHARGING_POWER=Victron:combined_battery_discharging_power

INFLUX_SENSOR_BATTERY_SOC=SENEC:bat_fuel_charge
INFLUX_SENSOR_CASE_TEMP=SENEC:case_temp
INFLUX_SENSOR_SYSTEM_STATUS=SENEC:current_state
INFLUX_SENSOR_SYSTEM_STATUS_OK=SENEC:current_state_ok
INFLUX_SENSOR_WALLBOX_POWER=KLIMA:power_auto
INFLUX_SENSOR_HEATPUMP_POWER=KLIMA:power_klima
INFLUX_SENSOR_CAR_BATTERY_SOC=Victron:pytes_soc

INFLUX_EXCLUDE_FROM_HOUSE_POWER=WALLBOX_POWER,HEATPUMP_POWER
```

ledermann, 2025-04-23:
Besten Dank dafür! Das ist tatsächlich ein Fehler, denn natürlich soll es auch funktionieren, wenn die Messwerte in unterschiedlichen Measurements liegen. 

Dein Lösungsvorschlag hat einen Performance-Nachteil, daher habe ich das etwas anders gemacht, zu sehen in `v0.19.1-4`. Wenn du das ausprobieren magst, kannst du auf das `develop`-Image wechseln.

Bezüglich "Hausverbrauch selbst berechnen" &dash; vielleicht ist das für dich interessant:
https://github.com/orgs/solectrus/discussions/4138

closed 2025-04-23 by ledermann

Oessi87, 2025-04-23:
Danke.

Habe das develop-Image getestet und es funktioniert.

Das ingest-Projekt brauche ich nicht, da der Hausverbrauch als Zwischenwert für die Wechselrichtersteuerung anfällt und bei mir auch alle 5 Sekunden (mit etlichen anderen Werten) in die InfluxDB geschrieben wird. Aber Danke für den Hinweis.