solectrus/solectrus #4251: Farbdarstellung Multi-inverter-93 und verbunden Status

StevieC121176, 2025-04-25:
### Beschreibung

2 Kleinigkeiten die mir mit der v93 aufgefallen sind. Die Farbdarstellung konnte ich bisher nur mit der mobilen Version testen.

- Neue Farbdarstellung ist bei mir deutlich schlechter als vorher zu erkennen bei den gestapelten Diagrammen
- er wechselt in der Erzeugeransicht aktuell alle 30 Sekunden etwa vom verbunden in den nicht verbunden Status. In der Ansicht Strombilanz aktuell wird es richtig angezeigt, immer verbunden. Werte werden auch ohne Lücken geschrieben und angezeigt.

[image]
[image]

### Welche Version von SOLECTRUS verwendest du?

V93

### Auf welchem Gerät läuft bei dir SOLECTRUS?

pi 4+5

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Deine vollständige `docker-compose.yml` oder `compose.yml`

```yaml

```

### Deine vollständige `.env`

```text

```

ledermann, 2025-04-25:
Zu den Farben: Ja, sehe ich ein. Farbschattierungen eignen sich nicht so gut zur klaren Unterscheidung. Fünf stark unterschiedliche Farben zu definieren, würde aber zu einer recht bunten Darstellung führen. Ein Selbstdefinieren von Farben möchte ich auch eher vermeiden. Ich habe jetzt mal eine Trennlinie ergänzt. Hilft das?

Zum "Nicht verbunden": Kann es seine, dass deine Erzeuger-Werte seltener hereinkommen als die Werte auf der Hauptseite? Ich habe jetzt noch etwas verbessert (durch Hinzunahme des Sensors `system_status`, auch wenn der auf der Seite gar nicht verwendet wird), was möglicherweise hilft. 

Beides zu sehen in `multi-inverter-95`

StevieC121176, 2025-04-25:
Mit der  Version 95 bekomme ich einen internen Server Fehler 500 auf der `Erzeuger aktuell` Seite. Alle anderen funktionieren.  
Es kann sein das sich die Werte bei den einzelnen Erzeugern so alle 15 Sekunden ändern. Bei der Gesamterzeugung meine ich so alle 3 Sekunden.

Zu den Farben, man kann es schon besser erkennen aber am besten war es wie vorher mit den verlaufsfarben. Das passt auch irgendwie besser zu den restlichen Diagrammen.

[image]

ledermann, 2025-04-26:
Der Fehler 500 ist behoben (`multi-inverter-96`), das war ein anderes Problem. Bei den Farben sehe ich das Problem, da überlege ich mir noch etwas.

StevieC121176, 2025-04-26:
Ja, Fehler 500 tritt mit der 96 nicht mehr auf.
Noch eine Frage, wäre es auch möglich in der Erzeuger aktuell Sicht ein gestapeltes Diagramm anzuzeigen? Aktuell wird das gleiche angezeigt wie in der Strombilanz aktuell. Bei Tag, Woche, monat, Jahr gibt es ja das gestapelte Diagramm in der Erzeugeransicht

ledermann, 2025-05-04:
Die Farben sind mittlerweile etwas bessert, aber die Darstellung der gestapelten Flächen ist weiterhin nicht optimal. Mangels weiterer Ideen schließe ich das Issue.

closed 2025-05-04 by ledermann