solectrus/solectrus #4254: Klammer bei Hausverbrauch und Erzeuger

StevieC121176, 2025-04-26:
### Beschreibung

Eine absolute Kleinigkeit, die Klammer in den oben genannten Sichten müsste nach oben hin etwas länger sein um den gesamten Bereich zu erfassen

Btw. Ich kann den issue Type nicht ändern, bei mir wird das immer als bug deklariert.

[image]

### Welche Version von SOLECTRUS verwendest du?

Alle

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Pi 4+5

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Deine vollständige `docker-compose.yml` oder `compose.yml`

```yaml

```

### Deine vollständige `.env`

```text

```

ledermann, 2025-04-27:
Bei mir ist der falsche Abstand eher unten zu erkennen, nicht oben. Aber wie auch immer, ich habe nun etwas korrigiert, das für eine verrutschte Darstellung verantwortlich war. Zu sehen in `multi-inverter-106`.

closed 2025-04-27 by ledermann

ledermann, 2025-04-27:
Hatte die kleineren Displays nicht bedacht, in `multi-inverter-107` ist es nun korrekt.

unclej84, 2025-04-27:
Vielleicht sollten die "Andere Verbraucher" auch einen Balken bekommen, weil es sonst so aussieht als würde es beim Hausverbrauch nicht so ganz passen:

[image]

ledermann, 2025-04-27:
@unclej84 Ist mir nicht ganz klar, was du meinst. Geht es darum, dass die Hintergrundfarbe von "Andere Verbraucher" nicht identisch mit der Rahmenfarbe sein sollte? 
Oder was meinst du mit "Balken"?

unclej84, 2025-04-27:
Ja, entweder einen Rand drum oder eine leicht andere Farbe (bei den Erzeugern hat der untere ja auch eine andere Farbe als der Hintergrund). Dachte erst die Klammer würde zu weit oben enden bis ich gemerkt hatte, dass da ja noch der Bereich "Andere Verbraucher" ist (der sich aber nicht vom Hintergrund abhebt)

ledermann, 2025-04-28:
@unclej84 Habe jetzt die Hintergrundfarbe des Segments "Andere Verbraucher" angepasst. Damit dürften nun keine Irritationen mehr auftreten. Zu sehen in `multi-inverter-112`.