solectrus/solectrus #4258: Erzeuger/Aktuell: Gestapeltes Diagramm

ledermann, 2025-04-27:
> Noch eine Frage, wäre es auch möglich in der `Erzeuger aktuell Sicht` ein gestapeltes Diagramm anzuzeigen? Aktuell wird das gleiche angezeigt wie in der `Strombilanz aktuell`. Bei `Tag, Woche, monat, Jahr` gibt es ja das gestapelte Diagramm in der Erzeugeransicht 

 _Originally posted by @StevieC121176 in [#4251](https://github.com/solectrus/solectrus/issues/4251#issuecomment-2831897113)_

ledermann, 2025-05-05:
Das hat etwas gedauert, weil ich die Logik, mit der ein Diagramm bei "AKTUELL" laufende fortgeschrieben wird, dafür umbauen musste. Nun können auch Diagramme mit mehreren Kurven fortgeführt werden. Daher ist jetzt bei Erzeuger/AKTUELL ein gestapeltes Diagramm zu sehen - ab `multi-inverter-163`.

closed 2025-05-05 by ledermann

StevieC121176, 2025-05-05:
Prima, das gefällt mir sehr gut. Funktioniert problemlos