solectrus/solectrus #4281: Gesamterzeugungsdiagramm wird nicht korrekt dargestellt

bennysweb, 2025-05-05:
### Beschreibung

Es sollte eigentlich mit Klick auf das Solectrus-Symbol die Standard-(Gesamt)Erzeugungs-Ansicht dargestellt werden. Der Graph stimmt, die Benennung aber nicht (Autarkie). Mobil oder über Desktop gleiches Phänomen. Die (Gesamt)Erzeugung kann auch nicht im DropDown-Menü ausgewählt werden.

[image]

### Welche Version von SOLECTRUS verwendest du?

multi-inverter-155

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Synology

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Deine vollständige `docker-compose.yml` oder `compose.yml`

```yaml

```

### Deine vollständige `.env`

```text

```

ledermann, 2025-05-05:
Das kann ich nicht reproduzieren. Tritt das wirklich immer auf? Von welcher Seite bist du gekommen? Der Klick auf das Logo führt kontext-abhhängig zu etwas anderen Zielen.

Den fehlenden Eintrag im Dropdown-Menü habe ich übrigens mit Version `multi-inverter-155` korrigiert. Hast du wirklich diese Version?

bennysweb, 2025-05-05:
Ja, das tritt immer auf. Du kannst gern selbst schauen... https://solar.familykoenig.de

bennysweb, 2025-05-05:
Dann kann dein Anschluss scheinbar kein IPv6? Habe einen Glasfaseranschluss, der kann nur von IPv6-Anschlüssen (z.B. Mobil) erreicht werden. Von meiner Arbeit aus (reiner IPv4-Anschluss) klappt der zugriff auch nicht.

Egal, der Fehler tritt immer auf im Energiedashboard, egal, von welcher Seite ich komme. Und es ist definitiv v-155 installiert.

ledermann, 2025-05-05:
Ja, von Mobil klappt es und da kann ich den Fehler auch sehen. Mal überlegen...

ledermann, 2025-05-05:
Ich kann es nun reproduzieren. Der Fehler tritt auf, wenn:

- `INFLUX_SENSOR_INVERTER_POWER` ist leer 

**und**

- Einstellungen/Sensoren/Erzeuger: Checkbox Summe ist deaktiviert

closed 2025-05-05 by ledermann