solectrus/solectrus #4286: Lücken im Hausverbrauch

67-Peter, 2025-05-06:
### Beschreibung

Guten Morgen,
seit März erfasse ich meine Daten mit SOLECTRUS.
Jetzt ist mir was aufgefallen:
An manchen Tagen fehlt bei mir der Hausverbrauch ( z.B.: Do 10.4.)
[image]

In der Tagesansicht werden die Daten protokoliert, also müssten sie doch am Ende des Tages integriert werden können.
[image]

Das ist mir nur beim Hausverbrauch aufgefallen!

vg. Peter 

### Welche Version von SOLECTRUS verwendest du?

0.19.01

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Lenovo 910q (Proxmox)

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Deine vollständige `docker-compose.yml` oder `compose.yml`

```yaml

```

### Deine vollständige `.env`

```text

```

ledermann, 2025-05-06:
Das ist tatsächlich etwas seltsam. Fehlt am 10.04.2025 wirklich nur der Hausverbrauch? Das heißt, ist Erzeugung, Einspeisung etc. vorhanden? Hast du vielleicht Messwerte nachträglich importiert oder anderweitig in InfluxDB verändert?

Vermutlich könnte es helfen, die Tageszusammenfassungen zurückzusetzen (Login als Admin, dann Einstellungen/Allgemein).

67-Peter, 2025-05-06:
In der Influx sieht auch alles ok aus. (10.4.)
[image]

Tageszusammenfassungen zurückzusetzen hat leider nichts gebracht!
War nicht nur am 10.4. sondern kommt öfters vor (Screenshot nach Tageszusammenfassungen zurückzusetzen):

[image]

Ich habe keine Daten nachträglich verändert.

67-Peter, 2025-05-06:
Ausser E-Auto (den lade ich ja nicht dauernd) habe ich keinen weiteren Lücken.

ledermann, 2025-05-06:
Ich würde das gerne näher untersuchen. Kannst du mir einen CSV-Export der Hausverbrauchs-Messwerte vom 10.04.2025 zukommen lassen? Ohne Aggregation, also das Ergebnis dieser Query:

```
from(bucket: "SENEC")
  |> range(start: 2024-04-10T00:00:00+02:00, stop: 2024-04-11T00:00:00+02:00)
  |> filter(fn: (r) => r["_measurement"] == "pv")
  |> filter(fn: (r) => r["_field"] == "house_power")
```

67-Peter, 2025-05-06:
Dein Query funktionierte , brachte aber keine Ergebnisse, ich vermute das liegt an dem UTC offset oder das ist das Problem ;-)
csv download bibt es wohl nicht mehr ohne https - ich wollte an dem Setup nichts ändern und habe die Daten via console gesammelt:

root@docker-33:~/solectrus# curl -s -X POST \
>   "http://192.168.178.33:8086/api/v2/query?org=solectrus" \
>   -H "Authorization: Token xxxxxxx" \
>   -H "Content-Type: application/vnd.flux" \
>   -H "Accept: application/csv" \
>   --data '
>     from(bucket: "solectrus")
>       |> range(start: 2025-04-10T00:00:00Z, stop: 2025-04-11T00:00:00Z)
>       |> filter(fn: (r) => r["_field"] == "house_power")
>   '  > house_power_2025-04-10.csv

[house_power_2025-04-10.csv](https://github.com/user-attachments/files/20055191/house_power_2025-04-10.csv)

ledermann, 2025-05-06:
Meine Query war in mehrfacher Hinsicht falsch: Bucket muss `solectrus` heißen und das Jahr `2025`. Vermutlich lag das daran. Habe das oben korrigiert. Sehe mir nun die Daten an.

67-Peter, 2025-05-06:
Das Jahr war es ... hab ich auch nicht gesehen, soll ich dir die Daten nochmal mit UTC +2:00 zusammenstellen?

ledermann, 2025-05-06:
Habe die CSV bei mir importiert und damit getestet. Die Balken sind da, ich kann das Problem somit noch **nicht** reproduzieren.

Der Sache muss ich näher auf den Grund gehen. Könntest du mir die vollständigen Messwerte übermitteln, also **alle** Fields und Measurements, vielleicht den ganzen April?

```
from(bucket: "solectrus")
|> range(start: 2025-04-01T00:00:00+02:00, stop: 2025-05-01T00:00:00+02:00)
```

(also ohne Filter - Zeitzone `+02:00` sollte kein Problem machen)

67-Peter, 2025-05-06:
Mit dem richtigen Jahr geht das auch :-) 

 curl -s -X POST   "http://192.168.178.33:8086/api/v2/query?org=solectrus"   -H "Authorization: Token xxxx"   -H "Content-Type: application/vnd.flux"   -H "Accept: application/csv"   --data '
    from(bucket: "solectrus")
      |> range(start: 2025-04-01T00:00:00+02:00, stop: 2025-05-01T00:00:00+02:00)
' > solectrus_67-Peter-2025-April.csv

[67-Peter-2025-April.zip](https://github.com/user-attachments/files/20055760/67-Peter-2025-April.zip)

ledermann, 2025-05-06:
Ok, habe es jetzt bei mir importiert. Ich gehe von folgender Konfiguration aus:

```env
INFLUX_SENSOR_INVERTER_POWER=pv:inverter_power
INFLUX_SENSOR_HOUSE_POWER=pv:house_power
INFLUX_SENSOR_GRID_IMPORT_POWER=pv:grid_import_power
INFLUX_SENSOR_GRID_EXPORT_POWER=pv:grid_export_power
INFLUX_SENSOR_BATTERY_CHARGING_POWER=pv:battery_charging_power
INFLUX_SENSOR_BATTERY_DISCHARGING_POWER=pv:battery_discharging_power
INFLUX_SENSOR_BATTERY_SOC=pv:battery_soc
INFLUX_SENSOR_WALLBOX_POWER=pv:wallbox_power
INFLUX_SENSOR_CASE_TEMP=pv:case_temp
INFLUX_SENSOR_SYSTEM_STATUS=
INFLUX_SENSOR_SYSTEM_STATUS_OK=
INFLUX_SENSOR_GRID_EXPORT_LIMIT=
INFLUX_SENSOR_INVERTER_POWER_FORECAST=forecast:watt
INFLUX_SENSOR_CAR_BATTERY_SOC=pv:vehicleSoc
INFLUX_EXCLUDE_FROM_HOUSE_POWER=WALLBOX_POWER
```

Das heißt, vom Hausverbrauch soll der Wallbox-Verbrauch **abgezogenen** werden.

Am 10. April findet sich in InfluxDB ein **Hausverbrauch** von 24,5 kWh. 

```
from(bucket: "solectrus")
  |> range(start: 2025-04-10T00:00:00+02:00, stop: 2025-04-11T00:00:00+02:00)
  |> filter(fn: (r) => r["_measurement"] == "pv")
  |> filter(fn: (r) => r["_field"] == "house_power")
  |> integral(unit: 1h)
```

Die **Wallbox** hat am selben Tag einen Verbrauch von 29,2 kWh.

```
from(bucket: "solectrus")
  |> range(start: 2025-04-10T00:00:00+02:00, stop: 2025-04-11T00:00:00+02:00)
  |> filter(fn: (r) => r["_measurement"] == "pv")
  |> filter(fn: (r) => r["_field"] == "wallbox_power")
  |> integral(unit: 1h)
```

Wallbox-Verbrauch ist also **größer** als der Hausverbrauch. Durch das Abziehen reduziert sich somit der Hausverbrauch auf weniger als 0. Der Balken in SOLECTRUS ist also korrekterweise nicht vorhanden.

Die Ursache liegt somit in deinen Messwerten. Entweder ist der gemeldete Wallbox-Verbrauch zu groß oder der Hausverbrauch zu klein.

67-Peter, 2025-05-06:
Verstehe! 
Dann wird das Problem an den Wallbox-Daten liegen, denn die - habe ich gerade festgestellt - sind manchmal zu hoch.
EVCC Daten stimmen mit geeichter Keba Wallbox überein. Danke dir vielmals.