solectrus/solectrus #4321: Einspeisevergütung direkt anzeigen

ledermann, 2025-05-17:
Vorgeschlagen in #4318 von @Zeppelin-PV:

In Zeiträumen sollte die Einspeisevergütung mehr hervorgehoben werden (anstatt sie nur im Tooltip des Segments "Einspeisung") anzuzeigen). Statt der beiden Kreise **Einsparung** und **Kosten** könnten drei Informationen (Einspeisung, Kosten, Einsparung) in einem Kasten zusammengefasst werden, was in etwa den gleichen Platz beanspruchen würde.

[image]

Zeppelin-PV, 2025-05-17:
... schön !!!
Und warum nicht auf Heller & Pfennig ???

musste 2024 an den Versorger  536,12€ bezahlen
vom Netzbetreiber habe ich 587,96€ bekommen

... die zwei runden auch nicht auf/ab  ;-)

ledermann, 2025-05-17:
Ich habe das jetzt umgesetzt, zu sehen in `v0.19.1-98` und `multi-inverter-235`.

Die Darstellung gerundeter Beträge erfolgt aufgrund der Messungenauigkeit, der Fehler beträgt meiner Erfahrung nach um die 1 %. Cents würden eine Genauigkeit suggerieren, die nicht gegeben ist. 
Versorger und Netzbetreiber arbeiten hingegen mit geeichten Zählern.

closed 2025-05-17 by ledermann