solectrus/solectrus #4330: neu eingeführte Anzeige Vergütung fehlt auf dem Smartphone

Zeppelin-PV, 2025-05-19:
### Beschreibung

die Anzeige Vergütung ist auf dem Smartphone Samsung A54 5G nicht vorhanden
auf dem Tablet und am PC wird sie angezeigt
[image]

### Welche Version von SOLECTRUS verwendest du?

multi-inverter-247-ged0ad2fc

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Pi 5

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Deine vollständige `docker-compose.yml` oder `compose.yml`

```yaml

```

### Deine vollständige `.env`

```text

```

ledermann, 2025-05-19:
Die Darstellung der Vergütung erfolgt nur, wenn Platz vorhanden ist. Ich habe die Grenze jetzt etwas verschoben, sodass die Vergütung jetzt nur auf winzigen Displays ausgeblendet wird.
Zu sehen in `develop` ab `v0.19.1-249`.

closed 2025-05-19 by ledermann

Zeppelin-PV, 2025-05-19:
top - DANKE
musste aber erst auf develop umstellen, auf PR kam die 249 nicht

ledermann, 2025-05-19:
> musste aber erst auf develop umstellen, auf PR kam die 249 nicht

So ist es, `pr-4195` gibt es nicht mehr, siehe [#4152](https://github.com/orgs/solectrus/discussions/4152#discussioncomment-13193810).