solectrus/solectrus #4394: Fehler in Diagramm Hausakku

spirit0607, 2025-06-08:
### Beschreibung

In der Jahres- und Gesamtübersicht wird die Akku Beladung teilweise nicht angezeigt (siehe Bilder). In der Ansicht Monat- und Woche werden die Daten korrekt angezeigt.

[image]
[image]
[image]
[image]

### Welche Version von SOLECTRUS verwendest du?

0.20.0

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Synology NAS

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Deine vollständige `docker-compose.yml` oder `compose.yml`

```yaml
version: "3.7"

services:
  app:
    image: ghcr.io/solectrus/solectrus:latest
    labels:
      - 'com.centurylinklabs.watchtower.scope=solectrus'
    depends_on:
      - db
      - influxdb
      - redis
    links:
      - db
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - INSTALLATION_DATE
      - ELECTRICITY_PRICE
      - FEED_IN_TARIFF
      - ADMIN_PASSWORD
      - DB_HOST=db
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_SENSOR_INVERTER_POWER
      - INFLUX_SENSOR_HOUSE_POWER
      - INFLUX_SENSOR_GRID_IMPORT_POWER
      - INFLUX_SENSOR_GRID_EXPORT_POWER
      - INFLUX_SENSOR_CUSTOM_POWER_01
      - INFLUX_SENSOR_CUSTOM_POWER_02
      - INFLUX_SENSOR_CUSTOM_POWER_03
      - INFLUX_SENSOR_CUSTOM_POWER_04
      - INFLUX_SENSOR_CUSTOM_POWER_05
      - INFLUX_SENSOR_CUSTOM_POWER_06
      - INFLUX_SENSOR_CUSTOM_POWER_07
[... 455 lines cut ...]
MAPPING_11_TOPIC=hassio/waschmaschine/watt
MAPPING_11_MEASUREMENT=SENEC
MAPPING_11_FIELD=waschmaschine
MAPPING_11_TYPE=integer
MAPPING_12_TOPIC=hassio/gefrierschrank/watt
MAPPING_12_MEASUREMENT=SENEC
MAPPING_12_FIELD=gefrierschrank
MAPPING_12_TYPE=integer
MAPPING_13_TOPIC=hassio/itdach/watt
MAPPING_13_MEASUREMENT=SENEC
MAPPING_13_FIELD=itdach
MAPPING_13_TYPE=integer
MAPPING_14_TOPIC=hassio/kuehlschrank/watt
MAPPING_14_MEASUREMENT=SENEC
MAPPING_14_FIELD=kuehlschrank
MAPPING_14_TYPE=integer
MAPPING_15_TOPIC=hassio/spuelmaschine/watt
MAPPING_15_MEASUREMENT=SENEC
MAPPING_15_FIELD=spuelmaschine
MAPPING_15_TYPE=integer
MAPPING_16_TOPIC=hassio/fernseher/watt
MAPPING_16_MEASUREMENT=SENEC
MAPPING_16_FIELD=fernseher
MAPPING_16_TYPE=integer
```

ledermann, 2025-06-08:
Freut mich, dass es nun klappt.

Mit einer älteren Version des Dashboards hat es eher nichts zu tun. Überhaupt habe ich es so geregelt, dass bei Pull eines veralteten Docker-Tags (`:pr-3747` und andere) immer das `develop`-Image geliefert wird - damit möglichst niemand versehentlich auf einem uralten Beta-Stand hängen bleibt.

Ich sehe die Ursache eher beim Power-Splitter. Es wäre zu untersuchen, ob der bei dir [aktuell](https://github.com/solectrus/power-splitter/releases/tag/v0.6.1) ist und Zugriff auf alle benötigten Variablen hat, insbesondere auf `INFLUX_SENSOR_BATTERY_CHARGING_POWER`. Doku hier:
https://docs.solectrus.de/referenz/power-splitter/konfiguration.html

spirit0607, 2025-06-09:
Okay, alles klar, vielen Dank! Das mache ich!

spirit0607, 2025-06-09:
Irgendwas hat scheinbar nicht funktioniert. Ich musste auch den Befehl in `docker-compose kill -s USR1 power-splitter` ändern, da ich noch Docker V1 habe. Nach dem Befehl wurde lt. Protokoll auch irgendwas aus der Influx gelöscht. Dabei ist dann aber irgendwann der Container gestoppt worden. Habe ihn anschließend über die GUI in der Docker Anwendung auf der NAS neu gestartet, also nicht mit `docker-compose up -d` falls das wichtig sein sollte. Daraufhin habe ich nun diesen Log erhalten:

```
power-splitter_1      | Power Splitter for SOLECTRUS, Version 0.6.1, built at 2025-05-24 04:49:44 +0000
power-splitter_1      | Using Ruby 3.4.4 on platform x86_64-linux-musl
power-splitter_1      | Copyright (c) 2024-2025 Georg Ledermann <georg@ledermann.dev>
power-splitter_1      | https://github.com/solectrus/power-splitter
power-splitter_1      | 
power-splitter_1      | Accessing InfluxDB at http://influxdb:8086, bucket my-solectrus-bucket
power-splitter_1      | Sensor initialization started
power-splitter_1      |   - Sensor 'grid_import_power' mapped to 'SENEC:grid_power_plus'
power-splitter_1      |   - Sensor 'house_power' mapped to 'SENEC:house_power'
power-splitter_1      |   - Sensor 'heatpump_power' mapped to 'SENEC:heatpump_power'
power-splitter_1      |   - Sensor 'wallbox_power' mapped to 'SENEC:wallbox_charge_power'
power-splitter_1      |   - Sensor 'battery_charging_power' mapped to 'SENEC:bat_power_plus'
power-splitter_1      |   - Sensor 'custom_power_01' mapped to 'SENEC:trockner'
power-splitter_1      |   - Sensor 'custom_power_02' mapped to 'SENEC:waschmaschine'
power-splitter_1      |   - Sensor 'custom_power_03' mapped to 'SENEC:gefrierschrank'
power-splitter_1      |   - Sensor 'custom_power_04' mapped to 'SENEC:itdach'
power-splitter_1      |   - Sensor 'custom_power_05' mapped to 'SENEC:kuehlschrank'
power-splitter_1      |   - Sensor 'custom_power_06' mapped to 'SENEC:spuelmaschine'
power-splitter_1      |   - Sensor 'custom_power_07' mapped to 'SENEC:fernseher'
power-splitter_1      |   - Sensor 'house_power' excluded 'heatpump_power'
power-splitter_1      | Sensor initialization completed
power-splitter_1      | 
power-splitter_1      | Starting endless loop for processing current data...
power-splitter_1      | 
power-splitter_1      | 2025-06-09 17:51:17 +0000 - Processing day 2025-06-09
power-splitter_1      |   Pushing 214 records to InfluxDB
power-splitter_1      |   Sleeping for 3600 seconds...
power-splitter_1      | 
power-splitter_1      | 
power-splitter_1      | --- Restarting...
power-splitter_1      | Thread exited cleanly.
power-splitter_1      | 
power-splitter_1      | --- Deleting all records from InfluxDB measurement 'power_splitter'
power-splitter_1      | bundler: failed to load command: app.rb (app.rb)
power-splitter_1      | /usr/local/bundle/gems/influxdb-client-3.2.0/lib/influxdb2/client/default_api.rb:109:in 'InfluxDB2::DefaultApi#_request': Net::ReadTimeout with #<TCPSocket:(closed)> (InfluxDB2::InfluxError)
power-splitter_1      | 	from /usr/local/bundle/gems/influxdb-client-3.2.0/lib/influxdb2/client/default_api.rb:54:in 'InfluxDB2::DefaultApi#_request_json'
power-splitter_1      | 	from /usr/local/bundle/gems/influxdb-client-3.2.0/lib/influxdb2/client/delete_api.rb:67:in 'InfluxDB2::DeleteApi#_delete'
power-splitter_1      | 	from /usr/local/bundle/gems/influxdb-client-3.2.0/lib/influxdb2/client/delete_api.rb:53:in 'InfluxDB2::DeleteApi#delete'
power-splitter_1      | 	from /power-splitter/lib/flux/deleter.rb:16:in 'Flux::Deleter#delete'
power-splitter_1      | 	from /power-splitter/lib/flux/deleter.rb:6:in 'Flux::Deleter#delete_measurement'
power-splitter_1      | 	from /power-splitter/lib/influx_push.rb:35:in 'InfluxPush#delete_measurement'
power-splitter_1      | 	from /power-splitter/lib/loop.rb:121:in 'Loop#delete_all'
power-splitter_1      | 	from /power-splitter/lib/loop.rb:26:in 'block in Loop#start'
power-splitter_1      | 	from <internal:kernel>:168:in 'Kernel#loop'
power-splitter_1      | 	from /power-splitter/lib/loop.rb:17:in 'Loop#start'
power-splitter_1      | 	from app.rb:30:in '<top (required)>'
power-splitter_1      | 	from /usr/local/bundle/gems/bundler-2.6.9/lib/bundler/cli/exec.rb:59:in 'Kernel.load'
power-splitter_1      | 	from /usr/local/bundle/gems/bundler-2.6.9/lib/bundler/cli/exec.rb:59:in 'Bundler::CLI::Exec#kernel_load'
power-splitter_1      | 	from /usr/local/bundle/gems/bundler-2.6.9/lib/bundler/cli/exec.rb:23:in 'Bundler::CLI::Exec#run'
power-splitter_1      | 	from /usr/local/bundle/gems/bundler-2.6.9/lib/bundler/cli.rb:452:in 'Bundler::CLI#exec'
power-splitter_1      | 	from /usr/local/bundle/gems/bundler-2.6.9/lib/bundler/vendor/thor/lib/thor/command.rb:28:in 'Bundler::Thor::Command#run'
power-splitter_1      | 	from /usr/local/bundle/gems/bundler-2.6.9/lib/bundler/vendor/thor/lib/thor/invocation.rb:127:in 'Bundler::Thor::Invocation#invoke_command'
power-splitter_1      | 	from /usr/local/bundle/gems/bundler-2.6.9/lib/bundler/vendor/thor/lib/thor.rb:538:in 'Bundler::Thor.dispatch'
power-splitter_1      | 	from /usr/local/bundle/gems/bundler-2.6.9/lib/bundler/cli.rb:35:in 'Bundler::CLI.dispatch'
power-splitter_1      | 	from /usr/local/bundle/gems/bundler-2.6.9/lib/bundler/vendor/thor/lib/thor/base.rb:584:in 'Bundler::Thor::Base::ClassMethods#start'
power-splitter_1      | 	from /usr/local/bundle/gems/bundler-2.6.9/lib/bundler/cli.rb:29:in 'Bundler::CLI.start'
[... cut ...]
power-splitter_1      |   Pushing 217 records to InfluxDB
power-splitter_1      | Flushing Redis cache failed: Waited 1.0 seconds
power-splitter_1      | --- Processing historical data successfully finished
power-splitter_1      | 
power-splitter_1      | Starting endless loop for processing current data...
power-splitter_1      | 
power-splitter_1      | 2025-06-09 18:05:17 +0000 - Processing day 2025-06-09
power-splitter_1      |   Pushing 217 records to InfluxDB
power-splitter_1      |   Sleeping for 3600 seconds...
power-splitter_1      | 
```

spirit0607, 2025-06-11:
So, ich bin leider nicht früher dazu gekommen das ganze auszuprobieren. Ich habe es nun aber geschafft. Ich habe mit Hilfe von ChatPGT den Befehl nur leicht anders aufgebaut, da meine Synology mit deinem Befehl nicht klarkam (schätze mal eine zu alte Version von Docker) nun hat es mit `docker exec -it solectrus_influxdb_1 sh` und anschließend
```
influx delete \
  --bucket 'my-solectrus-bucket' \
  --org 'solectrus' \
  --token 'my-super-secret-admin-token' \
  --start '1970-01-01T00:00:00Z' \
  --stop '2025-06-11T10:42:00Z' \
  --predicate '_measurement="power_splitter"'
```

geklappt. Vielen lieben Dank nochmal für deine Hilfe. Nun kann das Issue aber auch wirklich geschlossen werden.

closed 2025-06-11 by spirit0607

ledermann, 2025-06-18:
Soeben habe ich [v0.20.1](https://github.com/solectrus/solectrus/releases/tag/v0.20.1) veröffentlicht - wer also auf `develop` nur wegen dieser Korrektur ist, kann jetzt wieder auf `latest` wechseln.