solectrus/solectrus #4402: Unterstützung mehrerer Batterien/Akkus

unclej84, 2025-06-13:
### Beschreibung (auf Deutsch oder Englisch)

Es wäre schön, wenn man mehrere Akkus unterstützen könnte mit Charge, Discharge und optional SOC. Viele Balkon-Systeme werden mittlerweile ja auch mit eigenem Speicher zur Optimierung des Eigenverbrauchs verkauft oder man erweitert den SENEC Akku durch andere zusätzliche Speicher-Lösungen. Es wäre schön diese mit tracken zu können.

niceguy0815, 2025-06-28:
Ich überlege mir auch an meinen Growatt MID 11 KTL3-XH der seine Daten an einen SolarAssisten Instanz übergiebt und diese via MQTT im Netz bereit stellt einen zweiten Speicher ink Backup Box zu installieren Da der Senec home v 3 Hybrid duo nur schlecht erweiterbar ist und dort die Backupfunktin erstens teuer und zweites Scheiße ist da sie nicht das ganze Haus versorgen kann.

Also wäre schon cool wenn man die dDaten des Speichers auch nach Solectrus bringen könnte.
Aber das ist technisch warscheinlich nicht so einfach.

ledermann, 2025-10-18:
Danke für die allseitigen Bedarfsbekundungen :) Grundsätzlich ist das aufgrund des großen [Refactorings](https://github.com/solectrus/solectrus/pull/4793) auch leichter machbar, zeitlich möchte ich aber erst mal die [Version 1.0](https://github.com/orgs/solectrus/discussions/4805) fertigstellen, bevor ich weitere Neuerungen angehe.

Zeppelin-PV, 2026-09-10:
Bin immer noch in Lauerstellung bei der Anschaffung des Marstek venus e 3.0, der sicherlich im Preis nachgibt wenn demnächst der 4.0 am Markt ist.  Außer dem Wunsch diesen schön und vor allem richtig in Solectrus zu integrieren stelle ich mir die Frage - wie sich das auf das neue Feature auswirken würde bei der Bilanzkontrolle zwischen Herkunft und Verwendung .
Hauptaufgabe des 5kWh Speichers wäre diesen mit Tibber zu füllen und gezielt über HA zu leeren wenn dem Senec die Luft ausgeht

[image]

ledermann, 2026-09-10:
@Zeppelin-PV Solange dieses Issue nicht realisiert ist, muss man selbst dafür sorgen, dass die betroffenen Messwerte außerhalb von SOLECTRUS korrekt zusammengeführt werden. Es geht dabei um drei Sensoren:

* Beladung: `battery_charging_power`
* Entladung: `battery_discharging_power`
* Ladestand: `battery_soc`

Hat man mehrere Akkus, muss man also in etwa so rechnen:

* Bei Beladung und Entladung werden die jeweiligen Leistungen der Akkus **addiert**.
* Beim Ladestand wird ein **nach Speicherkapazität gewichteter Mittelwert** aus den Ladeständen der Akkus gebildet.

Für SOLECTRUS stellt sich das Ergebnis dann wie ein einzelner Akku dar.

Wird beispielsweise der zweite Akku aus dem Netz geladen, muss sich dieser Bezug natürlich auch im `grid_import_power` widerspiegeln. Dann bleibt die Strombilanz weiterhin ausgeglichen: Der zusätzliche Netzbezug steht der zusätzlichen Akkuladung gegenüber. Beim späteren Entladen gilt entsprechend das Umgekehrte.

Die Bilanzkontrolle von SOLECTRUS (siehe #5890) weist darauf hin, wenn Herkunft und Verwendung des Stroms deutlich voneinander abweichen. Eine solche Abweichung deutet auf unvollständige oder inkonsistente Messwerte hin und kann zu verschiedenen Folgefehlern führen. Das sollte daher möglichst vermieden werden.

Gabinger, 2026-11-10:
Hey zusammen,
ich wäre auch noch im Team 2. Speicher. Habe meine Senec Anlage erweitert bzw. einen 2. Sungrow WR mit einem 2. Speicher BYD parallel. Idealerweise dann auch mit +- Erkennung, da es nur einen Wert gibt bei Be-/Entladung mit Vorzeichenwechsel.... Vielleicht geht ja was :) (Ist das eigentlich so untypisch? Ich glaube das kommt eher vor als 5 E-Autos ;))

ledermann, 2026-11-11:
@Gabinger Zu den "5 E-Autos": Die sind nicht unbedingt für fünf Autos gleichzeitig gedacht, sondern vor allem für mehrere nacheinander. Wer das Auto wechselt, behält die Historie des alten. Viel Arbeit war auch nicht die Zahl fünf, sondern der Schritt von einem Auto auf mehrere. Genau dieser Schritt fehlt bei den Akkus noch. Übrigens: Laut Fragebogen in der Registrierung haben etwa gleich viele Nutzer einen zweiten Akku wie ein zweites E-Auto :-)

Warum das mit den mehreren Akkus gar nicht so trivial ist:

- **Ladestand:** Er muss nach Kapazität gewichtet werden, und zwar für jeden einzelnen Messpunkt. Sonst wäre kein gemeinsamer Ladestand-Verlauf darstellbar.
- **Power-Splitter:** Er liest die Akku-Werte direkt aus der InfluxDB und muss selbst lernen, mehrere Akkus zusammenzuzählen.
- **Ingest:** Ein Zusatzakku hinter dem Zähler verfälscht den Hausverbrauch, wie ein Balkonkraftwerk. Ingest muss ihn in die Korrektur einrechnen.
- **HELIOS:** Die neuen Sensoren und die Kapazitäten müssen dort einstellbar sein.
- **Oberfläche:** Diagramme, Tooltips und Einstellungen kennen heute nur einen Akku. Dazu käme eine eigene Seite für die Details.

Es geht also um **vier** Projekte, die zusammenpassen müssen. Bis das umgesetzt ist, bleibt der Workaround von oben: die Werte außerhalb von SOLECTRUS zusammenführen (Details in #5190).

Momentan kümmere ich mich um die [Fertigstellung der v1.4](https://github.com/orgs/solectrus/discussions/5966) - danach geht es auf zu neuen Ufern, da rangeln auch bereits verschiedene größere Themen um bevorzugte Behandlung...

@Zeppelin-PV Ein besonderer Dank an dich: Du unterstützt die Entwicklung ganz ohne Aufhebens mit häufigen Zusatz-Sponsorings. Zur [Nachahmung](https://buy.stripe.com/3csdRy0izf584Q8000) empfohlen :-)

Gabinger, 2026-11-11:
Hey,
ja war auch weniger als Kritik gemeint. Dazu kenn ich mich auch zu wenig aus um das zu beurteilen. Ich habe deine Empfehlungen umgesetzt und habe jetzt das richtige Ergebnis für meine zwei Anlagen. Danke dafür!
Nächste Herausforderung wird wohl die Wärmepumpe....
Grüße