solectrus/solectrus #4442: Top10: Bessere Gruppierung der Sensor-Auswahl

ledermann, 2025-06-28:
Das Dropdown-Menü bietet die Sensoren alphabetisch sortiert an. Das ist aber nicht ideal, da dies zu einer Vermischung von sehr unterschiedlichen Dingen führt. Eine gruppierte Darstellung wäre besser, z.B. so:

- Erzeugung (Gesamt und einzeln)
- Netz (Bezug und Einspeisung)
- Akku (Beladung und Entladung)
- Verbraucher (Hausverbrauch, Wärmepumpe, E-Auto und benutzerdefinierte)

Vorgeschlagen von @Zeppelin-PV in #4439

Zeppelin-PV, 2025-06-28:
... müsste es da nicht auch noch die Möglichkeit "Erzeugung-Gesamt" geben ?
ansonsten passt das genau zu meiner Vorstellung - Super - DANKE
[image]

Zeppelin-PV, 2025-06-28:
DANKE
jetzt passt's !!!