solectrus/solectrus #4456: Chart des aktuellen Tages: Zukunft wird auf Nullinie gezeichnet

gereons, 2025-07-03:
### Beschreibung

In meiner Solectrus-Instanz wird für den aktuellen Tag die Erzeugung als eine Nullinie gezeichnet und die Erzeugung auch als "0" im Tooltip angeben:

[image]

Das ist in der Demo-Instanz bei gleichem Softwarestand (v0.20.1-45-g463af19) anders, und nach meiner Erinnerung auch neu bei mir. Woran kann das liegen? 

### Welche Version von SOLECTRUS verwendest du?

v0.20.1

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Cloud-Server

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text

```

### Deine vollständige `docker-compose.yml` oder `compose.yml`

```yaml

```

### Deine vollständige `.env`

```text

```

ledermann, 2025-07-03:
Gut gesehen, danke für den Hinweis!

Das ist eine Regression einer gestrigen Änderung, die nur bei bestimmten Konfigurationen auftritt und `nil` als `0` auffasst. Ist jetzt korrigiert, siehe `v0.20.1-47`.

closed 2025-07-03 by ledermann

gereons, 2025-07-03:
Danke für den schnellen Fix, sieht wieder alles gut aus bei mir 👍